thesofproject/sof #4606: [BUG][BUG] dai_report_xrun(): underrun due to no data available when pause release on TGLU_RVP_NOCODEC_CI

XiaoyunWu6666, 2021-08-06:
**Describe the bug**
in inner daily 5738
 dai_report_xrun(): underrun due to no data available when pause release on TGLU_RVP_NOCODEC_CI

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec-ci.tplg ~/sof-test/test-case/multiple-pause-resume.sh -r 25

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: b7329735
SOF Branch: main
SOF Commit: d486a9ff9896
Topology:sof-tgl-nocodec-ci.tplg 
 Platform:  TGLU_RVP_NOCODEC_CI

**Screenshots or console output**
[console]
```
=== PAUSE ===                                                            
(25/25) pcm'DMIC' cmd'arecord' id'6': Wait for 145 ms before pause

(25/25) pcm'DMIC' cmd'arecord' id'6': Wait for 156 ms before resume

2021-08-05 22:24:56 UTC N pipeline: Port0 with aplay
2021-08-05 22:24:56 UTC N pipeline: DMIC with arecord
2021-08-05 22:24:56 UTC N Check expect exit status
declare -- cmd="journalctl_cmd --since=@1628202151"
2021-08-05 22:24:57 UTC [REMOTE_ERROR] pause resume PID 26233 had non-zero exit status
```
[etrace]
```
[  1183287404.022050] (     1386687.750000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[  1183287421.990799] (          17.968750) c0 dai          1.6            src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[  1183287439.282465] (          17.291666) c0 dai          1.6            src/audio/dai.c:675  ERROR comp_underrun(): dev->comp.id = 6, source->avail = 672, copy_bytes = 0
[  1183287578.449126] (         139.166656) c0 pipe         1.7   ....../pipeline-stream.c:170  ERROR pipeline_copy(): ret = -61, start->comp.id = 6, dir = 1
[  1183287597.407459] (          18.958332) c0 pipe         1.7   ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[  1184491415.692957] (     1203818.250000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[  1184491433.870039] (          18.177082) c0 dai          1.6            src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[  1184491450.849205] (          16.979166) c0 dai          1.6            src/audio/dai.c:675  ERROR comp_underrun(): dev->comp.id = 6, source->avail = 768, copy_bytes = 0
[  1184491578.609617] (         127.760414) c0 pipe         1.7   ....../pipeline-stream.c:170  ERROR pipeline_copy(): ret = -61, start->comp.id = 6, dir = 1
[  1184491597.463783] (          18.854166) c0 pipe         1.7   ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[  1185707408.509221] (     1215811.000000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[  1185707426.530053] (          18.020832) c0 dai          1.6            src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[  1185707443.977969] (          17.447916) c0 dai          1.6            src/audio/dai.c:675  ERROR comp_underrun(): dev->comp.id = 6, source->avail = 768, copy_bytes = 0
[  1185707573.040464] (         129.062500) c0 pipe         1.7   ....../pipeline-stream.c:170  ERROR pipeline_copy(): ret = -61, start->comp.id = 6, dir = 1
[  1185707592.155047] (          19.114582) c0 pipe         1.7   ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
Skipped 5720 bytes after the last statement.
```

lgirdwood, 2021-08-06:
@XiaoyunWu6666 can you always paste the 15 log messages before the first ```ERROR``` log message. This gives the devlopers some context prior to the error. Thanks.

XiaoyunWu6666, 2021-09-10:
This issue happened again in today's inner daily 

First happened on 2021-08-22T21:21:02.680Z (Report Id: 5994)
Last happened on 2021-09-09T21:24:54.582Z (Report Id: 6472)
3 hits within 19 days totally

keqiaozhang, 2021-12-13:
This issue is fixed on main and v2.0 branch.

closed 2021-12-13 by keqiaozhang