thesofproject/sof #4609: [BUG] wrong TGL nocodec CI topology

lyakh, 2021-08-06:
**Describe the bug**
Multiple links to a buffer sink in topology1/development/sof-tgl-nocodec-ci.m4 

BUF7.0 has two components sending data to it: "ECHO REF 7" and "MUXDEMUX1.0"
[image]

lyakh, 2021-08-06:
APL demux topology https://github.com/thesofproject/sof/blob/main/tools/topology/topology1/sof-apl-demux-pcm512x.m4 has the same bug:
[image]

lyakh, 2021-08-06:
https://github.com/thesofproject/sof/blob/main/tools/topology/topology1/sof-cml-demux-rt5682-max98357a.m4 and https://github.com/thesofproject/sof/blob/main/tools/topology/topology1/sof-cml-demux-rt5682.m4 have the same problem

lgirdwood, 2021-08-06:
Looks like a copy and paste error.
1) TGL needs a MUX as BUF7.0 source that muxes ECHOREF7 and MUXDEMUX1.0
2) APL  needs a MUX as BUF5.0 source that muxes ECHOREF5 and MUXDEMUX1.0

plbossart, 2021-08-06:
This reminds me of a thread with @juimonen and @ranj063 
I think the conclusion was it was a feature not a bug. Something to do with pipeline scheduling.

iuliana-prodan, 2021-08-06:
> This reminds me of a thread with @juimonen and @ranj063
> I think the conclusion was it was a feature not a bug. Something to do with pipeline scheduling.
> 
> Edit: I still think this sort of topology is "wrong", we need to move to 'host' and 'dai' pipelines, and connections between the two parts are handled with mixers, mux, demux, etc.

The discussion was here: https://github.com/thesofproject/sof/issues/4207

plbossart, 2021-08-06:
> > This reminds me of a thread with @juimonen and @ranj063
> The discussion was here: #4207

Yep. that's the one, thanks @iuliana-prodan

keyonjie, 2021-08-10:
@juimonen I don't think every pipeline has to have both 'host' and 'dai' component, if you check the mixer pipelines and WoV pipelines you will find one of it doesn't have a 'dai', so here the 'SSP5.IN' and 'ECHO REF 5' can be removed.

juimonen, 2021-08-10:
@keyonjie if I remember correctly, I tried to remove those and it just didn't work... but not sure. AFAIK this was coming from some ASoC layer code that there needs to be FE and BE in every pipeline. But needs to be checked again, things might have changed.

lgirdwood, 2021-10-15:
@lyakh probably can go in your bag for mixer, mux, smart amp updates to fix pipeline changes.