thesofproject/sof #4614: [BUG]ipc timed out when runtime suspend on TGLU_RVP_NOCODEC_MULTICORE

XiaoyunWu6666, 2021-08-09:
**Describe the bug**
found in inner daily 5772?model=TGLU_RVP_NOCODEC_CI&testcase=check-suspend-resume-50
Accordingly in round 28/50, when runtime suspend , ipc timed out when COMP_FREE

**To Reproduce**
 TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec-ci.tplg ~/sof-test/test-case/check-suspend-resume.sh -l 50
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: b7329735
SOF Branch: main
SOF Commit: f7982d88e3d8
Topology: sof-tgl-nocodec-ci.tplg
 Platform: TGLU_RVP_NOCODEC_CI

**Screenshots or console output**
[dmesg]
```shell
[ 6099.059006] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget BUF5.1 freed
[ 6099.059008] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 6099.059133] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 6099.059134] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget BUF5.2 freed
[ 6099.059136] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 6099.059301] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 6099.059302] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget SSP2.OUT freed
[ 6099.059304] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[ 6099.059916] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 6099.059933] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 6099.060004] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[ 6099.060005] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.5.SSP2.OUT freed
[ 6099.060007] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 6099.560079] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 6099.560192] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 6099.564125] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30020000 size 12
[ 6099.564140] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 6099.564145] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 6099.564162] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[ 6099.564169] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 6099.564178] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 6099.564183] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 6099.564188] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 6099.564209] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 6099.564559] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 6099.564565] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 6099.564571] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to free widget PCM2C
[ 6099.564580] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PCM2C freed
[ 6099.564593] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x20140000 successful
[ 6099.619891] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 6100.017598] kernel: asix 3-8.1:1.0 enx000ec667532b: Link is Up - 100Mbps/Full - flow control off
[ 6100.060066] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 6100.060189] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 6100.123196] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[ 6100.123214] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
[ 6100.123257] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x7010f0f successful
[ 6100.123288] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
[ 6100.123312] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 6100.123449] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
[ 6100.124123] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Current DSP power state: D3
```

keyonjie, 2021-08-10:
Looks it fails during the tearing down here, @ranj063 FYI.

XiaoyunWu6666, 2021-08-31:
in inner daily 6200 , on model TGLU_RVP_NOCODEC_CI , in case check-suspend-resume-with-playback-5
ipc timed out when COMP_FREE and brought DSP panic when freeing widget PCM2P
[dmesg]
```
[ 3013.270563] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 3013.770619] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 3013.770661] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 3013.774003] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x30020000 (msg/reply size: 12/12)
[ 3013.774014] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 3013.774018] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 3013.774032] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[ 3013.774037] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 3013.774044] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 3013.774048] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 3013.774051] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 3013.774057] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 3013.774431] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 3013.774435] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 3013.774440] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to free widget PCM2P
[ 3013.774450] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[ 3014.270586] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 3014.270637] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 3014.277985] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x40070000 (msg/reply size: 12/12)
[ 3014.277999] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to disable target core: 1 for widget PCM2P
[ 3014.278016] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x20140000 successful
[ 3014.278052] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 3014.781938] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x40010000 (msg/reply size: 76/12)
[ 3014.781950] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
```
[dmesg.txt](https://github.com/thesofproject/sof/files/7081418/dmesg.txt)
[sof-logger]
```
[       27689.061400] (         187.708328) c0 ipc                   src/ipc/ipc3/helper.c:585  ERROR ipc_comp_connect(): source component does not exist, source_id = 2 sink_id = 40
```

lgirdwood, 2021-08-31:
@XiaoyunWu6666 for some reason your logs above are missing the important information. i.e. from your attached dmesg
```
[ 3019.078660] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: setting up route PCM2P -&gt; BUF5.0
[ 3019.078662] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[ 3019.078795] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[ 3019.078797] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: setting up route MUXDEMUX1.0 -&gt; BUF7.0
[ 3019.078799] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
<span class="txtRed">[ 3019.078880] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30030000 (msg/reply size: 16/12): -22</span>
<span class="txtRed">[ 3019.078882] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: sof_route_setup_ipc: route setup failed -22</span>
<span class="txtRed">[ 3019.078883] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: sof_set_up_pipelines: restore pipeline connections failed</span>
<span class="txtRed">[ 3019.078884] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to restore pipeline after resume -22</span>
[ 3019.078885] kernel: PM: dpm_run_callback(): pci_pm_resume+0x0/0x80 returns -22
<span class="txtRed">[ 3019.078890] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: PM: failed to resume async: error -22</span>
<span class="txtRed">[ 3019.182993] kernel: atkbd serio0: Failed to deactivate keyboard on isa0060/serio0</span>
<span class="txtRed">[ 3019.559114] kernel: atkbd serio0: Failed to enable keyboard on isa0060/serio0</span>
[ 3019.575723] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 3019.575773] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 3019.578528] kernel: asix 3-8.1:1.0 enx000ec667532b: Link is Down
[ 3019.665026] kernel: OOM killer enabled.
[ 3019.665027] kernel: Restarting tasks ... done.
[ 3019.667400] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 3019.667759] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 3019.667774] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: prepare stream 2 dir 0
[ 3019.667776] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: hw params stream 2 dir 0
[ 3019.667777] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: generating page table for 00000000f6a8f163 size 0x10000 pages 16
[ 3019.667790] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x40000 successful
[ 3019.667832] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x40000 successful
[ 3019.667834] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda_dsp_stream_setup_bdl: period_bytes:0x4000
[ 3019.667836] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda_dsp_stream_setup_bdl: periods:4
[ 3019.667855] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: stream_tag 1
[ 3019.667864] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: complete pipeline PIPELINE.5.SSP2.OUT id 32
[ 3019.667867] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[ 3019.668582] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[ 3019.668602] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
<span class="txtRed">[ 3019.668918] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error : DSP panic!</span>
<span class="txtRed">[ 3019.668930] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480</span>
[ 3019.668932] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 3019.668945] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
<span class="txtRed">[ 3019.669048] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: runtime exception</span>
<span class="txtRed">[ 3019.669057] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace point 00004000</span>
<span class="txtRed">[ 3019.669067] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: panic at :0</span>
<span class="txtRed">[ 3019.669075] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops</span>
<span class="txtRed">[ 3019.669085] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access</span>
[ 3019.669100] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000040 PS       0x00060025 SAR     0x00000000
[ 3019.669115] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe045be5 EPC2     0xbe0467f2 EPC3     0x00000000 EPC4    0x00000000
[ 3019.669129] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 3019.669143] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00060920 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[ 3019.669156] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[ 3019.669170] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0xbe119900
[ 3019.669179] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe119900: 00000004 00000001 00000002 9e1fd600
[... cut ...]
[ 3019.669236] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe119914: 809f4400 ffff9c09 83fc0400 ffff9c09
[ 3019.669247] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe119918: a55e6200 fa20a28f 91acffa0 ffffffff
[ 3019.669256] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe11991c: 8093ca80 ffff9c09 91a8d710 ffffffff
[ 3019.669264] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 3019.672193] kernel: rfkill: input handler enabled
[ 3019.680071] kernel: PM: suspend exit
```
This looks like another NULL ptr dereference on stream params.

XiaoyunWu6666, 2021-09-06:
> @XiaoyunWu6666 for some reason your logs above are missing the important information. i.e. from your attached dmesg

@lgirdwood  Got it , I think I missed the latest DSP dump information , will take care next time I paste logs . thanks

XiaoyunWu6666, 2021-09-06:
in inner daily 6328,on model=TGLU_RVP_NOCODEC_CI, in testcase=check-suspend-resume-with-playback-5
ipc timed out happened again when COMP_FREE , with  ipc tx error when COMP_CONNECT  and DSP panic afterwards (showing _Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access_ also)
[dmesg.txt](https://github.com/thesofproject/sof/files/7112804/dmesg.txt)

lgirdwood, 2021-09-07:
@XiaoyunWu6666 thanks for update.

One more super useful thing todo with any crash dump that has non zero values in the ```EPCn``` registers (where n is between 1 and 7) is to provide the disassembly of the code. This is very helpful as it saves a lot of time for developers (and since you already have all the files needed). i.e. the registers show where we have crashed in the EPC regs.
```
[ 3019.669100] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000040 PS       0x00060025 SAR     0x00000000
[ 3019.669115] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe045be5 EPC2     0xbe0467f2 EPC3     0x00000000 EPC4    0x00000000
```
1) We see that EPC1 is contains a non zero address ```EPC1     0xbe045be5```
2) Remove the ```0x``` and the last two digits from EPC1 to get ```be045b``` and to give us 6 digit address.
3) Next search for the address by running  ```grep -rn be045b src/arch/xtensa/sof-tgl.dis``` where the ```.dis``` file is in the build directory for this target. 
4) Repeat with other EPC registers that are not zero.
@keqiaozhang  fyi too.

keqiaozhang, 2021-12-13:
Confirmed that this issue is fixed with latest main and stable v2.0 branch.

closed 2021-12-13 by keqiaozhang