thesofproject/sof #4621: [BUG]pcm create failure on multiple  ADL TGL platforms

XiaoyunWu6666, 2021-08-11:
**Describe the bug**
found in inner daily 5802, on multiple ADL TGL platforms including **ADLP_RVP_SDW ADLP_RVP_HDA TGLU_RVP_NOCODEC_CI 
TGLU_VOLT_SDW**

**To Reproduce**
boot the DUT with topology built by recipe built below ,  DUT failed to set up soundcards

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 2513c3d1
SOF Branch: main
SOF Commit: 9edfcfc03622
Topology and  Platform:sof-hda-generic-4ch-kwd for ADLP_RVP_HDA ,sof-adl-rt711-4ch for ADLP_RVP_SDW, sof-tgl-nocodec-ci for TGLU_RVP_NOCODEC_CI ,sof-tgl-sdw-max9837-rt5682 for  TGLU_VOLT_SDW

**Screenshots or console output**
take ADLP_RVP_HDA as example,when booting with given topology
[console]
```
[    7.983989] kernel: skl_hda_dsp_generic skl_hda_dsp_generic: No CPU DAIs support capture for stream DMIC
[    7.983994] kernel: skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: can't create pcm DMIC :-22
```

XiaoyunWu6666, 2021-08-11:
Reverting commit  "topology: remove side effects from macro definitions"  659266685b4358ded2f779b29d34166432108f71 can make this issue disappear. 
```shell
commit 659266685b4358ded2f779b29d34166432108f71 
Author: Bud Liviu-Alexandru <budliviu@gmail.com>
Date:   Thu Aug 5 00:26:42 2021 +0300

    topology: remove side effects from macro definitions

    This commit fixes the issue mentioned in #4583.

    A temporary definition is introduced that takes it's value from
    "CHANNELS_MIN" if it is already defined, otherwise it is set to
    a default value. This temporary is used instead of "CHANNELS_MIN"
    for the rest of the file. This avoids having the macro defined in
    files where it shouldn't be.

    In order to be completely sure it avoids side effects this temporary
    is undefined after it is no longer needed.

    Signed-off-by: Bud Liviu-Alexandru <budliviu@gmail.com>

```

XiaoyunWu6666, 2021-08-12:
hi @kv2019i could you please help to check why 659266685b4358ded2f779b29d34166432108f71 affect both DMIC and Bluetooth pcm creation , thanks so much

kv2019i, 2021-08-12:
Ack @XiaoyunWu6666  found the rootissue, will submit a patch soon.

XiaoyunWu6666, 2021-08-13:
thanks @kv2019i 's quick fix https://github.com/thesofproject/sof/pull/4630 , today's inner daily 5850 is good .
Let's close this issue.

closed 2021-08-13 by XiaoyunWu6666