thesofproject/sof #4629: [BUG] sof doesn't load properly on ALC3204

nklayman, 2021-08-12:
**Describe the bug**
sof doesn't initialized properly on boot so there are no input or output options in the sound settings.

**To Reproduce**
On a Dell Inspiron 16 Plus 7610 (Tiger Lake H, RTX 3000), fresh install of Fedora 34. Boot computer, check sound settings, see no outputs. Error messages printed in `dmesg`. Disabling sof with `options snd-intel-dspcfg dsp_driver=1` in `modprobe.d` makes speakers work, but mic is broken then.

**Reproduction Rate**
Happens every time. 

**Expected behavior**
sof should load properly and audio inputs/outputs should display in sound settings. There should not be errors in `dmesg` logs.

**Impact**
Audio doesn't work whatsoever on sof, and disabling sof only fixes speakers.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.13.8-200.fc34.x86_64 (default kernel on fedora 34)
    * SOF: 1:8:0-9e7a8 (latest sof version on fedora)
2) Name of the topology file
    * Topology: not sure how to get this
3) Name of the platform(s) on which the bug is observed.
    * Platform: Fedora 34

**Screenshots or console output**
Ouput from `dmesg | grep sof-audio-pci`:
```
[  177.595462] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[  177.595476] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[  177.595487] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[  177.595619] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[  177.595718] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[  177.602070] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[  177.623248] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[  177.623252] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[  177.623255] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[  177.624470] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 1:8:0-9e7a8
[  177.624472] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[  177.624476] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[  177.702816] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 1:8:0-9e7a8
[  177.702819] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[  177.708102] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:18:1 Kernel ABI 3:18:0
[  177.725290] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[  186.414478] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  186.414483] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  186.414484] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  186.414912] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  186.414916] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  186.414919] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  186.415341] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  186.415345] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  186.415347] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  186.415854] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  186.415858] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  186.415861] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  186.416293] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  186.416296] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  186.416299] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  186.416809] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  186.416814] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  186.416817] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  186.417271] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  186.417275] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  186.417277] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  186.417697] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  186.417701] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  186.417704] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  186.418132] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  186.418136] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  186.418138] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[... 14 lines cut ...]
[  191.015261] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  191.015682] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  191.015683] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  191.015684] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  191.016240] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  191.016241] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  191.016242] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  191.016668] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  191.016670] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  191.016670] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  191.017104] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  191.017106] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  191.017106] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  191.017553] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  191.017554] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  191.017555] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[  191.017969] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  191.017972] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[  191.017975] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
```

Full dmesg log: [dmesg.txt](https://github.com/thesofproject/sof/files/6972955/dmesg.txt)
Not sure how to get sof log, will be happy to update if someone sends instructions.

Thanks for taking a look at this!

plbossart, 2021-08-12:
oh and I forgot @nklayman if you can upload the results of alsa-info.sh that would be useful. It looks like you have a 'standard' HDaudio codec+dmic configuration but there might be additional details for us to look at. Thanks!

nklayman, 2021-08-12:
@plbossart Thanks for the quick response, here's the information you asked for:

`dmesg | grep sof-audio-pci` without display connected:
```
[   11.864091] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   11.864198] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   11.864233] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[   11.864366] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   11.864481] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   11.871282] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[   11.893348] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[   11.893350] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   11.893353] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[   11.894640] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 1:8:0-9e7a8
[   11.894643] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[   11.894648] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[   11.974553] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 1:8:0-9e7a8
[   11.974555] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[   11.980013] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:18:1 Kernel ABI 3:18:0
[   11.997009] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   22.358099] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   22.358105] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   22.358106] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   22.358571] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   22.358573] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   22.358574] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   22.359097] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   22.359099] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   22.359100] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   22.359602] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   22.359604] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   22.359605] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   22.360102] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   22.360103] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   22.360104] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   22.360667] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   22.360671] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   22.360675] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   22.361152] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   22.361154] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   22.361155] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   22.361611] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   22.361616] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   22.361619] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   22.362116] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   22.362118] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   22.362119] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   22.362560] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   22.362565] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   22.362567] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   71.077322] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   71.077333] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   71.077334] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   71.077781] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   71.077783] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   71.077784] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   71.078238] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[... cut ...]
[   76.410640] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
```
Full dmesg log: [dmesg.txt](https://github.com/thesofproject/sof/files/6977230/dmesg.txt)
The monitor plug in error is gone, but all the other errors remain.

Alsa info output (I used script from [here](https://www.alsa-project.org/wiki/AlsaInfo)): [alsa-info.txt](https://github.com/thesofproject/sof/files/6977377/alsa-info.txt)

After enabling dynamic debugging (still without display):
[dmesg.txt](https://github.com/thesofproject/sof/files/6977397/dmesg.txt)

Thanks for your help!

plbossart, 2021-08-12:
ok, so it's an error on DMIC capture

````
[   24.182200] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: open stream 6 dir 1
[   24.182201] sof-audio-pci-intel-tgl 0000:00:1f.3: period min 192 max 16384 bytes
[   24.182201] sof-audio-pci-intel-tgl 0000:00:1f.3: period count 2 max 16
[   24.182202] sof-audio-pci-intel-tgl 0000:00:1f.3: buffer max 65536 bytes
[   24.182257] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: hw params stream 6 dir 1
[   24.182258] sof-audio-pci-intel-tgl 0000:00:1f.3: generating page table for 00000000a306cc1b size 0x4af4 pages 5
[   24.182296] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   24.182309] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   24.182310] sof-audio-pci-intel-tgl 0000:00:1f.3: period_bytes:0x18fc
[   24.182311] sof-audio-pci-intel-tgl 0000:00:1f.3: periods:3
[   24.182318] sof-audio-pci-intel-tgl 0000:00:1f.3: stream_tag 2
[   24.182322] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000
[   24.182632] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   24.182634] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[   24.182634] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   24.182638]  DMIC: ASoC: soc_pcm_hw_params() failed (-12)
[   24.182641]  DMIC: ASoC: dpcm_fe_dai_hw_params failed (-12)
````

we may need to use the firmware trace to figure out what happens at the firmware level.

Please add this to /etc/modprobe.d/alsa-base.conf
options snd_sof sof_debug=1

If 'sof-logger' is not installed on your system, install it to e.g. /usr/local/bin from https://github.com/thesofproject/sof-bin/tree/main/v1.8.x/tools-v1.8 

And last, extract the firmware trace with 
sudo sof-logger -t -l /usr/lib/firmware/intel/sof-v1.8/sof-tgl-h.ldc | tee sof-log.txt

and attach sof-log.txt

I realize it's painful to do all this but it'll help us root cause what happens. Thank you for your time.

nklayman, 2021-08-12:
I created the `alsa-base.conf` file and added the line and rebooted, but I couldn't find the `.ldc` file. Inside of `/usr/lib/firmware/intel` there is only `sof` and `sof-tplg` (plus other non-sof stuff), no sof-v1.8. The `sof` dir has a `sof-tgl-h.ri` file, but that gave an `invalid ldc file signature` error. I also searched the entire system for a `.ldc` file but nothing showed up. Any ideas?

plbossart, 2021-08-12:
you do use firmware v1.8, this is what I see in the logs
````
Firmware info: version 1:8:0-9e7a8
````

Maybe your distro didn't pick the ldc files. Try to get it from https://github.com/thesofproject/sof-bin/blob/main/v1.8.x/sof-v1.8/sof-tgl-h.ldc 

if this doesn't work, pick both the .ri and .ldc files and add them in your firmware directory (do a backup first so that you don't wreck your distro and can revert that change).

the .ri file is the binary, and the .ldc contains a dictionary to translate error codes into human-readable strings. I have no idea why we split the two files, it generates more work for distros/users, but it is what it is...

nklayman, 2021-08-12:
I downloaded the file you linked and it seemed to work, here's the output from the `sof-log` command:
[sof-log.txt](https://github.com/thesofproject/sof/files/6977862/sof-log.txt)

I did plug the display in since booting and before running that command, is that an issue?

plbossart, 2021-08-12:
@nklayman I don't see any errors or smoking gun in the firmware trace. Did you get an error on the kernel side when you tried?

plbossart, 2021-08-12:
did you see an 'ipc error' message in the dmesg log prior to running the sof-logger command?

plbossart, 2021-08-12:
Note that in some cases enabling the trace makes things work by changing timings, that's why I was asking the question.

nklayman, 2021-08-12:
I didn't save the dmesg log from that boot unfortunately. Here's the dmesg and sof-log from the current boot (with monitor):
[sof-log.txt](https://github.com/thesofproject/sof/files/6978165/sof-log.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/6978166/dmesg.txt)

This time the sof-log has a ton of error messages. I'm gonna try rebooting a few times and see if I can get no errors to show up in the sof log again and see if sound works then.

plbossart, 2021-08-12:
Thanks @nklayman now we have enough information to ask @lgirdwood @ranj063 @keyonjie for help. I think we saw this before, looks like we need a hot fix for v1.8...

sof-logger output
````
[    42651132.576030] (          27.343748) c0 dai          10.70          src/audio/dai.c:357  dai_data_config() dai type = 2 index = 0 dd 0x9e122200
[    42651150.180196] (          17.604166) c0 dai          10.70          src/audio/dai.c:376  config->dmic.fifo_bits = 32 config->dmic.num_pdm_active = 1
[    42651192.732277] (          42.552082) c0 buffer                   src/audio/buffer.c:52   ERROR buffer_alloc(): could not alloc lock
[    42651209.398943] (          16.666666) c0 dai          10.70          src/audio/dai.c:638  ERROR dai_params(): failed to alloc dma buffer
[    42651229.034359] (          19.635416) c0 pipe         10.71 ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 64
[    42651248.096858] (          19.062500) c0 ipc                  src/ipc/handler-ipc3.c:284  ERROR ipc: pipe 10 comp 64 params failed -12
[    42651265.284358] (          17.187500) c0 pipe         10.71 ......./pipeline-graph.c:383  pipe reset
````

dmesg
````
[   26.140487] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: open stream 6 dir 1
[   26.140487] sof-audio-pci-intel-tgl 0000:00:1f.3: period min 192 max 16384 bytes
[   26.140488] sof-audio-pci-intel-tgl 0000:00:1f.3: period count 2 max 16
[   26.140489] sof-audio-pci-intel-tgl 0000:00:1f.3: buffer max 65536 bytes
[   26.140528] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: hw params stream 6 dir 1
[   26.140529] sof-audio-pci-intel-tgl 0000:00:1f.3: generating page table for 000000004fa59305 size 0x4af4 pages 5
[   26.140554] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   26.140567] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   26.140569] sof-audio-pci-intel-tgl 0000:00:1f.3: period_bytes:0x18fc
[   26.140570] sof-audio-pci-intel-tgl 0000:00:1f.3: periods:3
[   26.140578] sof-audio-pci-intel-tgl 0000:00:1f.3: stream_tag 3
[   26.140581] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000
[   26.140917] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   26.140918] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 3
[   26.140919] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   26.140923]  DMIC: ASoC: soc_pcm_hw_params() failed (-12)
[   26.140926]  DMIC: ASoC: dpcm_fe_dai_hw_params failed (-12
````

nklayman, 2021-08-12:
Glad that I could give you the information you need! I tried rebooting and checking the logs another five times and got errors in the log every time. Let me know if there's anything I can test or do to help.

plbossart, 2021-08-12:
IIRC the dma buffer lock is a rather recent issue, so maybe it's not there in previous version.

As a temporary work-around, maybe you can try going back to version 1.7. I'd be a hack, we've fixed a number of issues in 1.8 but we've obviously let a critter through. You can find the 1.7 version here: https://github.com/thesofproject/sof-bin/blob/main/v1.7.x/sof-v1.7/intel-signed/sof-tgl-h.ri

As usual for these hacks, *save your config* before experimenting.

plbossart, 2021-08-12:
I just checked and we have daily tests for SOF firmware and kernel on a TGL-H HDA platform.

@fredoh9 @aiChaoSONG is there a way we can try running the SOF firmware 1.8 with Linux stable 5.13.x on that TGLH_0A70_HDA platform?

aiChaoSONG, 2021-08-13:
@plbossart Yes, we have a kernel build job for stable kernel, I will post more details in teams channel.

plbossart, 2021-08-13:
Thanks @aiChaoSONG, the results with the following commits don't show any issues on our test TGL-H platform.

Kernel Branch: stable/linux-5.13.y commit a37da2be
SOF Branch: v1.8 commit 9e7a8f921efb

Not sure what we missed.

@nklayman can you upload the result of 'alsa-info'? I'd like to check for possible differences with our test device. Thanks!

nklayman, 2021-08-14:
@plbossart I uploaded the `alsa-info` output in [this comment](https://github.com/thesofproject/sof/issues/4629#issuecomment-897825601). If you think something might have changed since then, let me know and I'll upload a new one.

nklayman, 2021-08-14:
I'll also give v1.7 a shot tomorrow and will let you know if it works.

nklayman, 2021-08-15:
I downgraded to v1.6 (that's what `dnf downgrade` did, didn't bother to install 1.7 manually) and everything worked just fine. If you want me to test 1.7 I can as well.

plbossart, 2021-08-16:
@nklayman if you don't mind manually adding the 1.7 firmware and topology that would be great. 1.6 is relatively ancient and it'd be easier to track changes between 1.7 and 1.8.

@lgirdwood @singalsu if you have any ideas or suggestions to help identify what causes this IPC error on TGL-H we are all ears.

nklayman, 2021-08-16:
I downloaded the 1.7 `.ri` file to `intel-signed` and the speakers and mic work now.

lgirdwood, 2021-08-17:
This ones due to OOM for smallest allocator blocks used to alloc cache coherent buffer spinlocks for each buffer. Will be fixed in v1.9 as lock is part of new coherent API, but is topology specific for v1.8 (ie removing an unused pipeline from topology should fix it).

plbossart, 2021-08-17:
> This ones due to OOM for smallest allocator blocks used to alloc cache coherent buffer spinlocks for each buffer. Will be fixed in v1.9 as lock is part of new coherent API, but is topology specific for v1.8 (ie removing an unused pipeline from topology should fix it).

Not sure I get the explanation @lgirdwood, this is a bog-standard HDaudio topology that even our daily tests uses without issues. We also tried using 1.8 in our tests and didn't see any issues.

The only explanation I can find is PipeWire, this is reported on Fedora34 and we of course don't test this setup...

vchernin, 2021-08-18:
> The only explanation I can find is PipeWire, this is reported on Fedora34 and we of course don't test this setup...

Another Fedora user here, FYI on Fedora 34 you can swap PipeWire for PulseAudio with: `sudo dnf swap --allowerasing pipewire-pulseaudio pulseaudio` (and reboot).

It might be worth trying that to see if it's a PipeWire or Fedora 34 issue.

lgirdwood, 2021-08-18:
@plbossart this is about limitations in the heap allocator exposed by dynamic pipelines. We run out of small blocks as the ordering of allocation requests can impact free blocks of size X as the next size up is also checked if size X is empty (i.e. free/used size X and size X +1 can vary on allocation ordering).

plbossart, 2021-08-18:
@lgirdwood this is not using dynamic pipelines at all, the kernel code was not upstreamed yet.

plbossart, 2021-08-19:
@mrhpearson can you add this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/7016885/sof-dyndbg.conf.txt) as /etc/modprobe.d/sof-dyndbg.conf, reboot and provide the dmesg log without PipeWire enabled? Thanks!

mrhpearson, 2021-08-19:
No problem - attached (collected from journalctl as dmesg was truncated)
Please ignore the nouveau driver issues - known issue on this platform with the newer Nvidia card and I've not gotten around to installing the closed source driver. 
Let me know if you need anything else
Mark

[journalctl-sof-dyndbg.log](https://github.com/thesofproject/sof/files/7017098/journalctl-sof-dyndbg.log)

plbossart, 2021-08-19:
@mrhpearson The log says 'debian kernel', initially you mentioned Fedora?

mrhpearson, 2021-08-19:
@plbossart - I've been running both. I switched to Debian thinking they didn't have pipewire yet....but turns out they do. So I ended up disabling pipewire there.
I can run the collection on Fedora if you want :) The behaviour between the two OS's is the same - works with 1.8-rc2 but not with 1.8

plbossart, 2021-08-19:
@mrhpearson what causes the DMIC device to be opened then? Is this PulseAudio?

vchernin, 2021-08-19:
> I switched to Debian thinking they didn't have pipewire yet....but turns out they do. So I ended up disabling pipewire there.

I was under the impression PipeWire is available in Debian 11 and used for screen sharing, but not yet used by default for audio. See [the Debian Wiki](https://wiki.debian.org/PipeWire). Only Fedora uses it for audio by default currently AFAIK.

aiChaoSONG, 2021-08-20:
@plbossart  I think this issue may be TGL specific, because I am using a WHL lenovo laptop (CNL FW 1.8) with Manjaro distro, and both pipewire and pulseaudio are running.
```
➜  ~ ps -aux| grep -E "pipewire|pulseaudio"
chao        1391  0.0  0.2 1730016 16184 ?       Ssl  Aug19   0:01 /usr/bin/pulseaudio --daemonize=no --log-target=journal
chao       28199  0.0  0.0  36848  6096 ?        Ssl  10:04   0:00 /usr/bin/pipewire
chao       30587  0.0  0.0   6428  2316 pts/2    S+   10:26   0:00 grep --color=auto --exclude-dir=.bzr --exclude-dir=CVS --exclude-dir=.git --exclude-dir=.hg --exclude-dir=.svn --exclude-dir=.idea --exclude-dir=.tox -E pipewire|pulseaudio
```

Recipe on my laptop:
Kernel: stable 5.13.11 with Manjaro flavour kconfig
SOF: v1.8
Topology: sof-hda-generic-4ch.tplg

mrhpearson, 2021-08-20:
@plbossart - yes, pulseaudio

I've attached a log from Fedora, just in case it is useful as a comparison

[journalctl-sof-dyndbg2.log](https://github.com/thesofproject/sof/files/7022039/journalctl-sof-dyndbg2.log)

plbossart, 2021-08-20:
thanks @mrhpearson, do you mind trying to enable the SOF trace tools

a) fetch sof-logger from https://github.com/thesofproject/sof-bin/blob/main/v1.8.x/tools-v1.8/sof-logger, make it executable and add to /usr/local/bin
b) fetch sof-tgl-h.ldc from https://github.com/thesofproject/sof-bin/blob/main/v1.8.x/sof-v1.8/sof-tgl-h.ldc and add it to e.g. /usr/lib/firmware/intel/sof-v1.8/
c) add this in /etc/modprobe.d/alsa-base.conf (or any .conf)
options snd_sof sof_debug=1
d) sudo sof-logger -t -l /usr/lib/firmware/intel/sof-v1.8/sof-tgl-h.ldc | tee sof-log.txt

and attach the sof-log.txt

mrhpearson, 2021-08-23:
Thanks @plbossart 

With those steps I get: error: Invalid ldc file signature.
If I run it with the version in sof-bin (that is packaged with v1.8) I just get

 TIMESTAMP         (us)              DELTA  C# COMPONENT          LOCATION                      CONTENT	ktime=182.047s  @  2021-08-23 11:07:16 -0400 EDT

Is that expected? Any steps I'm missing (I needed a reboot between (c) and (d) above)

Mark

plbossart, 2021-08-23:
@mrhpearson this firmware packaging is a nightmare to support. Either you need to use the two .ri and .ldc files from the sof-bin repo, or the distro provides them. The requirement is that firmware and ldc files are aligned (the ldc contains a dictionary that's extracted when generated the .ri).

mrhpearson, 2021-08-23:
@plbossart - I tried replacing both .ldc and .ri from the sof repo and that didn't work (using the intel-signed version for the .ri)
Am I missing a step?

If I install the packages from sof-bin then it works - but no debug output. Wondering what the difference is between the sof repo .ldc and the sof-bin repo (for the same version of firmware)

Mark

plbossart, 2021-08-23:
No idea what's happening @mrhpearson. I provided the same instructions in issue https://github.com/thesofproject/linux/issues/2971 and that was successful. Something must be missing but I just don't know what...

mrhpearson, 2021-08-23:
@plbossart - well - I tried it again and this time got some results. Maybe a combo of a reboot or something?
Hopefully some useful clues in the attached.

[sof-log2.txt](https://github.com/thesofproject/sof/files/7034673/sof-log2.txt)

marc-hb, 2021-08-23:
>  I tried replacing both .ldc and .ri from the sof repo and that didn't work (using the intel-signed version for the .ri)
Am I missing a step?

Reloading the driver which reloads the .ri firmware? Happens :-)

> well - I tried it again and this time got some results. Maybe a combo of a reboot or something?

plbossart, 2021-08-23:
Thanks @mrhpearson this is super interesting. It's indeed the same error reported earlier by @nklayman. I didn't click before but I see two dai_prepare which fail. I am not fully clear on why this happens. @ranj063 any ideas?

````
<< first dai_prepare with channel = 3 works
[      131155.828122] (          61.718746) c0 dai          1.5            src/audio/dai.c:821  dai_prepare()
[      131174.994788] (          19.166666) c0 dai          1.5            src/audio/dai.c:769  dai_config_prepare(), channel = 3
[      131193.380204] (          18.385416) c0 dai          1.5            src/audio/dai.c:786  dai_config(): new configured dma channel index 3
[      145791.608790] (       14598.228516) c0 dai                           src/lib/dai.c:164  dai_get type 3 index 1 new sref 2
[      145810.879623] (          19.270832) c0 dai                           src/lib/dai.c:188  dai_put type 3 index 1 new sref 1
[      145835.410872] (          24.531248) c0 dai          2.15           src/audio/dai.c:706  dai_config() dai type = 3 index = 1 dd 0x9e121100
[      146233.744189] (         398.333313) c0 pipe         2.16  ....../pipeline-params.c:215  pipe params dir 1 frame_fmt 0 buffer_fmt 0 rate 48000
[      146251.348355] (          17.604166) c0 pipe         2.16  ....../pipeline-params.c:219  pipe params stream_tag 2 channels 2 sample_valid_bytes 2 sample_container_bytes 2
[      146376.087934] (         124.739578) c0 eq-iir       2.11  ../audio/eq_iir/eq_iir.c:614  eq_iir_params()
[      146403.744182] (          27.656248) c0 dai          2.15           src/audio/dai.c:357  dai_data_config() dai type = 3 index = 1 dd 0x9e121100
[      146531.400427] (         127.656242) c0 dai          2.15           src/audio/dai.c:509  dai_capture_params() src_dev = 0 stream_id = 0 src_width = 4 dest_width = 4
[      146548.952510] (          17.552082) c0 dai          2.15           src/audio/dai.c:515  dai_capture_params() fifo 0x0
[      146575.150426] (          26.197916) c0 pipe         2.16  ....../pipeline-params.c:297  pipe prepare
[      146637.025423] (          61.874996) c0 eq-iir       2.11  ../audio/eq_iir/eq_iir.c:765  eq_iir_prepare()
[      146656.660839] (          19.635416) c0 eq-iir       2.11  ../audio/eq_iir/eq_iir.c:799  eq_iir_prepare(), source_format=2, sink_format=0
[      146675.046255] (          18.385416) c0 eq-iir             ../audio/eq_iir/eq_iir.c:359  eq_iir_init_coef(), response assign for 2 channels, 1 responses
[      146693.171254] (          18.125000) c0 eq-iir             ../audio/eq_iir/eq_iir.c:429  eq_iir_init_coef(), ch 0 is set to response 0
[      146710.306670] (          17.135416) c0 eq-iir             ../audio/eq_iir/eq_iir.c:429  eq_iir_init_coef(), ch 1 is set to response 0
[      146738.171252] (          27.864582) c0 eq-iir       2.11  ../audio/eq_iir/eq_iir.c:815  eq_iir_prepare(), IIR is configured.

<< another dai_prepare with channel = 1 -> fails
[      146767.910835] (          29.739582) c0 dai          2.15           src/audio/dai.c:821  dai_prepare()
[      146785.515001] (          17.604166) c0 dai          2.15           src/audio/dai.c:769  dai_config_prepare(), channel = 1
[      146805.515000] (          20.000000) c0 dai          2.15           src/audio/dai.c:786  dai_config(): new configured dma channel index 1
[      153396.295988] (        6590.780762) c0 pipe         10.71 ....../pipeline-params.c:215  pipe params dir 1 frame_fmt 0 buffer_fmt 0 rate 48000
[      153414.212654] (          17.916666) c0 pipe         10.71 ....../pipeline-params.c:219  pipe params stream_tag 3 channels 2 sample_valid_bytes 2 sample_container_bytes 2
[      153542.181399] (         127.968742) c0 eq-iir       10.66 ../audio/eq_iir/eq_iir.c:614  eq_iir_params()
[      153569.577231] (          27.395832) c0 dai          10.70          src/audio/dai.c:357  dai_data_config() dai type = 2 index = 0 dd 0x9e122200
[      153587.181397] (          17.604166) c0 dai          10.70          src/audio/dai.c:376  config->dmic.fifo_bits = 32 config->dmic.num_pdm_active = 1
[      153630.045979] (          42.864582) c0 buffer                   src/audio/buffer.c:52   ERROR buffer_alloc(): could not alloc lock
[      153647.129311] (          17.083332) c0 dai          10.70          src/audio/dai.c:638  ERROR dai_params(): failed to alloc dma buffer
[      153668.639727] (          21.510416) c0 pipe         10.71 ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 64
[      153686.087643] (          17.447916) c0 ipc                  src/ipc/handler-ipc3.c:284  ERROR ipc: pipe 10 comp 64 params failed -12
````

ranj063, 2021-08-23:
@plbossart I tested a TGL-H HDA device with 1.8 and didnt see any issues. It looks like we're running out of memory when starting the 6th audio device during pulseaudio probing. But cant explain why it fails on this particular device and not mine

ranj063, 2021-08-24:
@mrhpearson I have a suggestion. Could you pelase try disabling runtime PM for the SOF driver and checking if the issue persists?

youc an disable it with "options snd-sof-pci sof_pci_debug=1" in one of the conf files in /etc/modprobe.d

mrhpearson, 2021-08-24:
Hi @ranj063  - Didn't seem to make any difference I'm afraid

kernel log and sof-logger files attached. I have that debug still enabled so let me know if it needs to be disabled.
[journalctl-sof-dyndbg3.log](https://github.com/thesofproject/sof/files/7035775/journalctl-sof-dyndbg3.log)
[sof-log3.txt](https://github.com/thesofproject/sof/files/7035776/sof-log3.txt)

Do we have any clues from the fact it worked in 1.8-rc2 but not in 1.8? Is it possible to try any test versions to narrow it down that way?

ranj063, 2021-08-24:
Thanks @mrhpearson. I wanted to check if there could be problems maybe we werent freeing the the memory properly during runtime suspend. Looks like it isnt the problem.

I did take a look at the diff between rc2 and 1.8 release commits and nothing stood out for me to explain the out of memory issue seen on your device.

ranj063, 2021-08-24:
@lgirdwood my suspicion is on these three commits that changes the allocation of the hda cha, and dai data to be on the runtime_shared heap. 

https://github.com/thesofproject/sof/commit/b3278b3c5145a18d08c89e10927b341f0d6f3a7b
https://github.com/thesofproject/sof/commit/2cede3159c2fc0d225e7a886c904895cccab9019
https://github.com/thesofproject/sof/commit/fc39c7689f950f3d601143cac28d583acfff65cd

Is there a way for us to verify if reverting these in 1,8 fixes this issue or if some of the latest changes Keyon made for heap allocation would make a difference? Unfortunately, we dont have a device that can reproduce this error and I'm guessing we'd need the Intel-signed binary to check. 

@keyonjie FYI

keyonjie, 2021-08-24:
Hi yes, I just back to the heap refinement PR https://github.com/thesofproject/sof/pull/4521 now, hopefully will finalize this within this week and then provide back-porting to stable-v1.8.

mrhpearson, 2021-08-24:
I'm very happy to be a pair of hands for testing if that's useful, but I would need the signed binary for this platform.

mrhpearson, 2021-09-03:
Hi @keyonjie - looks like you've been working hard on PR 4521 - do let me know if there's anything I can test here to confirm it fixes this issue as expected.

keyonjie, 2021-09-06:
@mrhpearson which tplg file does it used on? Can you build SOF firmware at your end? My PR #4521 is for main branch only, but I can do a simple hot fix commit for 1.8 if you can help to verify it.

mrhpearson, 2021-09-07:
Hi @keyonjie - I believe we're just using sof-hda-generic-4ch.tplg, but the firmware has to be Intel signed (at least from my experience) for it to work so I think you'd need to throw me a signed file to put on the system....let me know if there is something smarter I can do

keyonjie, 2021-09-07:
@mrhpearson I will submit a hotfix to the 1.8 branch, but we do need to go through the signing process for a new binary release. @mwasko

keyonjie, 2021-09-07:
@ktrzcinx @mwasko I just submitted a commit to address the issue, can you help to build and sign the tgl-h binaries with it, once @mrhpearson confirm it work, we can update the binaries to sof-bin @marc-hb @lgirdwood FYI.
https://github.com/thesofproject/sof/commit/07b9b87c9d4f5cd501620896fe1613d7e0305377

marc-hb, 2021-09-07:
@mrhpearson would you be able you contact the BIOS team for that device? They may have a way to allow additional keys on a sample of it.

marc-hb, 2021-09-07:
> I just submitted a commit to address the issue, 

If you really just submitted  it 11 minutes ago, I doubt we have validation results to make a release already.

keyonjie, 2021-09-07:
> > I just submitted a commit to address the issue,
> 
> If you really just submitted it 11 minutes ago, I doubt we have validation results to make a release already.

As we don't have reproduction here, I just performed a functional test on our CI tgl and tgl-h platforms so advanced full test may be needed, but first of all, we need the help from @mrhpearson to check if the change help to fix the issue reported here.

lgirdwood, 2021-09-07:
@keyonjie @mrhpearson fwiw, we should be signing v1.9-rc1 end of this week so fix could be verified on v1.9 ? Would this work ? (as the signing process can be resource intensive for @mwasko and team).

plbossart, 2021-09-07:
@keyonjie it bothers me that 'we don't have reproduction'... there's something really fishy here and trying to fix stuff without understanding what's broken doesn't seem quire right.

mrhpearson, 2021-09-07:
I'll test whatever is available - it's pretty quick and easy to do (if signed)
For root cause - I defer completely to you guys. From my point of view it broke between 1.8-rc2 and 1.8, but I don't have the tools to be able to bisect or anything like that. I'm a bit concerned with it being guesswork - but it does look like educated guesswork.

We're seeing this issue on multiple Lenovo platforms (and this bug was raised for a Dell platform so suggests they may be impacted too). I've delayed rolling out this version of firmware for the Debian package because of this issue - but I know Fedora has already made it available so it's only a matter of time before I get a heap of "audio is broken" bugs. I'm sure I'll hear from Arch folk soon :)

If it would be useful/important to get hands on some HW locally let me know and I can look at organising a loaner.

[15 more replies]

closed 2022-01-24 by XiaoyunWu6666