thesofproject/sof #4643: [BUG]spurious response 0x0:0x2 after system suspend-resume on EHL_RVP_I2S (with i915 suspending crtc failed with -22 calltrace)

XiaoyunWu6666, 2021-08-17:
**Describe the bug**
in innner daily 5595 and 5901
on EHL_RVP_I2S
when check system suspend resume with aplay running in the backgound
Call Trace occurred and " kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: spurious response 0x0:0x2, last cmd=0x20320031" error lead testcase to FAIL

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-ehl-rt5660.tplg ~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 5 -m playback

**Reproduction Rate**
rather low ,hard to reproduce manually 
It first happened in CI daily test  in late July and happened again on mid August.

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 9474f300
SOF Branch: main
SOF Commit: 53718867df85
Topology: /lib/firmware/intel/sof-tplg/sof-ehl-rt5660.tplg
Platform: EHL_RVP_I2S

**Screenshots or console output**
[console]
```
2021-07-29 23:17:33 UTC Sub-Test: [REMOTE_INFO] ===== Round(4/5) =====
2021-07-29 23:17:33 UTC Sub-Test: [REMOTE_COMMAND] Run the command: rtcwake -m mem -s 5
rtcwake: wakeup from "mem" using /dev/rtc0 at Thu Jul 29 23:17:39 2021
2021-07-29 23:17:39 UTC Sub-Test: [REMOTE_COMMAND] sleep for 5
2021-07-29 23:17:44 UTC Sub-Test: [REMOTE_INFO] Check for the kernel log status
declare -- cmd="journalctl_cmd --since=@1627600648"
2021-07-29 23:17:44 UTC [ERROR] Caught kernel log error
===========================>>
[ 6868.088771] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: spurious response 0x0:0x2, last cmd=0x20320031
<<===========================
2021-07-29 23:17:44 UTC Sub-Test: [REMOTE_ERROR] Caught error in kernel log
```
[dmesg]
```
[ 6826.875428] kernel: Call Trace:
[ 6826.875442] kernel:  i915_drm_suspend_late+0xe6/0x100 [i915]
[ 6826.875604] kernel:  ? pci_pm_poweroff_late+0x30/0x30
[ 6826.875614] kernel:  dpm_run_callback+0x6d/0x190
[ 6826.875629] kernel:  __device_suspend_late+0xb6/0x200
[ 6826.875639] kernel:  async_suspend_late+0x16/0x90
[ 6826.875648] kernel:  async_run_entry_fn+0x26/0x110
[ 6826.875659] kernel:  process_one_work+0x286/0x4f0
[ 6826.875676] kernel:  worker_thread+0x48/0x3d0
[ 6826.875686] kernel:  ? process_one_work+0x4f0/0x4f0
[ 6826.875696] kernel:  kthread+0x141/0x170
[ 6826.875707] kernel:  ? set_kthread_struct+0x40/0x40
[ 6826.875719] kernel:  ret_from_fork+0x1f/0x30
[ 6826.875745] kernel: ---[ end trace d90f18c5ef153897 ]---
```
```
[ 6867.991886] kernel: rfkill: input handler enabled
[ 6867.991973] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 6867.991977] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 791060
[ 6867.992045] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[ 6867.992057] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 6 dir 0 cmd 1
[ 6867.992970] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x14001e successful
[ 6867.992982] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 6867.993842] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 6868.025593] kernel: asix 1-4.1:1.0 enx000ec66547b7: Link is Down
[ 6868.078909] kernel: rfkill: input handler disabled
[ 6868.088771] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: spurious response 0x0:0x2, last cmd=0x20320031
[ 6868.456010] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 6868.456059] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 6868.955920] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 6868.955960] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 6870.088289] kernel: asix 1-4.1:1.0 enx000ec66547b7: Link is Up - 100Mbps/Full - flow control off
```
full dmesg log :
[dmesg.txt](https://github.com/thesofproject/sof/files/6996707/dmesg.txt)

XiaoyunWu6666, 2021-08-17:
hi @kv2019i could you please help to check it ? it is i915 relevant :) thx

lgirdwood, 2021-08-17:
looping in @plbossart as this could be driver related, lets wait for more info.

kv2019i, 2021-08-17:
This looks like non-audio related error in i915 driver. The wakerefs seem to be increasing with every suspend cycle:

<span class="txtRed">[ 6826.815248] kernel: i915 0000:00:02.0: [drm] *ERROR* Suspending crtc's failed with -22</span>
[ 6826.874461] kernel: ------------[ cut here ]------------
[ 6826.874473] kernel: i915 0000:00:02.0: i915 raw-wakerefs=147 wakelocks=147 on cleanup

<span class="txtRed">[ 6838.538985] kernel: i915 0000:00:02.0: [drm] *ERROR* Suspending crtc's failed with -22</span>
[ 6838.599692] kernel: ------------[ cut here ]------------
[ 6838.599704] kernel: i915 0000:00:02.0: i915 raw-wakerefs=149 wakelocks=149 on cleanup

<span class="txtRed">[ 6850.625206] kernel: i915 0000:00:02.0: [drm] *ERROR* Suspending crtc's failed with -22</span>
[ 6850.686577] kernel: ------------[ cut here ]------------
[ 6850.686589] kernel: i915 0000:00:02.0: i915 raw-wakerefs=151 wakelocks=151 on cleanup

<span class="txtRed">[ 6862.111514] kernel: i915 0000:00:02.0: [drm] *ERROR* Suspending crtc's failed with -22</span>
[ 6862.170342] kernel: ------------[ cut here ]------------
[ 6862.170353] kernel: i915 0000:00:02.0: i915 raw-wakerefs=153 wakelocks=153 on

<span class="txtRed">[ 6826.815248] kernel: i915 0000:00:02.0: [drm] *ERROR* Suspending crtc's failed with -22</span>
[ 6826.874461] kernel: ------------[ cut here ]------------
[ 6826.874473] kernel: i915 0000:00:02.0: i915 raw-wakerefs=147 wakelocks=147 on cleanup

... and so on. Audio driver traces show we release the i915 display reference.

I checked https://gitlab.freedesktop.org/drm/intel/ but could not find any existing issue, so a new one should be filed there.
And this is clearly Linux kernel side issue, so let's move issue to kernel.

kv2019i, 2021-08-17:
Aa, the spurious response trace is another thing. It doesn't seem to be related to the i915 errors and audio FW execution does not fail. I have not seen this before though. So this could need a separate bug to be filed if we want to track the i915 issue separately. But not sure if this is needed if no actually functionality is failing.

[ 6868.088771] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: spurious response 0x0:0x2, last cmd=0x20320031

XiaoyunWu6666, 2022-01-24:
haven't seen it in CI for months

closed 2022-01-24 by XiaoyunWu6666