thesofproject/sof #4645: [BUG] [APL] Zephyr SOF link failure on APL with XCC toolchain, .bss not within `ucram`

aiChaoSONG, 2021-08-17:
Zephyr SOF link failure on APL with XCC toolchain, .bss not within `ucram`, error log:
```
[5/5] Linking C executable logger/sof-logger
[238/248] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
/home/chao/sof/zephyrproject/zephyr/kernel/sched.c:641: warning: C99 inline functions are not supported; using GNU89
/home/chao/sof/zephyrproject/zephyr/kernel/sched.c:641: warning: to disable this warning use -fgnu89-inline or the gnu_inline function attribute
/home/chao/sof/zephyrproject/zephyr/kernel/sched.c:931: warning: C99 inline functions are not supported; using GNU89
/home/chao/sof/zephyrproject/zephyr/include/kernel/sched_priq.h:36: warning: ‘z_priq_dumb_add’ declared inline after being called
/home/chao/sof/zephyrproject/zephyr/include/kernel/sched_priq.h:36: warning: previous declaration of ‘z_priq_dumb_add’ was here
/home/chao/sof/zephyrproject/zephyr/kernel/sched.c:1037: warning: C99 inline functions are not supported; using GNU89
/home/chao/sof/zephyrproject/zephyr/kernel/sched.c:1045: warning: C99 inline functions are not supported; using GNU89
[240/248] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map 
: && /home/chao/xtensa/XtDevTools/install/tools/RG-2017.8-linux/XtensaTools/bin/xt-xcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/home/chao/sof/zephyrproject/build-apl/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/arch/arch/xtensa/core/startup/libarch__xtensa__core__startup.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/xtensa/intel_adsp/common/libintel_adsp_common.a  zephyr/drivers/ipm/libdrivers__ipm.a  modules/sof/libmodules_sof.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj  -L/home/chao/sof/zephyrproject/build-apl/zephyr  -lgcc  -lhal  zephyr/arch/common/libisr_tables.a  zephyr/soc/xtensa/intel_adsp/common/libintel_adsp_common.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N && cd /home/chao/sof/zephyrproject/build-apl/zephyr && /usr/bin/cmake -E echo
/home/chao/xtensa/XtDevTools/install/tools/RG-2017.8-linux/XtensaTools/bin/xt-ld: address 0x9e0896b0 of zephyr/zephyr_prebuilt.elf section `.bss' is not within region `ucram'
/home/chao/xtensa/XtDevTools/install/tools/RG-2017.8-linux/XtensaTools/bin/xt-ld: address 0x9e0896b0 of zephyr/zephyr_prebuilt.elf section `.bss' is not within region `ucram'
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/chao/sof/zephyrproject/build-apl
```

kv2019i, 2021-08-18:
We are simply running out of memory. I uploaded a temporary workaround for CI as https://github.com/thesofproject/sof/pull/4651 , but let's keep this bug open.

aiChaoSONG, 2021-08-19:
@kv2019i  With the workaround #4651 form Kai, CI now enables Zephyr SOF build on CNL/ICL/TGL-H with XCC, and APL stays with GCC