thesofproject/sof #4656: [BUG]APL_UP2_NOCODEC_ZEPHYR ipc timed out when PCM_FREE , when capturing with -Dhw:0,0

XiaoyunWu6666, 2021-08-19:
**Describe the bug**
APL_UP2_NOCODEC ipc timed out when PCM_FREE , when capturing with -Dhw:0,0 after capturing with -Dhw:0,10 and -Dhw:0,11 (DMIC and DMIC16)
since inner daily 5745

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg ~/sof-test/test-case/check-capture.sh -d 3 -l 1 -r 1 -F

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: b7329735
SOF Branch: main
SOF Commit: 5a42c023d2d1
Topology:/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg 
Platform: APL_UP2_NOCODEC_ZEPHYR

**Screenshots or console output**
【dmesg】
```
[  213.706809] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  213.707784] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  216.780966] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 0
[  216.780998] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  216.782252] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  216.782286] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: FW Poll Status: reg[0xa0]=0x240000 successful
[  216.782305] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  217.285068] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x60030000 size 12
[  217.285225] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: preventing DSP entering D3 state to preserve context
[  217.285231] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump start ]------------
[  217.285243] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[  217.285251] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  217.285260] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  217.285268] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump end ]------------
[  217.285274] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump start ]------------
[  217.285282] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: status: fw entered - code 00000005
[  217.285330] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  217.285338] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump end ]------------
[  217.285348] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at soc_component_trigger on 0000:00:0e.0: -110
[  217.285359] kernel:  Port0: ASoC: trigger FE cmd: 0 failed: -110
[  217.285435] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: free stream 0 dir 1
[  217.285446] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  217.789111] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x60030000 size 12
[  217.789157] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[  218.293160] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x30020000 size 12
[  218.293194] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to free widget PCM0C
[  218.293210] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PCM0C freed
[  218.293225] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[  218.797189] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x30210000 size 12
[  218.797225] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to free widget BUF2.0
[  218.797242] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget BUF2.0 freed
[  218.797258] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[  219.301261] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x30020000 size 12
[  219.301287] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to free widget PGA2.0
[  219.301300] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PGA2.0 freed
[  219.301312] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[  219.805256] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x30210000 size 12
[  219.805292] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to free widget BUF2.1
[  219.805309] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget BUF2.1 freed
[  219.805325] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[  220.309310] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x30110000 size 12
[  220.309347] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to free widget PIPELINE.2.SSP0.IN
[  220.309364] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget PIPELINE.2.SSP0.IN freed
[  220.309392] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:0e.0: -110
[  220.309698] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  220.813347] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x80010000 size 216
[  220.813382] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: failed resetting DAI config for SSP0.IN
[  220.813404] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[  221.317376] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x30020000 size 12
[  221.317412] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to free widget SSP0.IN
[  221.317429] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: widget SSP0.IN freed
[  221.317459] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: close stream 0 dir 1
[  221.337005] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: open stream 0 dir 1
[  221.337017] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: period min 192 max 16384 bytes
[  221.337022] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: period count 2 max 16
[  221.337025] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: buffer max 65536 bytes
[  221.337331] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: DAI config 0 matches pcm hw params
[  221.337337] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: DAI config 0 matches pcm hw params
[  221.337340] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[  221.337343] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: channels_min: 2 channels_max: 2
[  221.337351] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: DAI config 0 matches pcm hw params
[  221.337354] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: DAI config 0 matches pcm hw params
[  221.337356] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[  221.337359] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: channels_min: 2 channels_max: 2
[  221.337372] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  221.845549] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x30010000 size 96
[  221.845585] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to load widget SSP0.IN
[  221.845598] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: failed setting up DAI widget SSP0.IN
[  221.845610] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_dai_hw_params on SSP0 Pin: -110
[  221.845687] kernel:  NoCodec-0: ASoC: soc_pcm_hw_params() failed (-110)
[  221.845703] kernel:  Port0: ASoC: dpcm_fe_dai_hw_params failed (-110)
[  221.845730] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: free stream 0 dir 1
[  221.846773] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: close stream 0 dir 1

```

XiaoyunWu6666, 2021-08-19:
caused by commit  5a42c023d

commit 5a42c023d2d1705b594566643ce6cbf95e2660e7
Author: Ranjani Sridharan <ranjani.sridharan@linux.intel.com>
Date:   Wed Aug 4 11:45:29 2021 -0700
topology1: make all cavs nocodec topologies dynamic
All cAVS platforms can use dynamic pipelines with nocodec topology.
  Signed-off-by: Ranjani Sridharan <ranjani.sridharan@linux.intel.com>

kv2019i, 2021-08-20:
Thanks @XiaoyunWu6666 for finding out the commit, issue under debug.

kv2019i, 2021-08-24:
Potential bugfix: https://github.com/thesofproject/sof/pull/4673

XiaoyunWu6666, 2021-08-30:
havent' happened in daily test since Aug 25

closed 2021-08-30 by XiaoyunWu6666