thesofproject/sof #4658: v1.8 not working on Lenovo P1G4 (Tigerlake-h) based system (v1,8-rc2 works)

mrhpearson, 2021-08-19:
Hi,

I'm running Fedora 34 latest (5.13.10 kernel) on a Lenovo P1G4 and the audio is broken.

Doing some digging I found that v1.8-rc2 works correctly, but v1.8 doesn't. Error messsages are:

[    9.852559] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    9.852578] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 2
[    9.852579] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[    9.852582]  DMIC: ASoC: soc_pcm_hw_params() failed (-12)
[    9.852585]  DMIC: ASoC: dpcm_fe_dai_hw_params failed (-12)
(repeats 20 times)

Any suggestions on what might be happening and how to debug?
Note - I've tested the same firmware on X1C7 (Cometlake) and X1C9 (Tigerlake) and it works correctly

Thanks
Mark

lgirdwood, 2021-08-19:
Hi @mrhpearson are you able to send the full dmesg and the command line for the failed command. The error looks like the requested sample rate, channels or format was not available on for the given PCM device.
@plbossart fyi.

plbossart, 2021-08-19:
Hi @mrhpearson I think this issue is a duplicate of https://github.com/thesofproject/sof/issues/4658.

I think this is related to the use of PipeWire which we don't test in our current CI.

it's very interesting that 1.8-rc2 works, earlier we identified that 1.7 worked fine but I didn't think of asking for 1.8-rc2. That may make it easier to locate the problematic patch.

Do you mind if we continue this thread on https://github.com/thesofproject/sof/issues/4658 ?

mrhpearson, 2021-08-19:
Hi @lgirdwood - dmesg log attached (sorry - should have done that in the original)
[sof-dmesg.log](https://github.com/thesofproject/sof/files/7014924/sof-dmesg.log)

@plbossart - not sure how/why two issues were created. Happy to carry this on here.

I'll test the firmware without pipewire and let you know.

mrhpearson, 2021-08-19:
I see the same without pipewire - don't think that's the issue here 
(And just to check - I did 'systemctl disable --global pipewire', rebooted and confirmed the service wasn't running)
Mark

closed 2021-08-19 by plbossart