thesofproject/sof #4662: [BUG] sof-audio-acpi-intel-byt stops and give a screaming high sound

hedmo, 2021-08-20:
**Describe the bug**
after updating my system .. when i play video or listen to music , the sound gets cut of and leave me with a high screaming sound from the speakers

**To Reproduce**
install the latest kernel (5.13.12)
install the latest alsa deps...
install sof-firmware-1.8

**Reproduction Rate**
all the time

**Expected behavior**
enjoy the music/sound

**Impact**
cannot listen to music or watch videos

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel:5.13.12
    * SOF: 1.8
2) Name of the topology file
    * Topology: sof-audio-acpi-intel-byt
3) Name of the platform(s) on which the bug is observed.
    * Platform: gentoo linux

**Screenshots or console output**
no Screenshots/console output

**dmesg**

here is a snip of dmesg | grep sof-audio : 

[   10.575429] sof-audio-acpi-intel-byt 808622A8:00: Firmware info: version 1:8:0-9e7a8
[   10.575440] sof-audio-acpi-intel-byt 808622A8:00: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[   10.575459] sof-audio-acpi-intel-byt 808622A8:00: unknown sof_ext_man header type 3 size 0x30
[   10.616764] sof-audio-acpi-intel-byt 808622A8:00: Firmware info: version 1:8:0-9e7a8
[   10.616776] sof-audio-acpi-intel-byt 808622A8:00: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[   11.172497] sof-audio-acpi-intel-byt 808622A8:00: Topology: ABI 3:18:1 Kernel ABI 3:18:0
[   11.181617] sof-audio-acpi-intel-byt 808622A8:00: ASoC: Parent card not yet available, widget card binding deferred
[ 1137.442116] sof-audio-acpi-intel-byt 808622A8:00: error: reply expected 12 got 0 bytes

plbossart, 2021-08-20:
Thanks @hedmo for reporting this.

SOF should not (yet) be the default on BYT, we still have a number of bugs. I am not sure why your distro chose to enable SOF, but you can try a work-around by adding this in /etc/modprobe.d/alsa-base.conf

options snd-intel-dspcfg dsp_driver=2

that should select the 'SST' driver and fallback to what should still be the default.

If you don't mind though, we've never seen this IPC error so it'd be useful for us if you can copy this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/7022122/sof-dyndbg.conf.txt) as /etc/modprobe.d/sof-dyndbg.conf,  reboot and attach the dmesg log. I would be curious to see what causes this error.

You can go back to SOF by using 
options snd-intel-dspcfg dsp_driver=3

plbossart, 2021-08-20:
Maybe a red-herring but the only case I am aware of where we have this error reported in https://github.com/thesofproject/linux/issues/2971

That's for a PCI device and with a different codec (HDAudio v. I2S), so this might be a coincidence
````
Jun 16 12:29:59 Freja kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: reply expected 12 got 0 bytes
Jun 16 12:29:59 Freja kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: ipc error for 0x80010000 size 12
Jun 16 12:29:59 Freja kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
Jun 16 12:29:59 Freja kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on Analog CPU DAI: -22
````

hedmo, 2021-08-20:
@plbossart 

thank you for the fast reply. i will go slow forward and test : 

options snd-intel-dspcfg dsp_driver=2

some time before reporting back. i will provide what you with a dmesg.txt at the end of the day.

regards

hedmo, 2021-08-20:
the test went fast and the cut off with a sound came. here you go : 
[dmesg.txt](https://github.com/thesofproject/sof/files/7022442/dmesg.txt)

hedmo, 2021-08-20:
hm . i checked the other bug and have lockups to . i thought it was because of : 

https://bugzilla.kernel.org/show_bug.cgi?id=109051

but i did disable intel_idle and went with acpid . i still have lockups with acpid .

regards

plbossart, 2021-08-20:
@hedmo I don't see any errors in the log you provided in https://github.com/thesofproject/sof/issues/4662#issuecomment-902760390

hedmo, 2021-08-20:
no . i just did a reboot and provided the log . do you want me to check dmesg and send 
one when it happends ?

plbossart, 2021-08-20:
> no . i just did a reboot and provided the log . do you want me to check dmesg and send
> one when it happends ?

yes please.

hedmo, 2021-08-20:
it came again but it wont show in dmesg. when it happends i just need to close the 
window ,wait a min and it will stop.

hedmo, 2021-08-21:
hi again 
running  lts  kernel-5.10.52 gives me no sound . here is that dmesg : 
[dmesg.txt](https://github.com/thesofproject/sof/files/7025417/dmesg.txt)

regards

hedmo, 2021-08-23:
@plbossart

adding the 'dsp_driver=2' option brings the sound back on 5.10.x . ATM it does not sync fully but i have been trying to see if my problem where in userspace and switched form pulseaudio to pipewire this weekend . i will revert back and see if the out of sync is in userspace with this problem. i still hope we can find the source with 5.13.x kernels. do you know why on older kernel use legacy and  newer sof?. if you can think of anything i can provide you with for 5.13.x just let me know :)

regards

plbossart, 2021-08-23:
There was a Kconfig added SND_INTEL_BYT_PREFER_SOF but that should not be selected, that's what may cause the unintended switch to SOF.

plbossart, 2021-08-23:
The SND_INTEL_BYT_PREFER_SOF value will be overridden by the module option, so you can always go back to SOF and help us improve it by reporting issues/logs. As seen above in https://github.com/thesofproject/sof/issues/4662#issuecomment-903857134 there are some issues with the legacy driver as well, but they will likely never be fixed.

hedmo, 2021-08-24:
@plbossart 

today i changed the kernel option :  SND_INTEL_BYT_PREFER_SOF .
and yes it forced it to use the legacy driver . the odd sound started direct but with
an other sound more like when you are in a field and hear the crickets. hi have to add that the file : /etc/modprobe.d/alsa-base.conf was named wrong and did not have effects before but now it is working .

hedmo, 2021-08-28:
kernel-5.14-rc7 : 
[   11.141220] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[   11.141296] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[   11.141518] rt5645 i2c-10EC3270:00: Detected ASUS T100HAN platform
[   11.141530] rt5645 i2c-10EC3270:00: supply avdd not found, using dummy regulator
[   11.141577] rt5645 i2c-10EC3270:00: supply cpvdd not found, using dummy regulator
[   11.160500] dw_dmac INTL9C60:01: DesignWare DMA Controller, 8 channels
[   11.229194] sof-audio-acpi-intel-byt 808622A8:00: Firmware info: version 1:8:0-9e7a8
[   11.229207] sof-audio-acpi-intel-byt 808622A8:00: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[   11.229227] sof-audio-acpi-intel-byt 808622A8:00: unknown sof_ext_man header type 3 size 0x30
[   11.266628] sof-audio-acpi-intel-byt 808622A8:00: Firmware info: version 1:8:0-9e7a8
[   11.266640] sof-audio-acpi-intel-byt 808622A8:00: Firmware: ABI 3:18:1 Kernel ABI 3:18:0

still no go :(

hedmo, 2021-08-29:
after many hours of recompiling a kernel i have now manage to watch youtube for more then an hour on sof-firmware :) .
i have cleaned out all soc legacy audio drivers i the kernel  because with the legacy driver i had a strange feeling they conflicted some how . the legacy driver still have the bug and comes random. on the sof driver when i clean codecs i did find that if i only added only the right codec i did not get sound : 
rt5645 i2c-10EC3270:00: Detected ASUS T100HAN platform = SND_SOC_INTEL_CHT_BSW_RT5645_MACH ?

so i added them all and now it is working. i wont say it is solved ATM but it is a good way to be.
here is my .config : 

[config.txt](https://github.com/thesofproject/sof/files/7073098/config.txt)

regards

lgirdwood, 2021-09-20:
Please see https://github.com/thesofproject/sof/pull/4775 as a potential fix. GCC built binaries here https://github.com/thesofproject/sof/issues/3868 but may be slow since GCC does not support xtensa SIMD.

hedmo, 2021-09-21:
Thanks . I have not had any beeping sound more after THE kernel clean out. Still rocking :) .

fhost, 2021-11-05:
Hello, I've update my laptop, and I have the same issue as the one describe here originally, and I have the same card (rt5645). After some time (few seconds to several minutes), it stops playing and I have the screaming high sound.

The issue doesn't seems to generate any logs (I've looked in pulseaudio and dmesg).

I'm on Archlinux, sof 1.8 and kernel 5.12.15.

I did not try to recompile the kernel with option CONFIG_SND_SOC_SOF_STRICT_ABI_CHECKS as I don't have the error logs reported here. The only sof log I have are:
```
sof-audio-acpi-intel-byt 80860F28:00: Detected Baytrail-CR platform
sof-audio-acpi-intel-byt 80860F28:00: Detected Baytrail-CR platform
sof-audio-acpi-intel-byt 80860F28:00: Firmware info: version 1:8:0-9e7a8
sof-audio-acpi-intel-byt 80860F28:00: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
sof-audio-acpi-intel-byt 80860F28:00: unknown sof_ext_man header type 3 size 0x30
sof-audio-acpi-intel-byt 80860F28:00: Firmware info: version 1:8:0-9e7a8
sof-audio-acpi-intel-byt 80860F28:00: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
sof-audio-acpi-intel-byt 80860F28:00: Topology: ABI 3:18:1 Kernel ABI 3:18:0
sof-audio-acpi-intel-byt 80860F28:00: ASoC: Parent card not yet available, widget card binding deferred
Nov 05 15:06:39 laevatein kernel: input: sof-bytcht rt5645 Headset as /devices/platform/80860F28:00/cht-bsw-rt5645/sound/card0/input13
```

Also, I have "snd_soc_rt5645.quick=0x31", but I'm not sure it has any impact.

(PS : I'm using snd_intel_dspcfg.dsp_driver=2 to bring back the sound, but if I understand well, it makes it uses the "old" way, so it doesn't mean it "solve" the issue, right ?)

hedmo, 2021-11-05:
@fhost  try to remove all legacy drivers in your kernel . That worked for me.  When time comes i Will clean out My kernel more and test with legacy added again but for now i Will Keep legacy removed.

Regards.

fhost, 2021-11-05:
> @fhost try to remove all legacy drivers in your kernel . That worked for me. When time comes i Will clean out My kernel more and test with legacy added again but for now i Will Keep legacy removed.

Thanks ! Which drivers are the legacy ones ? I've just discover SOF today... So I still don't get way exactly it is.

plbossart, 2021-11-05:
@fhost there's no need to remove legacy drivers as suggested above.

We haven't been able to root-cause the 'screaming sound' issue (already reported on https://github.com/thesofproject/sof/issues/3868) , so at this point you have two options
a) stay with legacy snd_intel_dspcfg.dsp_driver=2
b) enable the trace with SOF snd_sof.sof_debug=1

snd_soc_rt5645.quick=0x31 is a separate configuration for the board-level settings (higher-level than internal DSP handling).

fhost, 2021-11-12:
Thanks! Both solution seems to work fine. I'll stick with solution b) for the moment because sof brings some improvements ;)

If I can help with something to find the root-cause, feel free to ask.

plbossart, 2021-11-12:
Thanks for the offer @fhost. 

It's a tricky race condition to fix really. With the trace disabled well we have no logs to figure out what went wrong. And with the trace enbled we have an Heisenbug effect where the problem doesn't happen.... And all our CI tests enable the trace, so it's an area where we don't have a good coverage.

hedmo, 2021-12-11:
hi again . 

i have started on an other device (t100ta) with an generic kernel and the it came back again . this time with : bytcr_rt5640 .

still no output in dmesg :( 

[    8.065738] iTCO_wdt iTCO_wdt.2.auto: initialized. heartbeat=30 sec (nowayout=0)
[    8.073296] bytcr_rt5640 bytcr_rt5640: quirk IN1_MAP enabled
[    8.073306] bytcr_rt5640 bytcr_rt5640: quirk realtek,jack-detect-source 3
[    8.073310] bytcr_rt5640 bytcr_rt5640: quirk realtek,over-current-threshold-microamp 2000
[    8.073313] bytcr_rt5640 bytcr_rt5640: quirk realtek,over-current-scale-factor 1
[    8.073317] bytcr_rt5640 bytcr_rt5640: quirk MCLK_EN enabled
[    8.073424] Loading firmware: intel/sof-tplg/sof-byt-rt5640.tplg
[    8.074941] sof-audio-acpi-intel-byt 80860F28:00: Topology: ABI 3:17:0 Kernel ABI 3:18:0
[    8.083427] sof-audio-acpi-intel-byt 80860F28:00: ASoC: Parent card not yet available, widget card binding deferred
[    8.091260] input: sof-bytcht rt5640 Headset as /devices/platform/80860F28:00/bytcr_rt5640/sound/card0/input21

regards

hedmo, 2021-12-26:
Backing down : 1.9.3 > 1.9 solve it . 

Regards

plbossart, 2022-01-10:
@hedmo it seems rather suspicious that downgrading the firmware makes BYT work, we've had this issue since 1.6.1. Likely a bit of luck here.

hedmo, 2022-01-10:
@plbossart . yes . that is why i added this because it was never a conflict  with legacy driver  and hope you may find a pattern or change between the versions. i have been using dist (prebuild) kernels to and they are working with 1.9 . 1.9.3 did not work on both 
rt5640 and rt5645 but works with 1.9 . i am using 1.9 for now with no issues but with tr5645.
it  stops working after hibernate. if you want me to make a new bug and send the dmesg log? . i can do that .

plbossart, 2022-01-10:
@hedmo It would be very useful to confirm
a) if you see any difference between SOF 1.9 and 1.9.3, and SOF 2.0 (which is now out it seems)

b) I am curious about the hibernate part, which is a point that no one else mentioned. maybe we are missing something when coming out of S3 that causes the 'screaming high sound' at some point. If you can try testing with a continuous test without ever hibernating this would be interesting.

hedmo, 2022-01-10:
@plbossart  i Will send dmesg logs from all versions and the log after hibernate . I Always listen to music or watching YouTube . The only thing that stops the sound is is to pause or Close the program ( kill the stream) . I can add that i am using intel_idle.max_cstate=2

hedmo, 2022-01-15:
@plbossart  i got some free time now . wil start with hibernate : 
 
dmesg before : 
[dmesg.txt](https://github.com/thesofproject/sof/files/7875526/dmesg.txt)

dmesg after : 
[dmesg-after.txt](https://github.com/thesofproject/sof/files/7875532/dmesg-after.txt)

plbossart, 2022-01-17:
@hedmo I don't see anything useful in the logs, did you notice any difference?

lgirdwood, 2022-01-18:
Dont see anything either in the logs, but I dont know if we should expect to see normal level SOF driver PM messages for hibernate (maybe these need debug log level).
@hedmo does restarting streams help after hibernate ? What about restarting pulseaudio or pipewire help after hibernate or unload and reload sof modules ?

hedmo, 2022-01-18:
@lgirdwood . I can provide a sig debug dmesg to . About hibernate .
Nothing works (restarting stream not restart pulse). I dont know what module to unload. tr5640 works after hibernate tho.

hedmo, 2022-01-23:
good morning .

i dont know if this is for any help but on legacy drivers i get this : 

`[   39.002061] intel_sst_acpi 808622A8:00: Wait timed-out condition:0x0, msg_id:0x1 fw_state 0x3
[   39.002075] intel_sst_acpi 808622A8:00: fw returned err -16
[   39.002088] sst-mfld-platform sst-mfld-platform: ASoC: error at snd_soc_dai_startup on ssp2-port: -16
[   39.002101]  SSP2-Codec: soc_pcm_open() failed (-16)
[   39.002106]  Audio Port: ASoC: dpcm_be_dai_startup() failed at SSP2-Codec (-16)
[   39.002111]  Audio Port: dpcm_fe_dai_startup() failed (-16)`

hedmo, 2022-01-23:
i have checked with sof debug but nothing useful there. legacy driver on rt5645 still does not work .

hedmo, 2022-01-23:
more info when hibernated.. if i turn sof-bytcht rt5645 off it will disappear frm the list and i cannot turn it on again but it is still in alsamixer...

lgirdwood, 2022-01-24:
@hedmo hmmm, if the legacy driver/fw also does not work then I suspect its similar issue to the same issue with the older Minnowboards where S4/S4 also did not recover the DSP (DSP PCI device never comes out of power off states). 

S3 works on Chrome/Windows though, so I suspect that Coreboot and the Windows driver are doing something extra at S3/4 -> S0 to the DSP PCI device flow (the same Linux drivers are used on Chrome). It's probably worth checking the coreboot code  for S3 -> S0 transition for any PCI or DSP programming.

hedmo, 2022-01-24:
@lgirdwood  . The legacy driver does not work at all.

plbossart, 2022-01-24:
> @lgirdwood . The legacy driver does not work at all.

I remember working on ASUS TA100 and HAN100 and having no problems with hibernate, this is really surprising that you would see this. I will have to find time to retry.

hedmo, 2022-02-03:
guys . i have made a mistake . i had not added : options snd_sof sof_debug=1
now it looks like 2.0 works without screaming sound . hibernate is still broken.

regards

hedmo, 2022-02-11:
checked some errors after hibernate : 
dmesg = pci 0000:00:03.0: can't change power state from unknown to D0 (config space inaccessible)

lspci = 00:03.0 Multimedia controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Imaging Unit (rev ff)

lgirdwood, 2022-02-14:
@hedmo ok, the same thing happened with the early Revs of Minnowboard. The audio DSP PCI device could not enter D0 after S3. However, it could enter D0 on other BYT/CHT devices with Linux + coreboot or with Windows which makes me think we are missing a register programming sequence that's done by coreboot or by the WIndows driver.  

The coreboot code for baytrail/cherrytrail based chromebooks is all opensource so it may be possible to drill down into the core boot D3 -> D0 code and see it's it's doing anything here for the DSP PCI device (which could be copied into the Linux driver if needed).

hedmo, 2022-02-14:
@lgirdwood yes you are right . Other BYT/CHT devices (t100ta RT 5640) works but this on does not .

danao81, 2022-02-22:
Hi all
After a long search I finally found this post
I'm going through the same issue, my audio playback randomly stops and starts a very annoying beep.
This happens when reproducing from any audio source (.mp3, video files, youtube streaming).

I tried to "add this to /etc/modprobe.d/alsa-base.conf

options snd_sof sof_debug=1"

but in my alsa-base.conf there's no "snd_sof_debug" string

Attached some system info:

              
[info.txt](https://github.com/thesofproject/sof/files/8116920/info.txt)

Thanks in advance for any help

plbossart, 2022-02-22:
@danao81 can you attach the results of 'alsa-info', that should show which modules are available in your system. lsmod | grep sof works are well...

danao81, 2022-02-22:
> @danao81 can you attach the results of 'alsa-info', that should show which modules are available in your system. lsmod | grep sof works are well...

Hi, thank you for the quick reply

I already tried to check alsa-info before posting , but it seems I don't have the file in the folder neither the command line works, I know it sounds weird...
I do have alsa-utils installed... 

Maybe a kernel update could help?

plbossart, 2022-02-22:
alsa-info is usually installed as /usr/sbin/alsa-info

Use lsmod | grep sof and paste the results

danao81, 2022-02-22:
> alsa-info is usually installed as /usr/sbin/alsa-info
> 
> Use "lsmod | grep sof" and paste the results

My bad, I was looking into "bin" folder and not "sbin"

Here it is the output:

dano@pop-os:/sbin$ lsmod | grep sof alsa-info
#This program is free software; you can redistribute it and/or modify
	local PATTERN='(alsa-(lib|oss|plugins|tools|(topology|ucm)-conf|utils|sof-firmware)|libalsa|tinycompress|sof-firmware)'

plbossart, 2022-02-23:
@danao81 do one of these

/usr/sbin/alsa-info
lsmod | grep sof

not both on the same line...

danao81, 2022-02-23:
[lsmod.txt](https://github.com/thesofproject/sof/files/8122627/lsmod.txt)
[alsa-info.txt](https://github.com/thesofproject/sof/files/8122628/alsa-info.txt)

Thank you for your patience

plbossart, 2022-02-23:
snd_sof is loaded
````
snd_sof               143360  4 snd_sof_intel_atom,snd_sof_acpi,snd_sof_acpi_intel_byt,snd_sof_intel_ipc
````
so if you do a 'modinfo snd_sof' you should see the sof_debug option, and then you can add it to an /etc/modprobe.d/alsa-base.conf file (or any file with a .conf extension).

options snd_sof sof_debug=1

Make sure to copy this verbatim, in your comment above you mentioned snd_sof_debug which does not exist indeed

danao81, 2022-02-23:
Ok, sorry for my lack of skill but i never go so deep in the file system except for some gui modding.

So I 
[modinfo _snd_sof.txt](https://github.com/thesofproject/sof/files/8126212/modinfo._snd_sof.txt)

and i can see "sof_debug:SOF core debug options (0x0 all off) (int)" as ending string.

Now how and where can I add 

options snd_sof sof_debug=1

to alsa-base.conf?

danao81, 2022-02-23:
> sudo gedit /etc/modprobe.d/sof.conf
> 
> add options snd_sof sof_debug=1 to first line
> 
> close and reboot.
> 
> I can't help you beyond that.

Done. 

All is working perfectly; i hope it will help other people with the same issue

Thank you very much for your help

hedmo, 2022-02-25:
hi again . is there a way to reload the sound stack?

plbossart, 2022-02-25:
> hi again . is there a way to reload the sound stack?

yes, see the scripts we use to unload/reload modules

https://github.com/thesofproject/sof-test/tree/main/tools/kmod

but that would not fully reset the DSP unfortunately.

lgirdwood, 2022-02-26:
@plbossart does this mean the BYT flows for S3 need to follow the module removal programming flows ?

plbossart, 2022-02-28:
Thanks @hedmo for the results. I trust that you've repeated the results multiple times

a) sof_remove.sh
b) hibernate
c) sof_insert.sh

I see no obvious difference @lgirdwood @ranj063 

````c
static int byt_suspend(struct snd_sof_dev *sdev, u32 target_state)
{
	byt_reset_dsp_disable_int(sdev);

	return 0;
}

static int byt_remove(struct snd_sof_dev *sdev)
{
	byt_reset_dsp_disable_int(sdev);

	return 0;
}

static int byt_resume(struct snd_sof_dev *sdev)
{
	/* enable BUSY and disable DONE Interrupt by default */
	snd_sof_dsp_update_bits64(sdev, DSP_BAR, SHIM_IMRX,
				  SHIM_IMRX_BUSY | SHIM_IMRX_DONE,
				  SHIM_IMRX_DONE);

	return 0;
}
static int byt_acpi_probe(struct snd_sof_dev *sdev)
{

<bunch of inits, then>

	/* enable BUSY and disable DONE Interrupt by default */
	snd_sof_dsp_update_bits64(sdev, DSP_BAR, SHIM_IMRX,
				  SHIM_IMRX_BUSY | SHIM_IMRX_DONE,
				  SHIM_IMRX_DONE);

	/* set default mailbox offset for FW ready message */
	sdev->dsp_box.offset = MBOX_OFFSET;

	return ret;
}
````

hedmo, 2022-02-28:
@plbossart  . Yes i have double checked many Times with Four different kernels to. To be more correct :

a) sof_remove.sh
b) hibernate
c) sof_remove.sh
d) sof_insert.sh

Is how i am doing it via hibernate script and Always worked.

Regards

lgirdwood, 2022-03-01:
It must be something down in the PCI layer ? i.e. when the PCI device is unregistered ?

[12 more replies]

closed 2022-10-10 by plbossart