thesofproject/sof #4676: [BUG] DSP panic when the FW reload or initialize the trace on the TGLH/ADLP/TGLU/ICL/CML/GLK

1994lwz, 2021-08-25:
**Describe the bug**
DSP panic when the FW reload or initialize the trace on the TGLH/ADLP/TGLU/ICL/CML/GLK

**To Reproduce**
do the test such as: suspend-resume, kmod-load-unload, or runtime-pm-status.

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: eb558167
SOF Branch: main
SOF Commit: 7df367451d0f
Platform: TGLH/ADLP/TGLU/ICL/CML/GLK

**Screenshots or console output**
[console]
```
rtcwake: assuming RTC uses UTC ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Aug 24 23:16:12 2021
2021-08-24 23:16:13 UTC [REMOTE_COMMAND] sleep for 5
2021-08-24 23:16:18 UTC [REMOTE_INFO] Check for the kernel log status
declare -- cmd="journalctl_cmd --since=@1629846961"
2021-08-24 23:16:18 UTC [ERROR] Caught kernel log error
===========================>>
[ 6843.082716] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error : DSP panic!
[ 6843.082744] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 6843.082755] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 6843.082982] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: runtime exception
[ 6843.082990] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace point 00004000
[ 6843.082998] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: panic at :0
[ 6843.083010] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops
[ 6843.083017] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[ 6843.083025] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000
[ 6843.083036] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe0313e5 EPC2     0xbe053696 EPC3     0x00000000 EPC4    0x00000000
[ 6843.083047] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 6843.083055] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00060520 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[ 6843.083064] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[ 6843.083073] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0xbe08aeb0
```
[dmesg]
```
[dmesg.txt](https://github.com/thesofproject/sof/files/7043218/dmesg.txt)

[ 6843.082105] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: firmware boot complete
[ 6843.082130] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[ 6843.082150] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000002 at  00000048
[ 6843.082166] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000004 at  00000044
[ 6843.082176] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000000 at  00000044
[ 6843.082205] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[ 6843.082258] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[ 6843.082268] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda_dsp_stream_setup_bdl: period_bytes:0x0
[ 6843.082276] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda_dsp_stream_setup_bdl: periods:1
[ 6843.082366] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: snd_sof_init_trace_ipc: stream_tag: 1
[ 6843.082390] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x90030000: GLB_TRACE_MSG
[ 6843.082716] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error : DSP panic!
[ 6843.082729] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[ 6843.082744] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 6843.082755] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 6843.082982] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: runtime exception
[ 6843.082990] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace point 00004000
[ 6843.082998] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: panic at :0
[ 6843.083010] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops
[ 6843.083017] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[ 6843.083025] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000
[ 6843.083036] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe0313e5 EPC2     0xbe053696 EPC3     0x00000000 EPC4    0x00000000
[ 6843.083047] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 6843.083055] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00060520 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[ 6843.083064] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[ 6843.083073] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0xbe08aeb0
```

keyonjie, 2021-08-25:
Thanks for reporting, looks there is deference to NULL pointer issue in the DMA trace part.

keyonjie, 2021-08-25:
@1994lwz can you please help to check if the PR https://github.com/thesofproject/sof/pull/4678 help?

plbossart, 2021-08-25:
This was detected in Intel daily test #6047

This can be root-caused to https://github.com/thesofproject/sof/pull/4636

The question is: how come this wasn't found in CI tests? there's a post-mortem analysis needed on such a regression.

lgirdwood, 2021-08-25:
Not sure why CI missed this as it tests the trace on all platforms (jenkins and Internal), the reload part is also tested on the PM tests. @keyonjie are we good to close ?

marc-hb, 2021-09-13:
> The question is: how come this wasn't found in CI tests? there's a post-mortem analysis needed on such a regression.

I spent some time looking at the test results of daily run 6047 and the reason this escaped PR testing is because even in the longer daily run 6047 this particular crash is relatively rare (there is a number of  other, unrelated failures in that run). Only a few platforms have experienced it, not on every test and even when it happens it's not immediately but only after a fair number of test iterations. Confirmed by comment https://github.com/thesofproject/sof/pull/4678#discussion_r695494407