thesofproject/sof #4680: [BUG]

kv2019i, 2021-08-25:
Describe the bug
TGLH_RVP_NOCODEC_ZEPHYR ipc timed out when issuing GLB_STREAM_MSG: TRIG_STOP.

To Reproduce
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-h-nocodec.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50

Reproduction Rate
100%

Environment
Kernel Branch: topic/sof-dev
Kernel Commit: eb55816743ecf7a68a94e0306d069c5e50783722
SOF Branch: main
SOF Commit: 7df367451d0fa638b530c1c92f02d645845dbe53
Topology:/lib/firmware/intel/sof-tplg/sof-tgl-h-nocodec.tplg
Platform: TGL_RVP_NOCODEC_ZEPHYR

Console output
```
[1374480.474916] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[1374480.475125] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 0
[1374480.982904] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[1374480.982912] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[1374480.982921] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000781 rirb 00
[1374480.982923] sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[1374480.982929] sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[1374480.982931] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[1374480.982933] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[1374480.982936] sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[1374480.983031] sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[1374480.983033] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[1374480.983053] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x20140000 successful
```

kv2019i, 2021-08-30:
@lgirdwood This is something else. When DSP is overloaded (>100% workload), processing of TRIG_STOP sometimes fails. The DSP doesn't crash and subsequent actions succeed.

lgirdwood, 2021-08-30:
@kv2019i ok, lets wait for @lyakh triggering PR to merge. The 100% utilisation means we get in a state - do we get any data from the Zephyr LL scheduler ? i.e. is tick late or does work overrrun in logs ?

ujfalusi, 2021-08-31:
> @lgirdwood This is something else. When DSP is overloaded (>100% workload), processing of TRIG_STOP sometimes fails. The DSP doesn't crash and subsequent actions succeed. The overload situation does occur because we moved the audio buffers back to uncached now.

@kv2019i, the TRIG_STOP ipc message times out, Linux would think that the DSP is crashed (we get timeout on IPC message). I have pending new fw_state to be taken by Linux in this case and this would block further IPC communications until the DSP is rebooted.
But in this case the DSP is _just_ too busy to respond. Should the IPC timeout be treated as a feature from now on and only the panic message from DSP should be treated as crash? I would think that the firmware can crash in a  not clean way to be able to tell Linux that it crashed..

lgirdwood, 2021-08-31:
@ujfalusi @kv2019i if FW is too busy to respond I would expect to see something in the logs to say this. e.g. LL should count number of times (per 1024 LL perf logging cycle) that there is little overhead for other work. This overhead could just be a subtraction of current time to next LL tick time and if it's less than limit we increment the count.

1994lwz, 2021-10-19:
The issue is fixed by #4884. 
In daily test: 7467, the issue is not reproduced, and not reproduce by manual test, so close it.

lyakh, 2021-10-20:
@kv2019i are you seeing any large "perf" entries in the trace? If you can reproduce it locally, maybe you could attach a trace here?

lgirdwood, 2021-11-12:
@lyakh @kv2019i still an issue with the sync STOP PR now merged ?

lyakh, 2021-11-15:
@lgirdwood multiple-pipeline-playback.sh passes now in daily tests, @kv2019i good to close or were you reproducing this manually?

kv2019i, 2021-11-15:
@lyakh @lgirdwood Good to close -- I think it's Seppo's mixer optimization PR that makes the tet pass. It was also now merged.

miRoox, 2021-11-22:
Reproduced in recently daily tests.

XiaoyunWu6666, 2021-11-22:
happened again in inner daily test result 8364
on model TGLH_RVP_NOCODEC_ZEPHYR
in testcase=multiple-pipeline-all-50

[dmesg]
```
[ 1308.063842] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[ 1308.063855] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 1308.063859] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 1308.063899] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc00007b7 rirb 00
[ 1308.063906] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 1308.063943] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 1308.063949] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 1308.063953] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 1308.063957] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware exception
[ 1308.063962] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 1308.063998] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 1308.064118] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
[ 1308.064124] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 1308.064173] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x20240000 successful
[ 1308.064183] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1308.064213] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 1308.064223] kernel:  DMIC: ASoC: trigger FE cmd: 0 failed: -110
[ 1308.064281] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 10 dir 1
[ 1308.064382] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
```

kv2019i, 2022-01-04:
This should be fixed with Zephyr PR https://github.com/zephyrproject-rtos/zephyr/pull/41547

lgirdwood, 2022-01-27:
@miRoox @kv2019i good to close ?

miRoox, 2022-01-28:
It seems there is only the sof-logger issue (#5120) in this platform. I think it's good to close.

closed 2022-01-28 by kv2019i