thesofproject/sof #4699: [BUG] FW boot failure with SOF main when trace mailbox is enabled

ranj063, 2021-08-31:
**Describe the bug**
FW fails to boot when built with trace mailbox enabled.

What have you tried to diagnose or workaround this issue?
Bisection points to https://github.com/thesofproject/sof/commit/7df367451d0fa638b530c1c92f02d645845dbe53
Reverting the commit fixes the problem.

**To Reproduce**
Build FW with Trace->Trace Mailbox enabled in config

**Reproduction Rate**
100%

**Expected behavior**
No issues during FW boot.

**Impact**
show-stopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/a2f06997cc7fdb02665f7d0b9bd6221849911437
    * SOF: https://github.com/thesofproject/sof/commit/6b650bd95fe003cc8f13715aa53feb293e629b70

    
Error log:
[    8.414944] sof-audio-pci-intel-tgl 0000:00:1f.3: error: firmware boot failure
[    8.414988] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[    8.415028] sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[    8.415098] sof-audio-pci-intel-tgl 0000:00:1f.3: error: extended rom status:  0x5 0x0 0x4000 0x0 0x0 0x0 0x256051f 0x0
[    8.415138] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[    8.415169] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware -5

ranj063, 2021-08-31:
@keyonjie FYI

keyonjie, 2021-08-31:
Thanks for reporting @ranj063 

After debugging, just root caused this to be deadlock in the trace->lock, will submit a fix for it soon.

Good finding here, maybe we should add Not-default kconfig validation coverage to our CI also. @marc-hb @keqiaozhang

keyonjie, 2021-08-31:
@ranj063 the PR here: https://github.com/thesofproject/sof/pull/4700

marc-hb, 2021-09-14:
> Good finding here, maybe we should add Not-default kconfig validation coverage to our CI also. @marc-hb @keqiaozhang

CONFIG_TRACEM is likely to have a severe performance impact.

marc-hb, 2021-09-14:
BTW I rely on CONFIG_TRACEM for my transition to Zephyr "strategy", see (now reverted) commit 0def9056300a

PR #4760 (2021-09-15): Revert trace enable commits, fix DEBUG_TRACE_PTR() macro so it can be used early
This reverts commit 7df3674 + reverts its followup commit + 1 line fix  for DEBUG_TRACE_PTR()

It has been successfully tested in TEST PR #4758 

See commit messages, the main one copied here:

This restores the ability to use CONFIG_TRACEM (copy everything to
mailbox) without crashing, in other words it fixes #4699

This also fixes the other DSP panic #4676 and removes the need for
logical changes in #4678, which can be reverted too.

commit 7df3674 ("trace: enable trace after it is ready") was meant
to fix a cra [... cut]
commit f2c13f5e (2021-09-15): Revert "trace: enable trace after it is ready"

This reverts commit 7df367451d0fa638b530c1c92f02d645845dbe53.

This restores the ability to use CONFIG_TRACEM (copy everything to
mailbox) without crashing, in other words it fixes #4699

This also fixes the other DSP panic #4676 and removes the need for
logical changes in #4678, which can be reverted too.

commit 7df367451d0f ("trace: enable trace after it is ready") was meant
to fix a crash when tr_xxx() was used early. However I've used very
early tracing for months and it never caused any crash (see #4334)

I tried adding a tr_err() statement [... cut]
commit 5dd268e2 (2021-11-02): Revert "trace: enable trace after it is ready"

This reverts commit 7df367451d0fa638b530c1c92f02d645845dbe53.

This restores the ability to use CONFIG_TRACEM (copy everything to
mailbox) without crashing, in other words it fixes #4699

This also fixes the other DSP panic #4676 and removes the need for
logical changes in #4678, which can be reverted too.

commit 7df367451d0f ("trace: enable trace after it is ready") was meant
to fix a crash when tr_xxx() was used early. However I've used very
early tracing for months and it never caused any crash (see #4334)

I tried adding a tr_err() statement [... cut]
closed 2021-09-15 by lgirdwood