thesofproject/sof #4703: [BUG] ICL build failure with zephyr and XCC

aiChaoSONG, 2021-09-01:
ICL build failed with zephyr and XCC
```
/srv/home/jenkins/workspace/sof_generic_build/zephyrproject
/srv/home/jenkins/workspace/sof_generic_build/zephyrproject
+ west build --build-dir build-icl --board intel_adsp_cavs20 zephyr/samples/subsys/audio/sof
[2/10] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map /srv/home/jenkins/workspace/sof_generic_build/zephyrproject/build-icl/zephyr/zephyr_prebuilt.map 
: && ccache /srv/home/jenkins/xcc/install/tools/RG-2017.8-linux/XtensaTools/bin/xt-xcc   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr/zephyr_prebuilt.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/srv/home/jenkins/workspace/sof_generic_build/zephyrproject/build-icl/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/arch/arch/xtensa/core/startup/libarch__xtensa__core__startup.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/xtensa/intel_adsp/common/libintel_adsp_common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/ipm/libdrivers__ipm.a  zephyr/drivers/timer/libdrivers__timer.a  modules/sof/libmodules_sof.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/srv/home/jenkins/workspace/sof_generic_build/zephyrproject/build-icl/zephyr  -lgcc  -lhal  zephyr/arch/common/libisr_tables.a  zephyr/soc/xtensa/intel_adsp/common/libintel_adsp_common.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N && cd /srv/home/jenkins/workspace/sof_generic_build/zephyrproject/build-icl/zephyr && /usr/bin/cmake -E echo
zephyr/linker_zephyr_prebuilt.cmd:436: non constant or forward reference address expression for section .noinit
ninja: build stopped: subcommand failed.
```

lgirdwood, 2021-09-01:
@aiChaoSONG any reason why CI did not catch this ? (I assume we are not building Zephyr for ICL yet ?)

lgirdwood, 2021-09-01:
@kv2019i @lyakh fyi

aiChaoSONG, 2021-09-01:
> @aiChaoSONG any reason why CI did not catch this ? (I assume we are not building Zephyr for ICL yet ?)

We don't test ICL Zephyr SOF in PR, but we do build it in daily, and the failure is observed in daily build.

lgirdwood, 2021-09-01:
> We don't test ICL Zephyr SOF in PR, but we do build it in daily, and the failure is observed in daily build.

Ok, can we add ICL to CI build.

aiChaoSONG, 2021-09-01:
@lgirdwood it is already there, for example, in PR https://github.com/thesofproject/sof/pull/4705,
[image]

kv2019i, 2021-09-01:
A couple of notes:
 - ICL zephyr build does not work (doesn't boot even when built with gcc) -> https://github.com/thesofproject/sof/issues/4539
 - the linker script for cavs20 was just changed on zephyr side-> https://github.com/zephyrproject-rtos/zephyr/pull/36055

kv2019i, 2021-09-01:
Needs to be fixed on Zephyr side, fix in https://github.com/zephyrproject-rtos/zephyr/pull/38201

keyonjie, 2021-09-02:
@aiChaoSONG can you help to check why the on device test is not run in my PR:
https://github.com/thesofproject/sof/pull/4521

keyonjie, 2021-09-02:
Thanks @aiChaoSONG it works now.

lgirdwood, 2021-09-02:
> > Can we at least remove the dependency of running on device test from the Zephyr building success, to unblock the CI PR test?
> > Or we should locally manage a downstream zephyr version where we can fix this kind of issues easily when needed? @kv2019i @lyakh @lgirdwood ?
> 
> The build failure on zephyr has no impact on test other than zephyr.

Thanks @aiChaoSONG I can see tests running again. Btw the APL Zephyr test has been removed, can we make the testing more granular so that only buildable targets are tested regardless of RTOS. i.e. non buildable targets would show up in the test results still and show "Not buildable"

aiChaoSONG, 2021-09-02:
@keyonjie @lgirdwood  we have planned several refinements/features to address this kind of issues, we will consider your suggestions to provide better CI user experience, thanks.

marc-hb, 2021-09-02:
Not sure why this bug is marked as specific to XCC. The same Zephyr commit regressed the ICL build with the Zephyr SDK too and the same zephyrproject-rtos/zephyr#38201 fix fixes the ICL build with the Zephyr SDK too.

With zephyr SDK 0.13.0:
```
Module Reading build-icl/zephyr/zephyr.elf.mod
  Found 46 sections, listing valid sections......
	No	LMA		VMA		End		Size	Type	Name
	1	0xbe010000	0xbe010000	0xbe01016a	0x16a	TEXT	.WindowVectors.text
	2	0xbe010180	0xbe010180	0xbe0101a9	0x29	TEXT	.Level2InterruptVector.text
	3	0xbe0101c0	0xbe0101c0	0xbe0101e9	0x29	TEXT	.Level3InterruptVector.text
	4	0xbe010200	0xbe010200	0xbe010229	0x29	TEXT	.Level4InterruptVector.text
	5	0xbe010240	0xbe010240	0xbe010269	0x29	TEXT	.Level5InterruptVector.text
	6	0xbe010280	0xbe010280	0xbe0102a9	0x29	TEXT	.DebugExceptionVector.text
	7	0xbe0102c0	0xbe0102c0	0xbe0102e9	0x29	TEXT	.NMIExceptionVector.text
	8	0xbe010300	0xbe010300	0xbe010303	0x3	TEXT	.KernelExceptionVector.text
	9	0xbe010340	0xbe010340	0xbe010352	0x12	TEXT	.UserExceptionVector.text
	10	0xbe0103c0	0xbe0103c0	0xbe0103c6	0x6	TEXT	.DoubleExceptionVector.text
	11	0xbe010400	0xbe010400	0xbe039000	0x28c00	TEXT	.text
	12	0xbe039000	0xbe039000	0xbe049b38	0x10b38	DATA	.rodata
	13	0xbe049b38	0xbe049b38	0xbe049ba0	0x68	DATA	initlevel
	14	0xbe049ba0	0xbe049ba0	0xbe049c30	0x90	DATA	devices
	15	0xbe049c30	0xbe049c30	0xbe049c48	0x18	DATA	k_p4wq_initparam_area
	16	0xbe049c48	0xbe049c48	0xbe049c80	0x38	DATA	log_const_sections
	17	0xbe049c80	0xbe049c80	0xbe049c90	0x10	DATA	log_backends_sections
	18	0xbe049c90	0xbe049c90	0xbe049ccc	0x3c	DATA	device_handles
	19	0xbe049ccc	0xbe049ccc	0xbe049d38	0x6c	DATA	.fw_ready
	20	0xbe049d40	0xbe049d40	0xbe04a1e0	0x4a0	DATA	.noinit
	21	0xbe04a200	0xbe04a200	0xbe04b000	0xe00	DATA	.data
	22	0xbe049d40	0xbe049d40	0xbe054540	0xa800	DATA	.cached
	23	0xbe04b000	0xbe04b000	0xbe04b4a8	0x4a8	DATA	sw_isr_table
	24	0xbe04b4a8	0xbe04b4a8	0xbe04b4c0	0x18	DATA	device_states
	25	0xbe04b4c0	0xbe04b4c0	0xbe04b4d0	0x10	DATA	k_sem_area
	26	0xbe04c000	0xbe04c000	0xbe1fee68	0x1b2e68	BSS	.bss

 module: input size 285475 (0x45b23) bytes 26 sections
 module: text 167547 (0x28e7b) bytes
    data 117928 (0x1cca8) bytes
    bss  1781352 (0x1b2e68) bytes

error: section overlap between build-icl/zephyr/zephyr.elf.mod:20 and build-icl/zephyr/zephyr.elf.mod:22
     [0xbe049d40 : 0xbe04a1e0] overlaps with [0xbe049d40 :0xbe054540]
```

marc-hb, 2021-09-02:
> i.e. non buildable targets would show up in the test results still and show "Not buildable"

Totally agree that build failures on one platform should not stop tests on other platforms. This being said, repeatedly building (or testing) that is not going to be fixed very soon is very time-consuming noise that forces everyone to constantly look at test results - or end up ignoring them.

Speaking of which: #4712 temporarily disables ICL in github actions (probably not in Jenkins)

marc-hb, 2021-09-03:
Fix zephyrproject-rtos/zephyr#38201 has been merged (no need to disable ICL)

lgirdwood, 2021-09-06:
@aiChaoSONG can we close now ?

aiChaoSONG, 2021-09-06:
@lgirdwood Yeah, Thi ICL build failure is fixed, but we have the same failure on CNL and TGL-H, @kv2019i FYI.

lgirdwood, 2021-09-06:
Closed by mistake - changed title.

aiChaoSONG, 2021-09-07:
related Zephyr issue: https://github.com/zephyrproject-rtos/zephyr/issues/38349

marc-hb, 2021-09-07:
It's the same error message but I doubt it's the same failure. The original issue affected only ICL with both GCC and XCC. It was fixed by  zephyrproject-rtos/zephyr#38201 and everything was building just fine. Then, later, a completely different Zephyr PR was merged and it caused zephyrproject-rtos/zephyr#38349 which fails in different places.

kv2019i, 2021-09-07:
Agree with @marc-hb , better to keep one bug per issue. We had two separate issues, one was a specific bug in cavs20 linker script. And now a more recent Zephyr regression that causes all cAVS XCC builds to fail. But yeah, water under the bridge, so let's keep this bug open. Basicly this is a SOF copy of https://github.com/zephyrproject-rtos/zephyr/issues/38349

kv2019i, 2021-09-10:
This is fixed now in main, closing.

closed 2021-09-10 by kv2019i