thesofproject/sof #4704: [BUG] imx8lup gcc build failed

fredoh9, 2021-09-01:
**Describe the bug**
I'm testing with SOF docker image.

With current docker image(image id=b6be3020af7f), there is no toolchain for it.
```
CMake Error at CMakeLists.txt:54 (project):
  The CMAKE_C_COMPILER:

    xtensa-imx8ulp-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:54 (project):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/sof/work/sof.git/build_imx8ulp_gcc/CMakeFiles/CMakeOutput.log".
See also "/home/sof/work/sof.git/build_imx8ulp_gcc/CMakeFiles/CMakeError.log".
```

With updated docker image(will share this information soon). it is looking for imx8ulp.toml file.
```
Scanning dependencies of target run_rimage
Scanning dependencies of target copy_dictionaries
error: can't open '/home/sof/work/sof.git/rimage/config/imx8ulp.toml' file
make[3]: *** [src/arch/xtensa/CMakeFiles/run_rimage.dir/build.make:57: src/arch/xtensa/CMakeFiles/run_rimage] Error 251
make[2]: *** [CMakeFiles/Makefile2:2164: src/arch/xtensa/CMakeFiles/run_rimage.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[100%] Built target copy_dictionaries
make[1]: *** [CMakeFiles/Makefile2:1980: src/arch/xtensa/CMakeFiles/bin.dir/rule] Error 2
make: *** [Makefile:684: bin] Error 2
```

**To Reproduce**
With docker image,
./scripts/docker-run.sh ./scripts/xtensa-build-all.sh -a 
./scripts/docker-run.sh ./scripts/xtensa-build-all.sh imx8lup

**Reproduction Rate**
100%

**Expected behavior**
build fine without any error

- sof HEAD: 8fa5ff579 ssp: only delay start for the first stream
- rimage HEAD: 9a26e45 config: apl: change image_size to the real SRAM size

dbaluta, 2021-09-01:
@fredoh9 can you update rimage at the latest HEAD.

Missing imx8ulp.toml file was introduced with 9716e10a3e000aa485b37c8cfd20ce184120809b

fredoh9, 2021-09-01:
with rimage's latest HEAD, imx8ulp builds fine.
`dcfcef80 config: Remove unused, platform specific, memory region definitions`

BTW, new SOF docker image was deployed from last night.

fredoh9, 2021-09-01:
For qemu test for imx8ulp, there are couple of places to change in scripts/qemu-check.sh. Do we need to update sofqemu docker image too?

lgirdwood, 2021-12-01:
@dbaluta close ?

dbaluta, 2021-12-01:
@dbaluta @fredoh9 there is no support for imx8ulp with qemu.

lgirdwood, 2021-12-02:
Thanks, fwiw - I see xtensa-build-all also works for all targets with GCC11. closing

closed 2021-12-02 by lgirdwood