thesofproject/sof #4706: [BUG]ipc timed out when TRIG_START and schedule ERROR on CML_RVP_SDW

XiaoyunWu6666, 2021-09-01:
**Describe the bug**
found in inner daily 6244 on CML_RVP_SDW
When trying to run aplay on multiple pipelines , IO error occurred with ipc timed out when issuing TRIG_START

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-cml-rt700-4ch.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50

**Reproduction Rate**
not sure
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: bed17ef
SOF Branch: main
SOF Commit: 8fa5ff579351
Topology:/lib/firmware/intel/sof-tplg/sof-cml-rt700-4ch.tplg
Platform: CML_RVP_SDW

**Screenshots or console output**
[console]
```
(in the first iteration)
2021-08-31 22:22:31 UTC [REMOTE_INFO] Testing: Jack Out [hw:0,0]
2021-08-31 22:22:31 UTC [REMOTE_COMMAND] aplay   -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/zero -q
2021-08-31 22:22:31 UTC [REMOTE_INFO] Testing: HDMI 1 [hw:0,5]
2021-08-31 22:22:31 UTC [REMOTE_COMMAND] aplay   -D hw:0,5 -c 2 -r 48000 -f S16_LE /dev/zero -q
2021-08-31 22:22:31 UTC [REMOTE_INFO] Testing: HDMI 2 [hw:0,6]
2021-08-31 22:22:31 UTC [REMOTE_COMMAND] aplay   -D hw:0,6 -c 2 -r 48000 -f S16_LE /dev/zero -q
2021-08-31 22:22:31 UTC [REMOTE_INFO] Testing: HDMI 3 [hw:0,7]
2021-08-31 22:22:31 UTC [REMOTE_INFO] wait 0.5s for aplay_opts()
2021-08-31 22:22:31 UTC [REMOTE_COMMAND] aplay   -D hw:0,7 -c 2 -r 48000 -f S16_LE /dev/zero -q
2021-08-31 22:22:31 UTC [REMOTE_INFO] checking pipeline status
2021-08-31 22:22:31 UTC [REMOTE_INFO] Letting playback/capture run for 5s
error: in logger_read(), fread(..., /sys/kernel/debug/sof/trace) failed: Operation not permitted(-1)
aplay: pcm_write:2061: write error: Input/output error
aplay: pcm_write:2061: write error: Input/output error
```
[dmesg]
```
[ 3489.592359] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 3489.592437] kernel: rt700 sdw:1:025d:0700:00: [rt700_sdw_write] 3501 <= 0003
[ 3489.592498] kernel: rt700 sdw:1:025d:0700:00: [rt700_sdw_write] 3501 <= 0000
[ 3489.592602] kernel: rt700 sdw:1:025d:0700:00: [rt700_sdw_write] 7321 83a1 <= b000
[ 3489.592643] kernel: rt700 sdw:1:025d:0700:00: [rt700_sdw_write] 3602 <= 0010
[ 3489.592726] kernel: rt700 sdw:1:025d:0700:00: [rt700_sdw_write] 7314 8394 <= b000
[ 3489.592782] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[ 3489.592948] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[ 3489.592952] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 6 dir 0 cmd 1
[ 3489.593079] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[ 3489.593084] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 7 dir 0 cmd 1
[ 3489.593181] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x14001e successful
[ 3489.593202] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x1a0]=0x34001e successful
[ 3489.593205] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 3489.593448] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 3489.593455] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 3489.593597] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x1c0]=0x44001e successful
[ 3489.593839] kernel: soundwire_intel soundwire_intel.link.1: Slave status change: 0x40
[ 3489.593877] kernel: rt700 sdw:1:025d:0700:00: [rt700_sdw_write] 00e0 <= 0001
[ 3489.593935] kernel: rt700 sdw:1:025d:0700:00: [rt700_sdw_write] 00f0 <= 0001
[ 3489.593936] kernel: rt700 sdw:1:025d:0700:00: rt700_clock_config complete, clk_freq=6000000
[ 3489.594500] kernel: rt700 sdw:1:025d:0700:00: Slave impl defined interrupt
[ 3489.594501] kernel: rt700 sdw:1:025d:0700:00: rt700_interrupt_callback control_port_stat=4
[ 3489.594689] kernel: soundwire_intel soundwire_intel.link.1: Slave status change: 0x20
[ 3489.850130] kernel: rt700 sdw:1:025d:0700:00: [rt700_sdw_read] b921 0000 => 80000000
[ 3489.850864] kernel: rt700 sdw:1:025d:0700:00: [rt700_sdw_read] 7520 85a0 9c20 aca0 => 00000400
[ 3489.851573] kernel: rt700 sdw:1:025d:0700:00: [rt700_sdw_read] 7520 85a0 9c20 aca0 => 00000000
[ 3489.851585] kernel: rt700 sdw:1:025d:0700:00: in rt700_jack_detect_handler, jack_type=0x3
[ 3489.851593] kernel: rt700 sdw:1:025d:0700:00: in rt700_jack_detect_handler, btn_type=0x0
[ 3490.049798] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 3490.049830] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 3490.097538] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x60040000 (msg/reply size: 12/12)
[ 3490.097560] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 3490.097566] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 3490.097581] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000781 rirb 00
[ 3490.097590] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 3490.097602] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 3490.097610] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 3490.097616] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 3490.097627] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[ 3490.097892] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 3490.097900] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 3490.097923] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 3490.097935] kernel:  HDMI 2: ASoC: trigger FE cmd: 1 failed: -110
[ 3490.097947] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 6 dir 0 cmd 0
[ 3490.097961] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 3490.097973] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: trace IO error
```
[etrace]
```
[       89202.756872] (       89202.757812) c0 edf-schedule       /schedule/edf_schedule.c:117  ERROR schedule_edf_task(), task already queued or running 3

```

lgirdwood, 2021-09-01:
@XiaoyunWu6666 thanks, do you have the FW register dump and stack trace ? (its part of dmesg)

plbossart, 2021-09-01:
@lgirdwood has anything changed in the firmware for triggers? I don't think we've had any changes for SoundWire merged on the kernel side.

lgirdwood, 2021-09-01:
@plbossart yes it's probably FW and not driver. @lyakh can you take a look. Thanks !

XiaoyunWu6666, 2021-09-02:
in inner daily 6272 on model=CML_RVP_NOCODEC in testcase=multiple-pipeline-playback-50, When trying to run aplay on multiple pipelines , IO error occurred with ipc timed out when issuing PIPE_NEW, in the 5th iteration
```
[ 4270.919815] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: stream_tag 3
[ 4270.920163] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 4270.920169] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[ 4271.417477] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 4271.417528] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 4271.425311] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x30100000 (msg/reply size: 48/20)
[ 4271.425332] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 4271.425338] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 4271.425359] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000181 rirb 00
[ 4271.425369] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 4271.425389] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 4271.425398] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 4271.425405] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 4271.425416] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[ 4271.425683] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 4271.425692] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 4271.425708] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to load widget PIPELINE.9.SSP2.OUT
[ 4271.425717] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed widget list set up for pcm 2 dir 0
[ 4271.425728] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -110
[ 4271.425743] kernel:  Port2: ASoC: soc_pcm_hw_params() failed (-110)
[ 4271.425764] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: trace IO error
[ 4271.425782] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 4271.917466] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 4271.917510] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 4271.929377] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx timed out for 0x80010000 (msg/reply size: 216/12)
[ 4271.929403] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed resetting DAI config for SSP2.OUT
```

lgirdwood, 2021-09-02:
@XiaoyunWu6666 we really need the FW stack trace from the FW crash AND the FW logs attached. 

The FW stack trace needs to be enabled in the kernel for all daily testing otherwise its impossible for developers to debug based on the message above. @keyonjie can you help @XiaoyunWu6666 configure her kernels to dump the FW stack trace ?

keyonjie, 2021-09-02:
@lgirdwood do you mean using the sof-coredump-reader.py to read out the FW call trace from the /sys/kernel/debug/sof/exception?

plbossart, 2021-09-02:
@lgirdwood if we have similar bugs for SoundWire and nocodec topologies, this points to a problem that can be checked on UpExtreme. My money is on the trace changes by @keyonjie  and the scheduling changes by @lyakh

plbossart, 2021-09-02:
Not able to reproduce the error reported in nocodec mode w/ test 6272:
TPLG=/lib/firmware/intel/sof-tplg/sof-cnl-nocodec.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50

SOF: 94db5c7b2 toolschain: update header files to support newer toolschain
kernel: 6a95ae1a6a50 fixup! ASoC: SOF: handle paused streams during system suspend

plbossart, 2021-09-02:
Never mind, I retried and it failed at iteration 35. See logs attached 
[test.tar.gz](https://github.com/thesofproject/sof/files/7100915/test.tar.gz)

XiaoyunWu6666, 2021-09-03:
@lgirdwood ,thanks for  @keyonjie 's help , I updated the coredump logs above

lyakh, 2021-09-03:
could somebody explain to me why time stamps are not intersecting between slogger and etrace output in daily test 6272 for this failing test case except for the banner trace "SHM: FW ABI:..."?

lyakh, 2021-09-03:
I managed to reproduce the problem and I think I have a fix - testing ATM. But the bad thing is, that PR CI doesn't catch this bug, so, when I submit my fix, we won't know whether it fixes the problem until we have merged it and have run a daily test... Can we adjust PR CI to also catch this? This bug isn't really CML-specific, it just happened to be the easiest to reproduce there. And it's also easier reproducible with dynamic pipelines. We do have (the same?) CML platforms in PR CI. Do they also have dynamic pipelines enabled in their topology?

lyakh, 2021-09-03:
#4715 should fix this issue

plbossart, 2021-09-03:
> I managed to reproduce the problem and I think I have a fix - testing ATM. But the bad thing is, that PR CI doesn't catch this bug, so, when I submit my fix, we won't know whether it fixes the [...]

@lyakh you can ask for a daily test on a specific PR, i'll send you the information through internal channels.

XiaoyunWu6666, 2021-09-07:
>  This bug isn't really CML-specific, it just happened to be the easiest to reproduce there. And it's also easier reproducible with dynamic pipelines. We do have (the same?) CML platforms in PR CI. Do they also have dynamic pipelines enabled in their topology?
@lyakh FYI, today we just found similar things happened to APL_UP2_NOCODEC too.
I just checked codes in sof , both APL_UP2_NOCODEC and CML_RVP_NOCODEC have dynamic pipelines enabled in their topology.
Thanks

lyakh, 2021-09-07:
> > This bug isn't really CML-specific, it just happened to be the easiest to reproduce there. And it's also easier reproducible with dynamic pipelines. We do have (the same?) CML platforms in PR CI. Do they also have dynamic pipelines enabled in their topology?
> 
> @lyakh FYI, today we just found similar things happened to APL_UP2_NOCODEC too.
> I just checked codes in sof , both APL_UP2_NOCODEC and CML_RVP_NOCODEC have dynamic pipelines enabled in their topology.
> Thanks
> 
> For APL_UP2_NOCODEC, please refer to the issue link here: #4720

@XiaoyunWu6666 nice, thanks for letting me know. If it's really the same issue, the same PR should fix them both.

XiaoyunWu6666, 2021-09-08:
did not happen in inner daily 6401 after the https://github.com/thesofproject/sof/pull/4715 is merged

closed 2021-09-10 by XiaoyunWu6666