thesofproject/sof #4717: [BUG] TestDaiGroupSdw Exception SofIpcErrorResponseException occurred  FW error code -11

wszypelt, 2021-09-06:
**Describe the bug**
Bug was found in:
23_00_TestDaiGroupSdw24000Hz24b32b4ch
23_00_TestDaiGroupSdw48000Hz32b32b4ch
23_00_TestDaiGroupSdw8000Hz24b32b2ch
23_00_TestDaiGroupSdw16000Hz16b16b8ch
23_01_TestDaiGroupSimpleStreamOperation8000Hz24b32b2ch
23_01_TestDaiGroupSimpleStreamOperation16000Hz16b16b8ch
23_01_TestDaiGroupSimpleStreamOperation24000Hz24b32b4ch
23_01_TestDaiGroupSimpleStreamOperation48000Hz32b32b4ch

the content of the error:
ERROR:Exception SofIpcErrorResponseException occurred at 'C:\\work\\KK\\cavs_scripts-py\\utilities\\ipc_driver.py':730 FW replied with error. Error code: -11 0x00000005 0x00000000 0x00004000

**Last working build: SOF_FW_MASTER_01.01.00.2602**

**Topology**
```
Topology:
                       pipe_plb1
                     +----------------------------+
    +-----------+    | +------+   +-------------+ |
    | Host1 plb |----->|Buffer|-->| Dai ALH G=1 | |------------------------------------------+
    +-----------+    | +------+   +-------------+ |                                          |
                     +----------------------------+                                          |
                       pipe_plb2                                                             |
                     +----------------------------+                                          |
    +-----------+    | +------+   +-------------+ |                                          |
    | Host2 plb |----->|Buffer|-->| Dai ALH G=1 | |---------------------------------------+  |
    +-----------+    | +------+   +-------------+ |                                       v  v
                     +----------------------------+                                     +-------+
                                                                                        |       |
                                                                                        |  ALH  |
                                                                                        |       |
                      pipe_cap1                                                         +-------+
    +------------------------------------------------------------+                        |  |
    | +----------+   +------+   +-----+   +------+   +---------+ |                        |  |
    | | Host cap |<--|Buffer|<--| Mux |<--|Buffer|<--| Dai ALH |<-------------------------+  |
    | +----------+   +------+   +-----+   +------+   +---------+ |                           |
    +------------------------------^-----------------------------+                           |
                                   |                                                         |
                                   |     pipe_cap2                                           |
                                   |    +------------------------+                           |
                                   |    | +------+   +---------+ |                           |
                                   +------|Buffer|<--| Dai ALH | |<--------------------------+
                                        | +------+   +---------+ |
                                        +------------------------+

```
Description:
Playback and capture on SdW interface on two internal loopbacks. Playback pipeline have the same group_id.
Start first plb pipeline, wait 0.1s and than start second plb pipeline.
"Muxing" two capture pipelines into one using Mux component.
Record signal on capture pipeline with audio verification, checking if two dai's can play synchronously and checking if
number of silence sample before signal are equal in every channel.

**To Reproduce**
Run tests with diagnostic driver
23_00_TestDaiGroupSdw24000Hz24b32b4ch
23_00_TestDaiGroupSdw48000Hz32b32b4ch
23_00_TestDaiGroupSdw8000Hz24b32b2ch
23_00_TestDaiGroupSdw16000Hz16b16b8ch
23_01_TestDaiGroupSimpleStreamOperation8000Hz24b32b2ch
23_01_TestDaiGroupSimpleStreamOperation16000Hz16b16b8ch
23_01_TestDaiGroupSimpleStreamOperation24000Hz24b32b4ch
23_01_TestDaiGroupSimpleStreamOperation48000Hz32b32b4ch

**Reproduction Rate**
100%

[... 41 lines cut ...]
[     3519684.895833] (           9.114583) c0 ll-schedule        ./schedule/ll_schedule.c:309  num_tasks 3 total_num_tasks 3
[     3519692.916667] (           8.020833) c0 ll-schedule        ./schedule/ll_schedule.c:390  task add 0xbe080280 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3>
[     3519700.703125] (           7.786458) c0 ll-schedule        ./schedule/ll_schedule.c:394  task params pri 0 flags 0 start 0 period 1000
[     3519712.369792] (          11.666667) c0 ll-schedule        ./schedule/ll_schedule.c:306  new added task->start 155448688 at 155447557
[     3519719.453125] (           7.083333) c0 ll-schedule        ./schedule/ll_schedule.c:309  num_tasks 4 total_num_tasks 4
[     3519772.031250] (          52.578125) c0 mux          4.11       src/audio/mux/mux.c:703  mux_trigger(), command = 9
[     3519783.958333] (          11.927083) c0 alh-dai      4.3    src/drivers/intel/alh.c:29   alh_trigger() cmd 9
[     3519793.541667] (           9.583333) c0 alh-dai      4.1    src/drivers/intel/alh.c:29   alh_trigger() cmd 9
[     3519841.276042] (          47.734375) c0 mux          4.11       src/audio/mux/mux.c:703  mux_trigger(), command = 1
[     3519854.270833] (          12.994792) c0 alh-dai      4.3    src/drivers/intel/alh.c:29   alh_trigger() cmd 1
[     3519866.953125] (          12.682292) c0 alh-dai      4.1    src/drivers/intel/alh.c:29   alh_trigger() cmd 1
[     3523471.822917] (        3604.869873) c0 pipe         1.3   ....../pipeline-stream.c:263  pipe trigger cmd 9
[     3523483.489583] (          11.666667) c0 ll-schedule        ./schedule/ll_schedule.c:390  task add 0xbe080080 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3>
[     3523492.005208] (           8.515625) c0 ll-schedule        ./schedule/ll_schedule.c:394  task params pri 0 flags 0 start 0 period 1000
[     3523504.557292] (          12.552083) c0 ll-schedule        ./schedule/ll_schedule.c:306  new added task->start 155602288 at 155593173
[     3523513.776042] (           9.218750) c0 ll-schedule        ./schedule/ll_schedule.c:309  num_tasks 5 total_num_tasks 5
[     3523818.177083] (         304.401031) c0 dai          1.1            src/audio/dai.c:845  ERROR dai_comp_trigger(), already processing atomic trigger
[     3523827.343750] (           9.166667) c0 pipe         1.3   ....../pipeline-stream.c:348  ERROR pipeline_trigger(): ret = -11, host->comp.id = 0, cmd = 9
[     3523835.859375] (           8.515625) c0 pipe         1.3   ..../pipeline-schedule.c:73   ERROR pipeline_task(): failed to trigger components: -11
[     3523846.718750] (          10.859375) c0 ll-schedule        ./schedule/ll_schedule.c:115  task complete 0xbe080080 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3>
Skipped 0 bytes after the last statement. Wrap possible, check the start of the output for later logs.

```
**Logs**
[2021_09_06_14_58_25.zip](https://github.com/thesofproject/sof/files/7116214/2021_09_06_14_58_25.zip)

lgirdwood, 2021-09-06:
@wszypelt can you check with https://github.com/thesofproject/sof/pull/4715

wszypelt, 2021-09-06:
@lgirdwood 
with #4715 unfortunately still the same error

lgirdwood, 2021-09-06:
@lyakh fyi

wszypelt, 2021-09-07:
Last working build: SOF_FW_MASTER_01.01.00.2602 hash: 6b650bd

lgirdwood, 2021-09-08:
@lyakh this is testing atomic DAI trigger where we have 2 PCM which trigger their DAIs at the same time i.e.
1) ALSA PCM trigger PCM0
2) FW checks and sees that PCM0 and PCM1 have joined atomic DAIs so does not trigger PCM0 DAI.
3) ALSA PCM trigger PCM1
4) FW checks and sees that both PCM0 and PCM1 have triggered so now atomically triggers both DAIs.

closed 2025-07-25 by github-actions[bot]