thesofproject/sof #4720: [BUG]ipc timed out when TRIG_START and schedule ERROR on APL_UP2_NOCODEC

XiaoyunWu6666, 2021-09-07:
**Describe the bug**
found in inner daily 6362 on APL_UP2_NOCODEC
When trying to run aplay on multiple pipelines , IO error occurred with ipc timed out when issuing TRIG_START , in the 1st iteration

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50
**Reproduction Rate**
not sure
After succeeding to reproduce it , will check whether FW stack trace is available and paste it here
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 6a95ae1a
SOF Branch: main
SOF Commit: 287a5f9a2b1e
Topology:/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg
Platform:  APL_UP2_NOCODEC

**Screenshots or console output**
[console]
```
(in the first iteration)
aplay: pcm_write:2061: aplay: set_params:1407: Unable to install hw params:
```

[dmesg]
```
[ 4402.733198] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 4402.733217] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 4403.231286] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 4403.231325] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 4403.238367] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx timed out for 0x60040000 (msg/reply size: 12/12)
[ 4403.238536] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: preventing DSP entering D3 state to preserve context
[ 4403.238540] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump start ]------------
[ 4403.238550] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000381 rirb 00
[ 4403.238555] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 4403.238562] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 4403.238566] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump end ]------------
[ 4403.238570] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump start ]------------
[ 4403.238576] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: status: fw entered - code 00000005
[ 4403.238620] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[ 4403.238624] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump end ]------------
[ 4403.238652] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at soc_component_trigger on 0000:00:0e.0: -110
```
[etrace]
```

[       49388.435537] (           0.000000) c0 edf-schedule       /schedule/edf_schedule.c:117  ERROR schedule_edf_task(), task already queued or running 3
[     7565757.824364] (     7565758.000000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[     7565763.553530] (           5.729167) c0 dai          3.14           src/audio/dai.c:892  ERROR dai_report_xrun(): underrun due to no data available
[     7565769.022280] (           5.468750) c0 dai          3.14           src/audio/dai.c:687  ERROR comp_underrun(): dev->comp.id = 14, source->avail = 0, copy_bytes = 0
[     7565775.168113] (           6.145833) c0 pipe         3.15  ....../pipeline-stream.c:318  ERROR pipeline_trigger_run(): ret = -22, host->comp.id = 12, cmd = 8
[     7565781.261863] (           6.093750) c0 pipe         3.15  ......../pipeline-xrun.c:159  ERROR pipeline_xrun(): Pipelines notification about XRUN failed, ret = -22
[     7565794.647279] (          13.385416) c0 pipe         3.15  ....../pipeline-stream.c:205  ERROR pipeline_copy(): ret = -61, start->comp.id = 14, dir = 1
[     7565800.220195] (           5.572917) c0 pipe         3.15  ..../pipeline-schedule.c:106  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.

```

XiaoyunWu6666, 2021-09-07:
related issue here : #4706

paulstelian97, 2021-09-07:
Coredump is full of zero, suspicious. Maybe hang?

XiaoyunWu6666, 2021-09-08:
did not happen in inner daily 6401 after the https://github.com/thesofproject/sof/pull/4715 is merged

closed 2021-09-10 by XiaoyunWu6666