thesofproject/sof #4742: [BUG]ipc timed out when pause-release while capturing on TGLH_RVP_NOCODEC_ZEPHYR

XiaoyunWu6666, 2021-09-10:
**Describe the bug**
When checking pause-release while capturing , TGLH_RVP_NOCODEC_ZEPHYR ipc timed out when TRIG_RELEASE.
This happened happened  twice in inner daily tests . One was 6422 , the other was 6472.
The 6422 was a daily test with Kernel PR: 3141 whille 6472 was today’s daily test ( alsothe first daily that after Kernel PR: 3141 got merged ).
So this issue is probably brought by Kernel PR: 3141 https://github.com/thesofproject/linux/pull/3141 Merge/sound upstream 20210907

**To Reproduce**
 TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-h-nocodec.tplg ~/sof-test/test-case/check-pause-resume.sh -c 100 -m capture

**Environment**
 Kernel Branch: topic/sof-dev
Kernel Commit: 85398998
storage: 85398998
SOF Branch: main
SOF Commit: 9a231915ea28
storage: 9a231915ea28
Topology: /lib/firmware/intel/sof-tplg/sof-tgl-h-nocodec.tplg
Platform: TGLH_RVP_NOCODEC_ZEPHYR

**Screenshots or console output**
[dmesg]
```
[ 2841.112477] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 11 dir 1 cmd 1
[ 2841.113081] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x2024001e successful
[ 2841.113090] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 2841.114381] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 2841.365848] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 11 dir 1 cmd 3
[ 2841.365867] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 2841.366402] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 2841.366430] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x20240000 successful
[ 2841.366507] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 11 dir 1 cmd 4
[ 2841.366521] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x24001e successful
[ 2841.366531] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[ 2841.868116] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60070000 (msg/reply size: 12/12)
[ 2841.868136] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 2841.868141] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 2841.868173] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00
[ 2841.868181] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 2841.868223] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 2841.868229] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 2841.868235] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 2841.868274] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 2841.868367] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 2841.868373] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 2841.868382] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 2841.868392] kernel:  DMIC16kHz: ASoC: trigger FE cmd: 4 failed: -110
[ 2841.868404] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 11 dir 1 cmd 3
[ 2841.868416] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 2842.372106] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60060000 (msg/reply size: 12/12)
[ 2842.372157] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x20240000 successful
[ 2842.372207] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 2842.372217] kernel:  DMIC16kHz: ASoC: trigger FE cmd: 3 failed: -110
```

XiaoyunWu6666, 2021-09-13:
@kv2019i  hi kai , do you have any new comments for this issue ?
I just found this issue 'gone' in daily tests, but I have no idea what happened because no fix PR is mentioned here.

I am double checking whether this issue did really go , or just become less likely to reproduce

XiaoyunWu6666, 2021-09-15:
happened in inner daily 6601 again

lgirdwood, 2021-09-20:
This looks like a deadlock or HW hang as stack is empty.

XiaoyunWu6666, 2021-11-01:
> Today's test is clean as well, can we close @XiaoyunWu6666 ?

YES , haven't seen it for quite long and also fix PR merged

closed 2021-11-01 by XiaoyunWu6666