thesofproject/sof #4746: [BUG] pipeline_comp_trigger() has a logic error

lyakh, 2021-09-10:
**Describe the bug**
`pipeline_comp_trigger()` effectively does this:
```
	err = comp_trigger(current, ppl_data->cmd);
	if (err < 0 || err == PPL_STATUS_PATH_STOP)
		return err;

	return pipeline_for_each_comp(current, ctx, dir);
```
and it is called from `pipeline_trigger_run()` as this:
```
	struct pipeline_walk_context walk_ctx = {
		.comp_func = pipeline_comp_trigger,
		.comp_data = &data,
		.skip_incomplete = true,
	};

	ret = walk_ctx.comp_func(host, NULL, &walk_ctx, host->direction);

	return ret;
```
Also note, that `pipeline_for_each_comp()` ignores non-negative return codes from the iterator function:
```
	int err = ctx->comp_func(buffer_comp, buffer,
				 ctx, dir);
	return 0;
```
This means, that if `pipeline_comp_trigger()` returns `PPL_STATUS_PATH_STOP` for the first component in the pipeline, i.e. when directly called from `pipeline_trigger_run()`, that return code will be propagated back to the caller. However, if that code is returned by one of later pipeline components, it will be ignored. This is probably not the intended behaviour. Note, that this code has been significantly altered by my recent commits, e.g. https://github.com/thesofproject/sof/pull/4559/commits/e8d2226fec15dafa11fecb7c0aeb31441e581ef8 but this behaviour has been preserved and can be verified in the code base before that commit.

lgirdwood, 2021-09-10:
@lyakh  - not intended behavior - we should respect the result and STOP. Can you fix ?

lyakh, 2021-11-16:
fixed by #4868

closed 2021-11-16 by lyakh