thesofproject/sof #4751: [BUG][APL] failed to boot DSP firmware - unknown sof_ext_man header type 3 size 0x30

TLMcNulty, 2021-09-12:
**Describe the bug**
Running an Ubuntu flavor called Gallium on an Acer Chromebook 15.
```
user@dut:~$ uname -a
Linux galliumos 5.14.2-051402-generic #202109080331 SMP Wed Sep 8 07:35:12 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

**To Reproduce**
Enable closed case debugging on an Apollo Lake device, in this case CB515-1H, install Gallium or Ubuntu, patch to 5.14 kernel. 

**Reproduction Rate**
100% of the time

**Expected behavior**
dWould expect codec probe to succeed and DSP to boot
```
[    8.985646] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[    8.985945] sof-audio-pci-intel-apl 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    8.991878] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[    9.039377] sof-audio-pci-intel-apl 0000:00:0e.0: hda codecs found, mask 4
[    9.062079] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 1:7:0-47d07
[    9.062090] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[    9.062107] sof-audio-pci-intel-apl 0000:00:0e.0: unknown sof_ext_man header type 3 size 0x30
[   12.094830] sof-audio-pci-intel-apl 0000:00:0e.0: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[   12.094857] sof-audio-pci-intel-apl 0000:00:0e.0: error: extended rom status:  0x81000012 0x2c 0x0 0x0 0x0 0x0 0x1522100 0x0
[   12.094860] sof-audio-pci-intel-apl 0000:00:0e.0: error: load fw failed ret: -110
[   12.094893] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to reset DSP
[   12.094896] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to boot DSP firmware -110
[   12.145519] sof-audio-pci-intel-apl 0000:00:0e.0: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[   12.145535] sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp core reset failed: core_mask 1
[   12.145948] sof-audio-pci-intel-apl 0000:00:0e.0: error: sof_probe_work failed err: -110
```

Would expect HDA driver fallback? Unsure on this platform
```
[  384.281993] snd_hda_intel 0000:00:0e.0: codec_mask forced to 0x0
[  384.282144] snd_hda_intel 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[  384.296471] snd_hda_intel 0000:00:0e.0: no codecs initialized
```

**Impact**
Annoyance. Bluetooth audio works but internal audio is an important part of platform validation for configuring these devices. If a machine has speakers they should work.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.14 {7d2a07b769330c34b4deabeed939325c77a7ec2f}
    * SOF: v1.8 {bd3a20e4745b56739d7471ef4a3d9bc9faee045f}
    * SOF: v1.7 {6bd5501e6ac3f29dde42a28ffe03a2942aa60640}
2) Name of the topology file
    * Topology: {FILE} - Not sure how to answer this one. 
3) Name of the platform(s) on which the bug is observed.
    * Platform: Apollo Lake

**Screenshots or console output**
Switching between DSP drivers
```
townson@galliumos:~$ sudo dmesg | grep -E 'snd|sof'
[    0.287102] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.287104] software IO TLB: mapped [mem 0x00000000769ee000-0x000000007a9ee000] (64MB)
[    7.123443] xor: measuring software checksum speed
[    8.985646] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[    8.985945] sof-audio-pci-intel-apl 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    8.991878] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[    9.039377] sof-audio-pci-intel-apl 0000:00:0e.0: hda codecs found, mask 4
[    9.062079] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 1:7:0-47d07
[    9.062090] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[    9.062107] sof-audio-pci-intel-apl 0000:00:0e.0: unknown sof_ext_man header type 3 size 0x30
[   12.094830] sof-audio-pci-intel-apl 0000:00:0e.0: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[   12.094857] sof-audio-pci-intel-apl 0000:00:0e.0: error: extended rom status:  0x81000012 0x2c 0x0 0x0 0x0 0x0 0x1522100 0x0
[   12.094860] sof-audio-pci-intel-apl 0000:00:0e.0: error: load fw failed ret: -110
[   12.094893] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to reset DSP
[   12.094896] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to boot DSP firmware -110
[   12.145519] sof-audio-pci-intel-apl 0000:00:0e.0: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[   12.145535] sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp core reset failed: core_mask 1
[   12.145948] sof-audio-pci-intel-apl 0000:00:0e.0: error: sof_probe_work failed err: -110
[  384.281993] snd_hda_intel 0000:00:0e.0: codec_mask forced to 0x0
[  384.282144] snd_hda_intel 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[  384.296471] snd_hda_intel 0000:00:0e.0: no codecs initialized
```

No sof paths under kernel/debug.

```
dut# ls /sys/kernel/debug/    
acpi	   cleancache	     devices_deferred  dynamic_debug	   gpio		 iosf_sb  mei0		    pm_genpd	resctrl		  suspend_stats  usb
asoc	   clear_warn_once   dma_buf	       energy_model	   hid		 iwlwifi  mmc0		    pwm		sched		  swiotlb	 virtio-ports
bdi	   clk		     dmaengine	       error_injection	   ieee80211	 kfence   mmc1		    ras		slab		  sync		 wakeup_sources
block	   cros_ec	     dma_pools	       extfrag		   iio		 kprobes  opp		    regmap	sleep_time	  telemetry	 x86
bluetooth  devfreq	     domains	       fault_around_bytes  intel_lpss	 kvm	  pinctrl	    regulator	soundwire	  tracing	 zram
cec	   device_component  dri	       frontswap	   interconnect  mce	  pkg_temp_thermal  remoteproc	split_huge_pages  ttm		 zswap
```

plbossart, 2021-09-13:
@TLMcNulty we've never enabled Chromebooks on ApolloLake. Well, I did for development but officially the only 'official' firmware for APL is the Intel SST one. I am not too use how you ended-up using SOF, on APL we only support the open Up2 board.

If you want to use SOF, you would have to use the Intel signed firmware, not the one with the community key used for official chromebooks since Gemini Lake.

TLMcNulty, 2021-09-13:
Thanks for the reply, I've just been seeing what's failing where while trying to get this platform to work. 

I'll focus on whatever the SST has been erroring on primarily. This bug is then not relevant or helpful. Thanks @plbossart

closed 2021-09-13 by TLMcNulty

bphd, 2025-03-09:
I use Intel proprietary firmware, not on a ChromeBook, and I experience [    4.206308] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30