thesofproject/sof #4756: [BUG] [ADL-M RVP]: Failed to load DSP topology during warm boot tests with RT711 on board Soundwire codec

jairaj-arava, 2021-09-13:
**Describe the bug**
During warm boot tests, we see DSP topology load failure randomly.

**To Reproduce**
1.Boot the Chromebook.
2. Copy the folder warm_cold_boot_device_check  script to the device.
3. Initiate test with command: ./start_test.sh 5 20 -> warm boot

**Reproduction Rate**
The issue happens randomly once or thrice so far in 1000 cycles.

**Expected behavior**
Sound card should be loaded in all warm boot cycles without any failure.

**Impact**
User may experience audio not working at any time due to random failure.

dmesg error logs:
	Line 2177: 2021-09-10T04:20:22,186809-07:00 sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
	Line 2206: 2021-09-10T04:20:22,425045-07:00 sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:18:1 Kernel ABI 3:18:0
	Line 2216: 2021-09-10T04:20:22,440359-07:00 sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: invalid widget size
	Line 2218: 2021-09-10T04:20:22,455418-07:00 sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: topology: could not load header: -22
	Line 2219: 2021-09-10T04:20:22,455428-07:00 sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
	Line 2221: 2021-09-10T04:20:22,466202-07:00 sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
	Line 2223: 2021-09-10T04:20:22,476542-07:00 sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
	Line 2224: 2021-09-10T04:20:22,476586-07:00 sof_sdw sof_sdw: ASoC: failed to instantiate card -22
	Line 2224: 2021-09-10T04:20:22,476586-07:00 sof_sdw sof_sdw: ASoC: failed to instantiate card -22
	Line 2228: 2021-09-10T04:20:22,505175-07:00 sof_sdw sof_sdw: snd_soc_register_card failed -22
	Line 2228: 2021-09-10T04:20:22,505175-07:00 sof_sdw sof_sdw: snd_soc_register_card failed -22
	Line 2230: 2021-09-10T04:20:22,522316-07:00 sof_sdw: probe of sof_sdw failed with error -22
	Line 2230: 2021-09-10T04:20:22,522316-07:00 sof_sdw: probe of sof_sdw failed with error -22

Attached the dmesg log: 
[dmesg.txt](https://github.com/thesofproject/sof/files/7153966/dmesg.txt)

jairaj-arava, 2021-09-13:
@Vamshigopal @sathyap-chrome

plbossart, 2021-09-13:
Can we add more information on the 'ASoC: invalid widget size' error? 
@ranj063 any ideas what this could be due to?

jairaj-arava, 2021-09-14:
@plbossart , shall I get the logs with dynamic debug logs enabled? or will there be any timings issues that we can expect if we enable full logs?