thesofproject/sof #4773: [BUG]

prof-dr-pepper, 2021-09-15:
**Describe the bug**
Sound driver seems to fail to load properly on Dell Latitude 7320 Detachable (see hardware probe here https://linux-hardware.org/?probe=21c50317b2) and therefore sound is not available at the device on linux.
Even though the sound adapter (hardware ID: 8086:a0c8) seems to be supported by the linux kernel the device does not work. Browsing on the internet this issue seems to happen on all Dell Latitude 7320 Detachable systems. I tested with kubuntu 21.04 with kernel 5.11.0-34-generic (however same issue occurs with latest Manjaro linux and kernels up to 5.14). Furthermore looking at dmesg I saw a lot of (thousands and millions) suppressed IRQ callbacks that keep the CPU busy (one core seems to be busy for about 50% because of that). Not loading the driver (via grub: snd_hda_intel.dmic_detect=0) the CPU issue is gone but of course then sound will be disabled.

**To Reproduce**
1. Run kubuntu 21.04 live image (sha256sum of ISO: e705e84ed8fa820f1b7f4d286d24e0eb97561a5b2d47421bf7ea0381103788b7) from USB stick
2. After boot look at dmesg (example see attachment)

**Reproduction Rate**
Issue is reproducible all the time (also seems with other linux kernel versions and linux distros).

**Expected behavior**
Linux boots up and loads driver without issues. Sound is working properly and CPU usage is near 0.

**Impact**
Using linux without sound is annoying at the least but during home office makes it also useless during audio conferences (however video is also not working on the device but that's another issue because there exists not webcam driver for this device yet). ;-)

**Environment**
- Dell Latitude 7320 Detachable
- Kubuntu 21.04 64bit (see ISO hash above)
- Stock ubuntu linux kernel 5.11.0-34-generic
- HW probe: https://linux-hardware.org/?probe=21c50317b2
- Docking station with SSD for linux installation was attached during HW probe (however issue also occurs when booting from kubuntu 21.04 live USB stick)
- 
**Screenshots or console output**
Here you can find my dmesg output directly after boot: [dmesg.txt](https://github.com/thesofproject/sof/files/7172540/dmesg.txt)

Please let me know if you need more information and how I can provide it.

Thanx,
Marcus

plbossart, 2021-09-15:
@prof-dr-pepper can you provide the results of 'alsa-info' please. Looks like you have a SoundWire device that is not known to us. It's usually a minor SKU value that we need to add, or that was added in more recent kernels.

prof-dr-pepper, 2021-09-16:
I added the alsa-info to the bug. You can find it here: http://alsa-project.org/db/?f=e7ce23315aee536587174f7f58f79812c1166d1a

Thanks for helping :)