thesofproject/sof #4779: [BUG] DMA trigger stop failed when suspend-resume during capturing on CML-SKU0983-SDW

1994lwz, 2021-09-17:
**Describe the bug**
DMA trigger stop failed when suspend-resume during capturing on CML-SKU0983-SDW

**To Reproduce**
Run command: "TPLG=/lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-mono-rt715.tplg ~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 5 -m capture"
The reproduction rate is 100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: a7bb845e
SOF Branch: main
SOF Commit: 9fadef789e80
Platform: CML-SKU0983-SDW

**Screenshots or console output**
[dmesg & slogger]
[dmesg.txt](https://github.com/thesofproject/sof/files/7182166/dmesg.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/7182167/slogger.txt)

[console]
```
2021-09-16 23:12:22 UTC Sub-Test: [REMOTE_COMMAND] Run the command: rtcwake -m mem -s 5
rtcwake: assuming RTC uses UTC ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Thu Sep 16 23:12:28 2021
2021-09-16 23:12:28 UTC Sub-Test: [REMOTE_COMMAND] sleep for 5
2021-09-16 23:12:33 UTC Sub-Test: [REMOTE_INFO] Check for the kernel log status
declare -- cmd="journalctl_cmd --since=@1631833937"
2021-09-16 23:12:34 UTC Sub-Test: [REMOTE_INFO] Check for the wakeup_count
2021-09-16 23:12:34 UTC Sub-Test: [REMOTE_INFO] ===== Round(4/5) =====
2021-09-16 23:12:34 UTC Sub-Test: [REMOTE_COMMAND] Run the command: rtcwake -m mem -s 5
rtcwake: assuming RTC uses UTC ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Thu Sep 16 23:12:40 2021
2021-09-16 23:12:48 UTC Sub-Test: [REMOTE_COMMAND] sleep for 5
arecord: suspend:1716: suspend: prepare error: Connection timed out
2021-09-16 23:12:53 UTC Sub-Test: [REMOTE_INFO] Check for the kernel log status
declare -- cmd="journalctl_cmd --since=@1631833949"
2021-09-16 23:12:53 UTC [ERROR] Caught kernel log error
===========================>>
[ 6554.616307] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: hda_dsp_stream_trigger: cmd 5 on dai_link "Jack In" (Capture, stream_tag: 2): timeout on STREAM_SD_OFFSET read
[ 6563.817077] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[ 6563.817746] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: hda_dsp_stream_trigger: cmd 0 on -- (Playback, stream_tag: 1): timeout on STREAM_SD_OFFSET read
[ 6563.817758] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: DMA trigger stop failed
[ 6563.817765] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 6563.817793] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: extended rom status:  0x5000001 0x0 0x0 0x0 0x0 0x0 0x1811102 0x0
[ 6563.817800] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 6563.817805] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: load fw failed ret: -110
[ 6563.817862] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to start DSP
[ 6563.817869] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to boot DSP firmware after resume -110
[ 6563.817875] kernel: PM: dpm_run_callback(): pci_pm_resume+0x0/0x80 returns -110
[ 6563.817906] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: PM: failed to resume async: error -110
[ 6563.819289] kernel: soundwire_intel soundwire_intel.link.0: Failed to power up link: -11
[ 6568.918927] kernel: rt711 sdw:0:025d:0711:00: Initialization not complete, timed out
[ 6568.918943] kernel: PM: dpm_run_callback(): acpi_subsys_resume+0x0/0x70 returns -110
[ 6568.918975] kernel: rt711 sdw:0:025d:0711:00: PM: failed to resume: error -110
[ 6568.920244] kernel: soundwire_intel soundwire_intel.link.1: Failed to power up link: -11
[ 6568.920603] kernel: soundwire_intel soundwire_intel.link.1: sdw_cdns_init failed: MCP_CONTROL_SW_RST is not cleared
[ 6568.920612] kernel: soundwire_intel soundwire_intel.link.1: sdw_cdns_init failed: MCP_CONTROL_CLK_STOP_CLR is not cleared
[ 6568.920620] kernel: soundwire_intel soundwire_intel.link.1: sdw_cdns_init failed: MCP_CONTROL_HW_RST is not cleared
[ 6568.922181] kernel: soundwire_intel soundwire_intel.link.1: intel_resume failed: MCP_CONTROL_SW_RST is not cleared
[ 6568.922187] kernel: soundwire_intel soundwire_intel.link.1: intel_resume failed: MCP_CONTROL_CLK_STOP_CLR is not cleared
```

XiaoyunWu6666, 2021-09-17:
this happened in 6200 6292 6655

plbossart, 2021-09-17:
Sorry @XiaoyunWu6666, not following you. Are you saying the "timeout on STREAM_SD_OFFSET" and "MCP_CONTROL_HW_RST is not cleared at iteration N"  errors are correlated somehow?

@bardliao this would be interesting indeed.

lgirdwood, 2021-09-17:
@XiaoyunWu6666 @1994lwz can someone try reverting 9fadef789e80b586c64cb963dad24f909d579afd and retest.
@ranj063 and I have a suspicion this is related to 
1) DSP -> host DMA traffic volume prior to suspend (in which case multiple capture streams would also show this).
2) Time between trace being disabled and DSP entering D3. More traffic may mean DMA takes longer to enter idle (and safe to enter D3).
@keyonjie fyi.

XiaoyunWu6666, 2021-09-18:
@lgirdwood @RanderWang @plbossart 
In today's inner daily 6686
both CML_SKU0983_SDW and CML_SKU0955_HDA have this issue

FYI , I tried to revert 9fadef7 and retest on CML_SKU0983_SDW (running suspend-resume-5-time during capture , for 100 rounds ) 
so far so good . haven't reproduce the issue (noted that this issue isn't 100% reproducible so this is just a reference)

ranj063, 2021-09-18:
@1994lwz @XiaoyunWu6666 could you please check if https://github.com/thesofproject/linux/pull/3166 helps fix the issue?

1994lwz, 2021-09-18:
I tried to revert 9fadef7 and retest on CML_SKU0955_HDA (running suspend-resume-5-time during capture , for 50 rounds ), haven't reproduce the issue.

XiaoyunWu6666, 2021-09-18:
> @1994lwz @XiaoyunWu6666 could you please check if [thesofproject/linux#3166](https://github.com/thesofproject/linux/pull/3166) helps fix the issue?

hi @ranj063  . I am afraid not , the  same error appears again.
@1994lwz will make a double confirm

1994lwz, 2021-09-18:
> > @1994lwz @XiaoyunWu6666 could you please check if [thesofproject/linux#3166](https://github.com/thesofproject/linux/pull/3166) helps fix the issue?
> 
> hi @ranj063 . I am afraid not , the same error appears again.
> kernel commit : 0620fb04b2d01
> sof commit ：509eef9c96b8
> PR ：3166
> @1994lwz will make a double confirm

the same error appears again on CML_SKU0955_HDA.

ranj063, 2021-09-18:
@XiaoyunWu6666 @1994lwz I have update my PR https://github.com/thesofproject/linux/pull/3166 to fix the issues during suspend-resume on capture on the CML Helios and the CML-HDA. Could you please double confirm?

@plbossart I need your help for the CML_SDW case.

lgirdwood, 2021-09-20:
@RanderWang @1994lwz @XiaoyunWu6666 has anyone tried reverting https://github.com/thesofproject/sof/commit/9fadef789e80b586c64cb963dad24f909d579afd AND trying multiple capture streams for this test. i.e. try 2 then 3 capture streams in this test.
This will help us determine if its more likely to be
1) Capture DMA traffic prior to D3
2) Capture (trace) DMA shutdown prior to D3.

ranj063, 2021-09-20:
> @ranj063 @1994lwz @XiaoyunWu6666 has anyone tried reverting [9fadef7](https://github.com/thesofproject/sof/commit/9fadef789e80b586c64cb963dad24f909d579afd) AND trying multiple capture streams for this test. i.e. try 2 then 3 capture streams in this test.
> This will help us determine if its more likely to be
> 
> 1. Capture DMA traffic prior to D3
> 2. Capture (trace) DMA shutdown prior to D3.

@lgirdwood I tried 2 capture streams during suspend with my kernel PR on the helios and it works fine. My PR fixes this issue on both the helios and the HDA laptop.

plbossart, 2021-09-20:
This issue and related PRs are really hard to follow. I ran a quick test on CML_SKU09C6_SDW (same as CML-SKU0983-SDW) and I don't see any issue. 

"TPLG=/lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-mono-rt715.tplg ~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 5 -m capture"
The reproduction rate is 100% <<< NOT REPRODUCED AT ALL.

We should really deal with separate sightings and validations, it's not helpful if we conflate HDaudio, Chromebook and SoundWire platforms in the same bucket. They use different links and there's no real rationale for errors being common.

kernel: 73e5ae02dd8c
SOF: 4de66271e

plbossart, 2021-09-20:
ok, so the command above is a bit misleading. I changed it to "TPLG=/lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-mono-rt715.tplg ~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 200 -m capture" and I see errors happening somewhat randomly, e.g. round 19, 2, 25.

we probably need to change the '5' value since it leads to missed errors in CI.

plbossart, 2021-09-20:
PR thesofproject/linux#3166 does not solve anything on  CML_SKU09C6_SDW, now trying to with a revert of 9fadef789e80b586c64cb963dad24f909d579afd

plbossart, 2021-09-20:
Reverting 9fadef7 works in my local tests, submitted PR #4785 for more tests.

@lgirdwood @ranj063 @1994lwz @keyonjie @marc-hb FYI

marc-hb, 2021-09-20:
Found another recent CML suspend timeout but it looks a bit different: https://sof-ci.01.org/sofpr/PR4771/build10385/devicetest/?model=CML_HEL_RT5682&testcase=check-suspend-resume-with-capture

```
[ 1481.879185] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1481.879870] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1481.880649] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x240002 timedout
[ 1481.880692] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: hda_dsp_stream_trigger: cmd 0 on dai_link "Port1" (Capture, stream_tag: 2): timeout on STREAM_SD_OFFSET read
[ 1481.880744] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1481.881007] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1481.881230] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: free stream 0 dir 1
[ 1481.896939] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
```

keyonjie, 2021-09-22:
> Reverting [9fadef7](https://github.com/thesofproject/sof/commit/9fadef789e80b586c64cb963dad24f909d579afd) works in my local tests, submitted PR #4785 for more tests.
> 
> @lgirdwood @ranj063 @1994lwz @keyonjie @marc-hb FYI

Not quite following here, @plbossart do you mean changing some Kconfig items will impact the DMA of the normal pipelines?

XiaoyunWu6666, 2021-09-22:
#4785 has been merged ,and in today's inner daily 6739 , CML_SKU0983_SDW and CML_SKU0955_HDA are good
No DMA trigger stop error happened.
Let's keep observing it

marc-hb, 2021-09-22:
> Not quite following here, @plbossart do you mean changing some Kconfig items will impact the DMA of the normal pipelines?

When the system is brittle or unstable then all bets are off.

ranj063, 2021-09-22:
@lgirdwood I think the right fix for this issue and #4558 is in Linux PR https://github.com/thesofproject/linux/pull/3167 and SOF PR #4793

plbossart, 2021-09-22:
@lgirdwood @keyonjie My theory for this timeout issue is a race condition where the firmware removes access to the L2 SRAM when we stop the DMA (by clearing the DGCS_GEN bit), and as a result if there are pending DMA transactions clearing the RUN bit from the driver side times out. 

By changing the way the trace works, and possibly adding more trace transactions, we may see this race condition more often.

The recommended sequence is to remove access to the L2 SRAM *AFTER* the DMA is stopped with a follow-up IPC. This is clearly indicated in the hardware programming sequence, it's implemented in the Windows driver but we didn't follow it in the Linux firmware/driver for some reason.

1994lwz, 2021-09-27:
Use the Linux PR #3167 and SOF PR #4793 to build the kernel and firmware, not reproduce on the cml-sku0983-sdw/ cml-sku0955-hda by test the suspend-resume during capturing 100 times

1994lwz, 2021-10-14:
This issue is fixed by #3167 and not reproduce on the recent daily, so close it.

marc-hb, 2021-11-02:
#4793 was replaced by #4820 and #4844

XiaoyunWu6666, 2021-11-08:
Since relevant PRs were already merged and  havent happened in inner daily tests for nearly 50 days . So close it

 relevant PRs : thesofproject/linux#3167 #4820 and #4844

keqiaozhang, 2021-11-30:
Observed this issue again when running suspend/resume stress testing on CML_SKU0983_SDW. Test ID:8054.
```
[ 1592.161340] kernel: rt1308 sdw:1:025d:1308:00: sdw_modify_slave_status: signaling enumeration completion for Slave 1
[ 1592.292468] kernel: rt1308 sdw:1:025d:1308:00: rt1308_io_init m_btl_l=0xffe2, m_btl_r=0x8
[ 1592.292481] kernel: rt1308 sdw:1:025d:1308:00: rt1308_io_init c_btl_l=0x2ef, c_btl_r=0x10
[ 1592.301784] kernel: rt1308 sdw:1:025d:1308:00: rt1308_io_init hw_init complete
[ 1592.301790] kernel: rt1308 sdw:1:025d:1308:00: sdw_handle_slave_status: signaling initialization completion for Slave 1
[ 1592.314783] kernel: soundwire_intel soundwire_intel.link.2: intel_resume: MCP_CONTROL_HW_RST is not cleared at iteration 0
[ 1592.316316] kernel: soundwire_intel soundwire_intel.link.2: intel_resume: MCP_CONTROL_HW_RST is not cleared at iteration 1
[ 1592.317774] kernel: soundwire_intel soundwire_intel.link.2: intel_resume: MCP_CONTROL_HW_RST is not cleared at iteration 2
[ 1592.319304] kernel: soundwire_intel soundwire_intel.link.2: intel_resume: MCP_CONTROL_HW_RST is not cleared at iteration 3
[ 1592.320779] kernel: soundwire_intel soundwire_intel.link.2: intel_resume: MCP_CONTROL_HW_RST is not cleared at iteration 4
[ 1592.322308] kernel: soundwire_intel soundwire_intel.link.2: intel_resume: MCP_CONTROL_HW_RST is not cleared at iteration 5
[ 1592.323777] kernel: soundwire_intel soundwire_intel.link.2: intel_resume: MCP_CONTROL_HW_RST is not cleared at iteration 6
[ 1592.325361] kernel: soundwire_intel soundwire_intel.link.2: intel_resume: MCP_CONTROL_HW_RST is not cleared at iteration 7
[ 1592.326844] kernel: soundwire_intel soundwire_intel.link.2: intel_resume: MCP_CONTROL_HW_RST is not cleared at iteration 8
[ 1592.328468] kernel: soundwire_intel soundwire_intel.link.2: intel_resume: MCP_CONTROL_HW_RST is not cleared at iteration 9
[ 1592.329841] kernel: soundwire_intel soundwire_intel.link.2: intel_resume failed: MCP_CONTROL_HW_RST is not cleared
```

plbossart, 2021-11-30:
wrong bug @keqiaozhang, this had nothing to do with DMA. This should be added in https://github.com/thesofproject/linux/issues/3012

closed 2021-11-30 by plbossart