thesofproject/sof #4794: [BUG][TGL] Stack pointer is somewhere in the buffer heap

cujomalainey, 2021-09-22:
**Describe the bug**
In debugging some memory issues, we realized the stack pointer is never in the region we expect `__sof_stack_start <-> __sof_stack_end` Our stack pointer is always pointing to somewhere in the buffer heap. This has been observed from tgl-012-drop-stable to main

**To Reproduce**
Log your stack pointer

**Reproduction Rate**
100%

**Expected behavior**
Stack pointer is in stack region

**Impact**
When scaling to larger stacks this breaks a lot things, a show stopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: N/A
    * SOF: d8a19a4d2062cde129f2a18e414b07ce9ab31c73
2) Name of the topology file
    * Topology: N/A
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL

**Screenshots or console output**
Test code
```
diff --git a/src/lib/alloc.c b/src/lib/alloc.c
index 8ea9e1db1..f2a642692 100644
--- a/src/lib/alloc.c
+++ b/src/lib/alloc.c
@@ -581,6 +581,8 @@ void heap_trace(struct mm_heap *heap, int size)
 void heap_trace_all(int force)
 {
        struct mm *memmap = memmap_get();
+       tr_err(&mem_tr, "%p", &memmap);
+
 
        /* has heap changed since last shown */
        if (memmap->heap_trace_updated || force) {
```

Error message
`[ 10799297.604167] (       16.979166) c0 memory                      src/lib/alloc.c:584  ERROR 0xbe251d80`

sof.map BSS section
```
 *(.bss.*)                                                                                                                                                                                                                                                                                  
 *(.gnu.linkonce.b.*)                                                                                                                         
 *(COMMON)                                                                                                                                                                                                                                                                                  
                0xbe0b0000                . = ALIGN (0x10000)                                                                                                                                                                                                                               
 *fill*         0xbe0a9500     0x6b00                                                                                                                                                                                                                                                       
                0xbe0b0000                _system_heap_start = ABSOLUTE (.)                                                                                                                                                                                                                 
                0xbe0b8000                . = (. + 0x8000)                                                                                                                                                                                                                                  
 *fill*         0xbe0b0000     0x8000                                                                                                                                                                                                                                                       
                0xbe0b8000                _system_heap_end = ABSOLUTE (.)                                                                                                                                                                                                                   
                0xbe0b8000                . = ALIGN (0x40)                                                                                                                                                                                                                                  
                0xbe0b8000                _system_runtime_heap_start = ABSOLUTE (.)                                                                                                                                                                                                         
                0xbe0bd000                . = (. + 0x5000)                                                                                                                                                                                                                                  
 *fill*         0xbe0b8000     0x5000                                                                                                         
                0xbe0bd000                _system_runtime_heap_end = ABSOLUTE (.)                                                             
                0xbe0bd000                . = ALIGN (0x1000)                                                                                  
                0xbe0bd000                _sof_stack_start = ABSOLUTE (.)                                                                     
                0xbe0be000                . = (. + 0x1000)                                                                                                                                                                                                                                  
 *fill*         0xbe0bd000     0x1000                                                                                                                                                                                                                                                       
                0xbe0be000                _sof_stack_end = ABSOLUTE (.)                                                                                                                                                                                                                     
                0xbe0be000                . = ALIGN (0x40)                                                                                                                                                                                                                                  
                0xbe0be000                _runtime_shared_heap_start = ABSOLUTE (.)                                                                                                                                                                                                         
                0xbe19e000                . = (. + 0xe0000)                                                                                                                                                                                                                                 
 *fill*         0xbe0be000    0xe0000                                                                                                                                                                                                                                                       
[... 14 lines cut ...]
                0xbe1d0000                . = ALIGN (0x10000)                                                                                                                                                                                                                               
                0xbe1d0000                _runtime_heap_start = ABSOLUTE (.)                                                                                                                                                                                                                
                0xbe250000                . = (. + 0x80000)                                                                                                                                                                                                                                 
 *fill*         0xbe1d0000    0x80000                                                                                                         
                0xbe250000                _runtime_heap_end = ABSOLUTE (.)                                                                                                                                         [... cut ...]
                0xbe1d0000                _module_heap = _runtime_heap_start                                                                                                                                                                                                                
                0xbe250000                _buffer_heap = _buffer_heap_start                                                                                                                                                                                                                 
                [0xbe2e0000]                _buffer_heap_end = _buffer_heap_end 
```
FYI @johnylin76 @bzhg

cujomalainey, 2021-09-22:
I think I answered my own issue, it appears our tasks are full blown threads with their own stack whereas I was under the impression they were more of a workqueue with deadlines. This raises some questions though. If we have a component that requires a tall stack, how do we allocate this for strictly this task? Also I was under the impression tasks are not preemtable, so why give them their own stack if we can't preempt them?

keyonjie, 2021-09-23:
@cujomalainey I think we allocate separated stack for each EDF task only, for those pipeline tasks which usually are LL task, they share the CORE specific stack.

paulstelian97, 2021-09-23:
> @cujomalainey I think we allocate separated stack for each EDF task only, for those pipeline tasks which usually are LL task, they share the CORE specific stack.

To add more info, at least on i.MX, LL tasks usually run in the context of the DMA IRQ handler (as we have DMA based scheduling; I'd guess you'd use the timer IRQ handler for the timer based scheduling). And yeah, IRQ handlers don't have their own stack (as far as I know)

lgirdwood, 2021-09-23:
@cujomalainey we can allocate individual stacks (with variable sizes) when we use the Zephyr RTOS for any task LL or preemptive.

cujomalainey, 2021-09-23:
Understood, I think I have a better picture now, thanks.

closed 2021-09-23 by cujomalainey