thesofproject/sof #4796: [FEATURE] Reject PCM_PARAMS IPC when in ACTIVE or PAUSED state

lyakh, 2021-09-23:
**Is your feature request related to a problem? Please describe.**
Certain CI tests erroneously submit PCM_PARAMS IPC when the pipeline is in ACTIVE or PAUSED state, the firmware accepts and tries to process it.

**Describe the solution you'd like**
Such IPCs should return an error to the host.

plbossart, 2021-09-23:
isn't it a bug rather?

lyakh, 2021-09-23:
> isn't it a bug rather?

@plbossart I have been creating too many bugs recently, so I decided to make this a feature :-) It doesn't seem to create any problems at the moment.

plbossart, 2021-09-23:
Man, I found so many bugs lately, if I started called them features there would be several pages of enhancements!

lgirdwood, 2021-09-24:
@lyakh thinking the easiest approach here is to ignore the request for compatibility with Windows. i.e. just return success and report this as a warning in the logs.

lyakh, 2022-10-26:
fixed by #4869

closed 2022-10-26 by lyakh