thesofproject/sof #4809: [BUG] Input from microphone gets choppy

brianjmurrell, 2021-09-27:
**Describe the bug**
At some point after an unknown amount of up-time, input from a headset microphone becomes choppy.  This doesn't happen immediately after a reboot but only some (unknown) time subsequently.

**To Reproduce**
Unknown

**Reproduction Rate**
100% after an unknown amount of time since boot

**Expected behavior**
Input should not be choppy

**Impact**
Makes use of the headset microphone impossible

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Fedora 33 - 5.13.16-100.fc33.x86_64
    * SOF: Unknown
2) Name of the topology file
    * Topology: Not sure what this is
3) Name of the platform(s) on which the bug is observed.
    * Platform: x86_64?

`alsa-info` report:
```
upload=true&script=true&cardinfo=
!!################################
!!ALSA Information Script v 0.5.0
!!################################

!!Script ran on: Mon Sep 27 14:56:16 UTC 2021

!!Linux Distribution
!!------------------

Fedora release 33 (Thirty Three) NAME=Fedora ID=fedora PRETTY_NAME="Fedora 33 (Workstation Edition)" LOGO=fedora-logo-icon CPE_NAME="cpe:/o:fedoraproject:fedora:33" HOME_URL="https://fedoraproject.org/" DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f33/system-administrators-guide/" SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help" BUG_REPORT_URL="https://bugzilla.redhat.com/" REDHAT_BUGZILLA_PRODUCT="Fedora" REDHAT_BUGZILLA_PRODUCT_VERSION=33 REDHAT_SUPPORT_PRODUCT="Fedora" REDHAT_SUPPORT_PRODUCT_VERSION=33 PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy" Fedora release 33 (Thirty Three) Fedora release 33 (Thirty Three)

!!DMI Information
!!---------------

Manufacturer:      HP
Product Name:      HP ZBook Fury 15 G7 Mobile Workstation
Product Version:   
Firmware Version:  S92 Ver. 01.04.01
System SKU:        2V8E5UP#ABA
Board Vendor:      HP
Board Name:        8783

!!ACPI Device Status Information
!!---------------

/sys/bus/acpi/devices/ACPI000C:00/status 	 15
/sys/bus/acpi/devices/ELAN0720:00/status 	 15
/sys/bus/acpi/devices/HPIC000C:00/status 	 15
/sys/bus/acpi/devices/HPQ6001:00/status 	 15
/sys/bus/acpi/devices/HPQ6007:00/status 	 15
/sys/bus/acpi/devices/IFX0785:00/status 	 15
/sys/bus/acpi/devices/INT3400:00/status 	 15
/sys/bus/acpi/devices/INT3403:00/status 	 15
/sys/bus/acpi/devices/INT3403:01/status 	 15
/sys/bus/acpi/devices/INT3403:02/status 	 15
/sys/bus/acpi/devices/INT3403:03/status 	 15
/sys/bus/acpi/devices/INT3403:04/status 	 15
/sys/bus/acpi/devices/INT340E:00/status 	 15
/sys/bus/acpi/devices/INT3450:00/status 	 15
/sys/bus/acpi/devices/INT3F0D:00/status 	 15
/sys/bus/acpi/devices/LNXPOWER:01/status 	 1
/sys/bus/acpi/devices/LNXPOWER:02/status 	 1
/sys/bus/acpi/devices/LNXPOWER:03/status 	 15
[... 2751 lines cut ...]
[68227.087678] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[68227.087680] snd_hda_codec_realtek ehdaudio0D0: ASoC: error at snd_soc_dai_startup on Digital Codec DAI: -22
[68227.087682]  Digital Playback and Capture: soc_pcm_open() failed (-22)
[68227.087683]  HDA Digital: ASoC: dpcm_be_dai_startup() failed at Digital Playback and Capture (-22)
[68227.087684]  HDA Digital: dpcm_fe_dai_startup() failed (-22)
[68228.652120] usb usb3: root hub lost power or was reset
[68228.652128] usb usb4: root hub lost power or was reset
[68228.661707] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[68228.661710] snd_hda_codec_realtek ehdaudio0D0: ASoC: error at snd_soc_dai_startup on Digital Codec DAI: -22
[68228.661712]  Digital Playback and Capture: soc_pcm_open() failed (-22)
[68228.661714]  HDA Digital: ASoC: dpcm_be_dai_startup() failed at Digital Playback and Capture (-22)
[68228.661715]  HDA Digital: dpcm_fe_dai_startup() failed (-22)
[68293.993652] usb usb3: root hub lost power or was reset

!!Packages installed
!!--------------------

alsa-lib-1.2.4-5.fc33.x86_64
alsa-utils-1.2.4-2.fc33.x86_64
alsa-plugins-pulseaudio-1.2.2-4.fc33.x86_64
alsa-sof-firmware-1.6.1-4.fc33.noarch
alsa-tools-firmware-1.2.2-3.fc33.x86_64
```

I don't really know much of anything about SOF.  I'm really just an end-user of it.  As such, I am quite sure there is not enough information here, but I don't know what else to add that would be relevant.  I am of course happy to provide whatever additional information is necessary.

plbossart, 2021-09-27:
@brianjmurrell the headset is not directly controlled by the Intel PCH but by the HDaudio codec.

Can you try using the legacy driver by adding  “options snd-intel-dspcfg dsp_driver=1” to /etc/modprobe.d/alsa-base.conf.

This will prevent the internal microphones from working, but at least we should be able to see if the headset is functional without SOF.

lgirdwood, 2021-09-27:
@plbossart hmm, this is CML capture related, so could benefit from the stream offset fixes done by @ranj063 in v1.9.1
@brianjmurrell v1.9.1 is a few weeks away, but could solve this problem.

plbossart, 2021-09-27:
Also @brianjmurrell is there any reason why you have both PipeWire and PulseAudio? Usually it's one or the other.

````
PipeWire:
      Installed - Yes (/usr/bin/pipewire)
      Running - Yes

Pulseaudio:
      Installed - Yes (/usr/bin/pulseaudio)
      Running - Yes
````

plbossart, 2021-09-27:
> @plbossart hmm, this is CML capture related, so could benefit from the stream offset fixes done by @ranj063 in v1.9.1
> @brianjmurrell v1.9.1 is a few weeks away, but could solve this problem.

I don't think so, the changes are only to stop the DMA and avoid a timeout, and you would see a big issue requiring a reboot.

brianjmurrell, 2021-09-27:
> Can you try using the legacy driver by adding “options snd-intel-dspcfg dsp_driver=1” to /etc/modprobe.d/alsa-base.conf.

Wow!  The stack of `snd_*` kernel modules has gotten huge while I was not looking.  Any hints/tricks on how to easily `rmmod snd_intel_dspcfg` and it's entire stack of dependencies?  I'd rather not have to completely reboot.  It would be pretty disruptive to my other ongoing work.

> Also @brianjmurrell is there any reason why you have both PipeWire and PulseAudio? Usually it's one or the other.

Just because pulseaudio and bluetooth simply don't work without having to jump through hoops.  Pipewire was reported to fix it, but my tests there reveal it has other issues.  So they are just there to be able to switch back and forth while debugging one or the other.

brianjmurrell, 2021-09-27:
> Wow! The stack of snd_* kernel modules has gotten huge while I was not looking. Any hints/tricks on how to easily rmmod snd_intel_dspcfg and it's entire stack of dependencies? I'd rather not have to completely reboot. It would be pretty disruptive to my other ongoing work.

Damn!  Trying to remove some `snd_*` module (I don't recall which one) hung my system.  The mouse pointer was still active but everything else was locked up hard.  Couldn't even ssh into the thing even though ping still works.  :frowning:

brianjmurrell, 2021-09-27:
So, given that I had to reboot, I added the module parameter.

So far recording from "Built-in Audio" seems good.  But SOF sounded fine just after a reboot.  Time will tell.

plbossart, 2021-09-27:
> Damn! Trying to remove some `snd_*` module (I don't recall which one) hung my system. The mouse pointer was still active but everything else was locked up hard. Couldn't even ssh into the thing even [...]

we have scripts in https://github.com/thesofproject/sof-test/tree/main/tools/kmod to remove/insert modules, but they may not always work, e.g. if userspace keeps a reference on a stream, etc.

brianjmurrell, 2021-10-01:
> Can you try using the legacy driver by adding “options snd-intel-dspcfg dsp_driver=1” to /etc/modprobe.d/alsa-base.conf.

This does provide for stable headset input.

So what next?

plbossart, 2021-10-01:
@brianjmurrell thanks for the experiments. The good news is that there is no issue with the hardware codec management. That's beyond the scope of our SOF driver, and it is difficult for us to check since it's really very hardware specific.

if you have spare cycles and able to help us debug further, re-enabling SOF without the "options snd-intel-dspcfg dsp_driver=1” but with the latest firmware would be nice.

You have fedora33, not sure if they have picked the new firmware just yet, if not you can install it yourself. The firmware can be found here: https://github.com/thesofproject/sof-bin and you can follow the instructions with just 1.9 instead of 1.7.

brianjmurrell, 2021-10-01:
RPM says that the current _sof_ firmware files come from alsa-sof-firmware-1.6.1-4.fc33.noarch and [the specfile](https://src.fedoraproject.org/rpms/alsa-sof-firmware/blob/f33/f/alsa-sof-firmware.spec) says they are from: [sof-bin commit f31c9ce](https://github.com/thesofproject/sof-bin/archive/f31c9ce30ac2792ec39e19da489aee7631e00654/sof-bin-f31c9ce.tar.gz) for whatever that is worth.

plbossart, 2021-10-01:
yes, 1.9 was released last week so it's not a surprise that there's a delay. 1.6 is relatively old, we've done a lot of work since, so it's really advised to try 1.9.

brianjmurrell, 2021-10-01:
> we have scripts in https://github.com/thesofproject/sof-test/tree/main/tools/kmod to remove/insert modules, but they may not always work, e.g. if userspace keeps a reference on a stream, etc.

Pretty useful.  I still ended up with:

```
$ lsmod | grep snd
snd_hda_codec_realtek   155648  1
snd_hda_codec_generic    98304  1 snd_hda_codec_realtek
ledtrig_audio          16384  1 snd_hda_codec_generic
snd_hda_codec_hdmi     73728  2
snd_usb_audio         335872  1
snd_hda_intel          57344  2
snd_usbmidi_lib        45056  1 snd_usb_audio
snd_intel_dspcfg       28672  1 snd_hda_intel
snd_rawmidi            45056  1 snd_usbmidi_lib
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg
snd_hda_codec         172032  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_hda_core          110592  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
snd_hwdep              16384  2 snd_usb_audio,snd_hda_codec
snd_seq_device         16384  1 snd_rawmidi
snd_pcm               135168  5 snd_hda_codec_hdmi,snd_hda_intel,snd_usb_audio,snd_hda_codec,snd_hda_core
mc                     65536  5 videodev,snd_usb_audio,videobuf2_v4l2,uvcvideo,videobuf2_common
snd_timer              49152  1 snd_pcm
snd                   110592  18 snd_hda_codec_generic,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_usb_audio,snd_usbmidi_lib,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm,snd_rawmidi
soundcore              16384  1 snd
```

I did stop `pulseaudio.{socket,service}` but I also had to kill an `alsactl` process.  After that:

```
$ sudo rmmod snd_usb_audio snd_hda_intel snd_usbmidi_lib snd_intel_dspcfg snd_rawmidi snd_intel_sdw_acpisnd_intel_sdw_acpi snd_hda_codec snd_hda_core snd_hwdep snd_seq_device snd_pcm snd_timer snd
```

caused:

```
[353469.054160] BUG: unable to handle page fault for address: 000000000000173c
[353469.054172] #PF: supervisor read access in kernel mode
[353469.054177] #PF: error_code(0x0000) - not-present page
[353469.054183] PGD 0 P4D 0 
[353469.054190] Oops: 0000 [#1] SMP NOPTI
[353469.054198] CPU: 2 PID: 2906589 Comm: kworker/2:3 Not tainted 5.13.16-100.fc33.x86_64 #1
[353469.054206] Hardware name: HP HP ZBook Fury 15 G7 Mobile Workstation/8783, BIOS S92 Ver. 01.04.01 01/05/2021
[353469.054211] Workqueue: events set_brightness_delayed
[353469.054222] RIP: 0010:micmute_led_set+0x14/0x80 [snd_hda_codec_realtek]
[353469.054253] Code: 85 f6 41 0f 95 c0 eb b8 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 0f 1f 44 00 00 48 8b 47 50 48 8b 78 40 48 8b 97 58 04 00 00 <8b> 8a 3c 17 00 00 8b 82 48 17 00 00 85 c9 75 46 85 f6 41 0f 94 c0
[353469.054261] RSP: 0018:ffffae654410fe70 EFLAGS: 00010286
[353469.054267] RAX: ffffa010c9ce0400 RBX: ffffa010ccdf26e8 RCX: ffffa01fff4aa620
[353469.054272] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffffa010c8317800
[353469.054277] RBP: ffffa010ccdf2628 R08: ffffa01fff4aa620 R09: ffffa0179245072c
[353469.054280] R10: 0000000000000007 R11: 0000000000000002 R12: ffffa01fff4aa600
[353469.054285] R13: ffffa01fff4b0700 R14: 0000000000000000 R15: 0000000000000001
[353469.054290] FS:  0000000000000000(0000) GS:ffffa01fff480000(0000) knlGS:0000000000000000
[353469.054296] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[353469.054301] CR2: 000000000000173c CR3: 000000069c960002 CR4: 00000000007706e0
[353469.054306] PKRU: 55555554
[353469.054309] Call Trace:
[353469.054315]  set_brightness_delayed+0x71/0xb0
[353469.054325]  process_one_work+0x1e9/0x380
[353469.054336]  worker_thread+0x53/0x3e0
[353469.054344]  ? process_one_work+0x380/0x380
[353469.054352]  kthread+0x124/0x150
[353469.054359]  ? set_kthread_struct+0x40/0x40
[353469.054366]  ret_from_fork+0x1f/0x30
[353469.054380] Modules linked in: sctp ip6_udp_tunnel udp_tunnel nfnetlink_queue nfnetlink_log r8153_ecm cdc_ether usbnet ib_core xt_addrtype xt_REDIRECT xt_hl rfcomm xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_nat_tftp nf_conntrack_tftp tun bridge stp llc bonding tls nft_objref nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat nf_tables ebtable_nat ebtable_broute ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat nf_conntrack nouveau nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_raw iptable_security mxm_wmi drm_ttm_helper ttm ip_set nfnetlink ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter bnep iTCO_wdt intel_pmc_bxt ee1004 iTCO_vendor_support mei_wdt mei_hdcp intel_rapl_msr ucsi_ccg intel_tcc_cooling x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass rapl intel_cstate vfat fat
[353469.054503]  intel_uncore snd_hda_codec_realtek hp_wmi platform_profile snd_hda_codec_generic iwlmvm intel_wmi_thunderbolt ledtrig_audio sparse_keymap snd_hda_codec_hdmi wmi_bmof ac97_bus mac80211 snd_hda_intel(-) snd_usbmidi_lib snd_intel_dspcfg snd_rawmidi i2c_i801 snd_intel_sdw_acpi i2c_smbus snd_hda_codec libarc4 uvcvideo videobuf2_vmalloc thunderbolt iwlwifi snd_hda_core btusb videobuf2_memops btrtl snd_hwdep mei_me videobuf2_v4l2 btbcm videobuf2_common btintel mei snd_seq_device idma64 bluetooth cfg80211 videodev snd_pcm hid_sensor_als hid_sensor_trigger hid_sensor_iio_common ecdh_generic industrialio_triggered_buffer joydev mc snd_timer rfkill kfifo_buf processor_thermal_device processor_thermal_rfim ecc industrialio snd processor_thermal_mbox p [... cut ...]
[353469.054735] R10: 0000000000000007 R11: 0000000000000002 R12: ffffa01fff4aa600
[353469.054739] R13: ffffa01fff4b0700 R14: 0000000000000000 R15: 0000000000000001
[353469.054743] FS:  0000000000000000(0000) GS:ffffa01fff480000(0000) knlGS:0000000000000000
[353469.054748] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[353469.054752] CR2: 000000000000173c CR3: 000000069c960002 CR4: 00000000007706e0
[353469.054757] PKRU: 55555554

```
:-(  At least I commented out the `options snd-intel-dspcfg dsp_driver=1` so when I reboot I should get back to using _sof_.

brianjmurrell, 2021-10-01:
> yes, 1.9 was released last week so it's not a surprise that there's a delay. 1.6 is relatively old, we've done a lot of work since, so it's really advised to try 1.9.

Since 1.8 is easily/readily available from F34, going to try that first.  But will roll a 1.9 RPM if that still doesn't work.

brianjmurrell, 2021-10-01:
Hrm.  Didn't expect it to be this bad.  But I now only have a _TU106 High Definition Audio Controller_ in `pavucontrol` now.

```
$ $ lsmod | grep snd
snd_ctl_led            24576  0
snd_soc_skl_hda_dsp    24576  6
snd_soc_intel_hda_dsp_common    20480  1 snd_soc_skl_hda_dsp
snd_soc_hdac_hdmi      45056  1 snd_soc_skl_hda_dsp
snd_soc_dmic           16384  1
snd_hda_codec_realtek   155648  1
snd_hda_codec_generic    98304  1 snd_hda_codec_realtek
snd_sof_pci_intel_cnl    16384  5
snd_sof_intel_hda_common   110592  1 snd_sof_pci_intel_cnl
soundwire_intel        45056  1 snd_sof_intel_hda_common
snd_sof_intel_hda      20480  1 snd_sof_intel_hda_common
snd_sof_pci            20480  2 snd_sof_intel_hda_common,snd_sof_pci_intel_cnl
snd_sof_xtensa_dsp     16384  1 snd_sof_intel_hda_common
snd_sof               147456  2 snd_sof_pci,snd_sof_intel_hda_common
ledtrig_audio          16384  3 snd_ctl_led,snd_hda_codec_generic,snd_sof
snd_soc_skl           180224  0
snd_soc_hdac_hda       24576  2 snd_sof_intel_hda_common,snd_soc_skl
snd_hda_ext_core       36864  5 snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_soc_skl,snd_sof_intel_hda
snd_soc_sst_ipc        20480  1 snd_soc_skl
snd_soc_sst_dsp        36864  1 snd_soc_skl
snd_soc_acpi_intel_match    49152  3 snd_sof_intel_hda_common,snd_soc_skl,snd_sof_pci_intel_cnl
snd_soc_acpi           16384  3 snd_soc_acpi_intel_match,snd_sof_intel_hda_common,snd_soc_skl
snd_soc_core          331776  8 soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_soc_skl,snd_soc_dmic,snd_soc_skl_hda_dsp
snd_hda_codec_hdmi     73728  2
snd_usb_audio         335872  1
snd_compress           32768  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
ac97_bus               16384  1 snd_soc_core
snd_hda_intel          57344  5
snd_usbmidi_lib        45056  1 snd_usb_audio
snd_intel_dspcfg       28672  3 snd_hda_intel,snd_sof_intel_hda_common,snd_soc_skl
snd_intel_sdw_acpi     20480  2 snd_sof_intel_hda_common,snd_intel_dspcfg
snd_rawmidi            45056  1 snd_usbmidi_lib
snd_hda_codec         172032  7 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek,snd_soc_intel_hda_dsp_common,snd_soc_hdac_hda,snd_soc_skl_hda_dsp
snd_hda_core          110592  12 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_hda_codec_realtek,snd_soc_intel_hda_dsp_common,snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_soc_skl,snd_sof_intel_hda
snd_hwdep              16384  2 snd_usb_audio,snd_hda_codec
snd_seq                90112  0
snd_seq_device         16384  2 snd_seq,snd_rawmidi
snd_pcm               135168  17 snd_hda_codec_hdmi,snd_hda_intel,snd_usb_audio,snd_hda_codec,soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_compress,snd_soc_core,snd_soc_skl,snd_hda_core,snd_pcm_dmaengine
snd_timer              49152  2 snd_seq,snd_pcm
mc                     65536  5 videodev,snd_usb_audio,videobuf2_v4l2,uvcvideo,videobuf2_common
snd                   110592  38 snd_ctl_led,snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_usb_audio,snd_usbmidi_lib,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_soc_hdac_hdmi,snd_compress,snd_soc_core,snd_pcm,snd_soc_skl_hda_dsp,snd_rawmidi
soundcore              16384  2 snd_ctl_led,snd
$ dmesg | grep snd
[   21.496210] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   21.496218] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   21.496281] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[   21.496332] snd_hda_intel 0000:01:00.1: Disabling MSI
[   21.496335] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[   21.590446] snd_soc_skl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   21.590464] snd_soc_skl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   22.010637] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC285: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[   22.010641] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   22.010642] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[   22.010644] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[   22.010645] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[   22.010646] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[   22.083137] usbcore: registered new interface driver snd-usb-audio
[   22.160470] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[   22.160475] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[   22.167548] snd_hda_codec_hdmi ehdaudio0D2: Monitor plugged-in, Failed to power up codec ret=[-13]
[   84.507038] snd_hda_intel 0000:01:00.1: bound 0000:01:00.0 (ops nv50_audio_component_bind_ops [nouveau])
```

plbossart, 2021-10-01:
could be a UCM issue as well, it's handled with the alsa-ucm-conf package.
Can you provide the list to the results of 'alsa-info' to see what the driver exposes?

brianjmurrell, 2021-10-01:
It does appear in `pavucontrol` now but any attempt to choose `Comment Lake PCH cAVS *` as an input device, for example simply doesn't actually make the choice.

[alsa-info.txt](https://github.com/thesofproject/sof/files/7269687/alsa-info.txt)

brianjmurrell, 2021-10-01:
These have appeared in `dmesg` since the above log was generated:

```
[ 2333.228987] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[ 2333.228990] snd_hda_codec_realtek ehdaudio0D0: ASoC: error at snd_soc_dai_startup on Digital Codec DAI: -22
[ 2333.228992]  Digital Playback and Capture: soc_pcm_open() failed (-22)
[ 2333.228993]  HDA Digital: ASoC: dpcm_be_dai_startup() failed at Digital Playback and Capture (-22)
[ 2333.228994]  HDA Digital: dpcm_fe_dai_startup() failed (-22)
[ 2333.230068] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[ 2333.230070] snd_hda_codec_realtek ehdaudio0D0: ASoC: error at snd_soc_dai_startup on Digital Codec DAI: -22
[ 2333.230071]  Digital Playback and Capture: soc_pcm_open() failed (-22)
[ 2333.230073]  HDA Digital: ASoC: dpcm_be_dai_startup() failed at Digital Playback and Capture (-22)
[ 2333.230074]  HDA Digital: dpcm_fe_dai_startup() failed (-22)
```

plbossart, 2021-10-01:
> It does appear in `pavucontrol` now but any attempt to choose `Comment Lake PCH cAVS *` as an input device, for example simply doesn't actually make the choice.
> 

Some of these devices are playback, the only thing you can select for input is DMIC.

> [alsa-info.txt](https://github.com/thesofproject/sof/files/7269687/alsa-info.txt)

Doesn't look so bad, we have the right filter of HDMI devices handled by the nVidia card.

The part that's not good is the HDaudio digital part. we've seen such issues on other boards, it depends on BIOS verbs and the presence of a digital output proper. we should probably remove this option or make it dependent on a run-time information to disable it if not present. @kv2019i @ranj063 FYI

The only thing that I see as possibly problematic is the 
````
snd_hda_codec_realtek ehdaudio0D0: Unable to sync register 0x2b8000. -5
````

that could point to some power management issue? You have 5.13.19 for the kernel, that's recent enough and should work.

kv2019i, 2021-10-04:
Hmm, this is an interesting config. Both the GPUs seem to be enabled, Intel and the Nvidia one. This may create issues as SOF driver does not support the vga-switcheroo (dynamic switching of multiplexed GPUs). But OTOH, there are no errors related to this in the log, so this would not seem to be the source of problems here.

The "Unable to sync" smells of the regressions that were found related to commit 215a22ed31a1
("ALSA: hda: Refactor codec PM to use direct-complete optimization"), but I checked that 5.13.19 has all the fixes.

brianjmurrell, 2021-10-04:
Latest messages from the kernel as I was seeing if it was working this morning:

```
[233030.272528] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[233030.272531] snd_hda_codec_realtek ehdaudio0D0: ASoC: error at snd_soc_dai_startup on Digital Codec DAI: -22
[233030.272533]  Digital Playback and Capture: soc_pcm_open() failed (-22)
[233030.272534]  HDA Digital: ASoC: dpcm_be_dai_startup() failed at Digital Playback and Capture (-22)
[233030.272535]  HDA Digital: dpcm_fe_dai_startup() failed (-22)
[233030.273955] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[233030.273956] snd_hda_codec_realtek ehdaudio0D0: ASoC: error at snd_soc_dai_startup on Digital Codec DAI: -22
[233030.273958]  Digital Playback and Capture: soc_pcm_open() failed (-22)
[233030.273959]  HDA Digital: ASoC: dpcm_be_dai_startup() failed at Digital Playback and Capture (-22)
[233030.273960]  HDA Digital: dpcm_fe_dai_startup() failed (-22)
```

How likely is the 1.9 (vs. the 1.8 that I have currently) firmware to fix this?

brianjmurrell, 2021-10-04:
> Both the GPUs seem to be enabled, Intel and the Nvidia one

That's not something that *I* set up.  Probably just the default state.  Currently it seems that the nouveau module is loaded but honestly, I have no idea which GPU is actually driving my display output at the moment.  I probably don't really care either.  I am not doing anything on this machine that the Nvidia GPU is likely to be better at performing than the Intel one.

I don't really know how to disable one or the other though.

brianjmurrell, 2021-10-04:
pulseaudio seems to be blocked in the kernel now:

```
[234932.386752] sysrq: Show Blocked State
[234932.387276] task:kworker/3:2     state:D stack:    0 pid:1275858 ppid:     2 flags:0x00004000
[234932.387279] Workqueue: pm pm_runtime_work
[234932.387282] Call Trace:
[234932.387284]  __schedule+0x2f0/0x990
[234932.387287]  schedule+0x46/0xb0
[234932.387288]  schedule_timeout+0x7d/0x120
[234932.387290]  ? __raw_callee_save___native_queued_spin_unlock+0x11/0x1e
[234932.387292]  ? __bpf_trace_tick_stop+0x10/0x10
[234932.387296]  snd_hdac_bus_get_response+0x16f/0x230 [snd_hda_core]
[234932.387301]  ? finish_wait+0x80/0x80
[234932.387303]  snd_hdac_bus_exec_verb_unlocked+0x4b/0x140 [snd_hda_core]
[234932.387306]  ? __mutex_unlock_slowpath.constprop.0+0xae/0xf0
[234932.387307]  codec_exec_verb+0x72/0x110 [snd_hda_codec]
[234932.387312]  snd_hda_codec_set_power_to_all+0x58/0xb0 [snd_hda_codec]
[234932.387316]  hda_set_power_state+0x9c/0x100 [snd_hda_codec]
[234932.387320]  ? hda_call_codec_suspend+0xa0/0xa0 [snd_hda_codec]
[234932.387324]  hda_call_codec_suspend+0x6f/0xa0 [snd_hda_codec]
[234932.387327]  hda_codec_runtime_suspend+0x27/0x80 [snd_hda_codec]
[234932.387331]  ? hda_call_codec_suspend+0xa0/0xa0 [snd_hda_codec]
[234932.387334]  __rpm_callback+0x7e/0x140
[234932.387335]  ? hda_call_codec_suspend+0xa0/0xa0 [snd_hda_codec]
[234932.387339]  rpm_callback+0x1f/0x70
[234932.387340]  ? hda_call_codec_suspend+0xa0/0xa0 [snd_hda_codec]
[234932.387343]  rpm_suspend+0x137/0x6c0
[234932.387344]  ? psi_task_switch+0xc5/0x240
[234932.387346]  ? __switch_to_asm+0x42/0x70
[234932.387347]  ? __switch_to+0x114/0x450
[234932.387350]  pm_runtime_work+0x8e/0x90
[234932.387351]  process_one_work+0x1e9/0x380
[234932.387353]  worker_thread+0x53/0x3e0
[234932.387354]  ? process_one_work+0x380/0x380
[234932.387355]  kthread+0x124/0x150
[234932.387356]  ? set_kthread_struct+0x40/0x40
[234932.387357]  ret_from_fork+0x1f/0x30
[234932.387373] task:pulseaudio      state:D stack:    0 pid:1285436 ppid: 10420 flags:0x00004224
[234932.387375] Call Trace:
[234932.387375]  __schedule+0x2f0/0x990
[234932.387376]  schedule+0x46/0xb0
[234932.387377]  schedule_preempt_disabled+0xa/0x10
[234932.387378]  __mutex_lock.constprop.0+0x279/0x440
[234932.387380]  codec_exec_verb+0x4f/0x110 [snd_hda_codec]
[234932.387384]  hdmi_pin_setup_infoframe+0x15c/0x230 [snd_hda_codec_hdmi]
[234932.387387]  hdmi_setup_audio_infoframe+0x145/0x170 [snd_hda_codec_hdmi]
[234932.387390]  generic_hdmi_playback_pcm_prepare+0x15b/0x260 [snd_hda_codec_hdmi]
[234932.387392]  snd_hda_codec_prepare+0x4d/0x110 [snd_hda_codec]
[234932.387396]  hdac_hda_dai_prepare+0x69/0x90 [snd_soc_hdac_hda]
[234932.387398]  snd_soc_pcm_dai_prepare+0x51/0xb0 [snd_soc_core]
[234932.387409]  ? snd_pcm_common_ioctl+0x9ae/0x1120 [snd_pcm]
[234932.387413]  soc_pcm_prepare+0x50/0x100 [snd_soc_core]
[234932.387421]  dpcm_be_dai_prepare+0xba/0x120 [snd_soc_core]
[234932.387428]  dpcm_fe_dai_prepare+0xaa/0x1d0 [snd_soc_core]
[234932.387435]  snd_pcm_prepare+0x146/0x1a0 [snd_pcm]
[234932.387439]  snd_pcm_ioctl+0x23/0x30 [snd_pcm]
[234932.387442]  __x64_sys_ioctl+0x80/0xb0
[234932.387444]  do_syscall_64+0x3d/0x80
[234932.387446]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[234932.387448] RIP: 0033:0x7f981fecff7b
[234932.387450] RSP: 002b:00007fffc28eaed8 EFLAGS: 00000202 ORIG_RAX: 0000000000000010
[234932.387451] RAX: ffffffffffffffda RBX: 000055834f84a020 RCX: 00007f981fecff7b
[234932.387452] RDX: 0000000000000000 RSI: 0000000000004140 RDI: 000000000000002e
[234932.387453] RBP: 00007fffc28eb4c0 R08: 0000000000000000 R09: 00000000000012bd
[234932.387453] R10: 0000000000000004 R11: 0000000000000202 R12: 0000000000000000
[234932.387454] R13: 00007fffc28eb190 R14: 0000000000000000 R15: 00007fffc28eaf20
```

plbossart, 2021-10-04:
The latest dmesg still hints at codec power management issues when we use the hdac_hda codec with SOF.

kv2019i, 2021-10-04:
Hmm, this hints at the display codec getting stuck:

[234932.387380]  codec_exec_verb+0x4f/0x110 [snd_hda_codec]
[234932.387384]  hdmi_pin_setup_infoframe+0x15c/0x230 [snd_hda_codec_hdmi]

I think we are talking about multiple issues here. The "DAI Digital Codec DAI" errors you talk about earlier are completely harmless. Pipewire is opening a PCM that is not connected to any hw so you get an error. This won't cause any harm.

The resume/suspend error. I think this is related to the multiplexed GPU setup (and at @brianjmurrell , this is a feature of the laptop's design and needs additional kernel support on top of Nvidia and Intel hw drivers). The above log showing pulseaudio stuck clearly points to Intel audio driver trying to talk to Intel GPU, but latter has been switched off (and will not response).
For this, you can try following to disable audio driver for Intel GPU:
/etc/modprobe/sof-options.conf
```
# limit SOF driver to only handle the external HDA codec 
options snd_sof_intel_hda_common codec_mask=0x1
```

But then the original problem of microphone getting choppy. I don't see how this would relate to problems #1 and #2 above. That could still get solved by upgrade to SOF1.8 or SOF1.9. @brianjmurrell Could you try a newer Fedora image, maybe Fedora35 beta https://getfedora.org/workstation/download/ ...?

brianjmurrell, 2021-10-04:
> For this, you can try following to disable audio driver for Intel GPU:

I've added that for my next reboot.

> But then the original problem of microphone getting choppy

Maybe it's too early to determine sufficiently, but I didn't have choppy input audio today using SOF.  Maybe the update to 1.8 firmware fixed it.  I will report as time goes on.

brianjmurrell, 2021-10-05:
> The resume/suspend error. I think this is related to the multiplexed GPU setup (and at @brianjmurrell , this is a feature of the laptop's design and needs additional kernel support on top of Nvidia and Intel hw drivers).

Are you referring to something like https://01.org/linuxgraphics/gfx-docs/drm/gpu/vga-switcheroo.html or some yet-unknown-unimplemented support?  Is there something I should be doing to avoid this:

> points to Intel audio driver trying to talk to Intel GPU, but latter has been switched off (and will not response).

other than manually setting `snd_sof_intel_hda_common` options?

I guess I am looking for the answer to the question of what needs changing where to make this all work out of the box without the need to get help and do manual kernel module settings.  Is there something the distro should be doing differently so that this problem doesn't happen to everyone with muxed GPUs?

kv2019i, 2021-10-07:
@brianjmurrell wrote:
> Are you referring to something like https://01.org/linuxgraphics/gfx-docs/drm/gpu/vga-switcheroo.html or 
> some yet-unknown-unimplemented support? Is there something I should be doing to avoid this:

Exactly that. This would seem to require new feature development. FIled
https://github.com/thesofproject/sof/issues/4854

galibozek, 2022-01-27:
Same error on Arch - 5.16.3-arch1-1, SOF: 2.0
```
sty 27 20:08:52 kernel: snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
sty 27 20:08:52 kernel: snd_hda_codec_realtek ehdaudio0D0: ASoC: error at snd_soc_dai_startup on Digital Codec DAI: -22
sty 27 20:08:52 kernel:  Digital Playback and Capture: soc_pcm_open() failed (-22)
sty 27 20:08:52 kernel:  HDA Digital: ASoC: dpcm_be_dai_startup() failed at Digital Playback and Capture (-22)
sty 27 20:08:52 kernel:  HDA Digital: dpcm_fe_dai_startup() failed (-22)
sty 27 20:08:52 kernel: snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
sty 27 20:08:52 kernel: snd_hda_codec_realtek ehdaudio0D0: ASoC: error at snd_soc_dai_startup on Digital Codec DAI: -22
sty 27 20:08:52 kernel:  Digital Playback and Capture: soc_pcm_open() failed (-22)
sty 27 20:08:52 kernel:  HDA Digital: ASoC: dpcm_be_dai_startup() failed at Digital Playback and Capture (-22)
sty 27 20:08:52 kernel:  HDA Digital: dpcm_fe_dai_startup() failed (-22)
```

plbossart, 2022-01-27:
@galibozek this error is usually harmless, we don't actually use the digital output in any device. Do you see any real issue or just reporting those logs?

galibozek, 2022-01-30:
@plbossart I did not observe any real problem

plbossart, 2022-01-31:
thanks @galibozek. we'll need to work on this and remove this path when it's not fully enabled in the codec driver. Or alternatively remove the digital path since it's not used by anyone.

lgirdwood, 2022-01-31:
@singalsu fyi - are there any frag API users in the DMIC pipes ? I assume we can fix those all for v2.1 to give a DMIC performance boost that may help here.

singalsu, 2022-02-01:
Attach of a sine wave recording would be useful, see tone 440 Hz comment above. Or get a sine wave player app to your smartphone and use it as sound source. https://apps.apple.com/us/app/sine-wave-generator/id1118935547 https://play.google.com/store/apps/details?id=sine.wave.signal.generator&hl=en_IE&gl=US

jike212, 2022-07-06:
This is an older issue but I just began experiencing the same thing on my X1 Carbon Gen 9 (Tigerlake) and Fedora 36. It just started happening a few days ago and I'm not sure why.

My `dmesg` resembles @brianjmurrell and @galibozek's posts. Anything else I can test?

plbossart, 2022-07-06:
@jike212 please check if by any chance you had an update of the SOF firmware. There was a release last week, that's the only thing that I can think of.

lgirdwood, 2022-07-07:
@jike212 can you provide the audio driver output in your dmesg thanks.

closed 2025-07-25 by github-actions[bot]