thesofproject/sof #4813: [BUG] IPC timed out when multiple-pause-resume on TGLU-VOLT-SDW

1994lwz, 2021-09-28:
**Describe the bug**
in inner daily 6885 testcase=multiple-pause-resume-25
ipc timed out when multiple-pause-resume on TGLU-VOLT-SDW

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-sdw-max98373-rt5682.tplg ~/sof-test/test-case/multiple-pause-resume.sh -r 25
not 100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 0715e9e6
SOF Branch: main
SOF Commit: 3e260cc8de71
Topology:/lib/firmware/intel/sof-tplg/sof-tgl-sdw-max98373-rt5682.tplg
Platform: TGLU-VOLT-SDW

**Screenshots or console output**
[console]
```
=== PAUSE ===                                                            
(6/25) pcm'Speaker' cmd'aplay' id'0': Wait for 28 ms before pause

(6/25) pcm'Speaker' cmd'aplay' id'0': Wait for 23 ms before resume

(5/25) pcm'Amplifier Reference' cmd'arecord' id'1': Wait for 49 ms before resume
    error: fetch_entry() failed with: -1, aborting
arecord: do_pause:1581: pause release error: Connection timed out
                                                                          

=== PAUSE ===                                                            arecord: do_pause:1570: pause push error: File descriptor in bad state
                                                                          

(6/25) pcm'Amplifier Reference' cmd'arecord' id'1': Wait for 25 ms before resume
 aplay: do_pause:1570: pause push error: Connection timed out
                                                                          

=== PAUSE ===                                                            
(7/25) pcm'Speaker' cmd'aplay' id'0': Wait for 38 ms before resume
 aplay: do_pause:1570: pause push error: Connection timed out
```

[dmesg]
```
[dmesg.txt](https://github.com/thesofproject/sof/files/7241792/dmesg.txt)
[ 4221.638414] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[ 4221.641786] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 2 dir 0 cmd 3
[ 4221.934872] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 4221.934905] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 4222.144656] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60070000 (msg/reply size: 12/12)
[ 4222.144673] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 4222.144676] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 4222.144686] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000083 rirb 00
[ 4222.144692] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 4222.144706] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 4222.144710] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 4222.144714] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 4222.144720] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 4222.144947] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 4222.144951] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 4222.144968] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 4222.144977] kernel:  Amplifier Reference: ASoC: trigger FE cmd: 4 failed: -110
[ 4222.144987] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 3 dir 1 cmd 3
[ 4222.144997] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 4222.145016] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace IO error
[ 4222.434915] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 4222.434933] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

bkokoszx, 2021-10-05:
@1994lwz 
Is it reproduced only in inner testes?

XiaoyunWu6666, 2021-10-08:
@bkokoszx  found ipc timed out when multiple pause release on TGLU_RVP_SDW again in inner daily test 7104

keqiaozhang, 2021-11-08:
@lyakh any updates on this issue?

lgirdwood, 2021-11-12:
@1994lwz can you retest with HEAD, we have a pause/STOP update merged by @lyakh so would be interested

XiaoyunWu6666, 2022-01-24:
seen together with #4434

lgirdwood, 2022-02-15:
@mwasko fyi since Bartosz has left. Please re-assign.

lgirdwood, 2022-02-16:
@XiaoyunWu6666 what the reproduction rate here ? "it says not 100%" are we meaning > 50%

XiaoyunWu6666, 2022-02-21:
> @XiaoyunWu6666 what the reproduction rate here ? "it says not 100%" are we meaning > 50%

I would say the reproduction rate had increased a lot now , almost can be seen in every daily report together with 4434.

plbossart, 2022-02-22:
didn't @lyakh push some pipeline state changes recently in the firmware?

lgirdwood, 2022-02-22:
> didn't @lyakh push some pipeline state changes recently in the firmware?

Yep, updates have been merged for v2.1 to make everything more reproducible wrt IPC and pipeline copies. Issue is tracked for v2.1.

mengdonglin, 2022-02-28:
This issue won't be fixed for the same reason as https://github.com/thesofproject/sof/issues/4434:
When smart amplifier is present, the playback and capture (feedback) pipelines are not independent. The playback pipeline is data provider and feedback pipeline is consumer. We cannot do random pause/release on these 2 pipelines in parallel. In addition, Chrome doesn't use PAUSE/RELEASE function of SOF driver/firmware.

closed 2022-02-28 by mengdonglin