thesofproject/sof #4838: Codec Adapter: Initial setup data problem

markbartonxperi, 2021-10-04:
I'm hoping someone here will be able to advise.... I am currently using the SOF codec adapter mechanism as a way to integrate my company's technoloy for use in SOF. So far everything has gone fairly smoothly, so thanks to everyone for that!

The last issue I have, which I'm a little stuck on is passing my initial setup configuration to my codec adapter using the pipeline .m4 file that I have written. I've pasted a working version of my .m4 at the end of this message.

The problem I have is that I need to include ~900 bytes in CA_SETUP_CONTROLBYTES. Through trial and error, the 'limit' I see to hit is around 220 bytes. By contrast, I'm able to send the same 900 byte configuration 'blob' to my runtime control via ALSA (snd_ctl_elem_tlv_write) with no problems at all - all the data I expect to arrive at my codec adapter arrives sucessfully and I'm able to update the settings in my underlying software with no issue.

When I run my topology (on an UP! Xtreme dev board) with the larger, non-working, CA_SETUP_CONTROLBYTES, I see the following in dmesg output.

```
[    7.391056] sof-audio-pci-intel-cnl 0000:00:1f.3: widget PCM0P setup complete
[    7.391060] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30200000
[    7.391145] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30200000
[    7.391146] sof-audio-pci-intel-cnl 0000:00:1f.3: widget BUF1.0 setup complete
[    7.391154] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30010000
[    7.391263] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30010000
[    7.391274] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x50030000
[    7.391646] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x50030000
[    7.391648] sof-audio-pci-intel-cnl 0000:00:1f.3: widget EQIIR1.0 setup complete
[    7.391651] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30200000
[    7.391739] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30200000
[    7.391742] sof-audio-pci-intel-cnl 0000:00:1f.3: widget BUF1.1 setup complete
[    7.391766] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30010000
[    7.391864] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 116/20): -22
[    7.391869] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to load widget CODEC_ADAPTER1.0
[    7.391872] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30020000
[    7.392131] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30020000
[    7.392134] sof-audio-pci-intel-cnl 0000:00:1f.3: widget PCM0P freed
[    7.392137] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30210000
[    7.392201] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30210000
[    7.392203] sof-audio-pci-intel-cnl 0000:00:1f.3: widget BUF1.0 freed
[    7.392205] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30020000
[    7.392410] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30020000
[    7.392411] sof-audio-pci-intel-cnl 0000:00:1f.3: widget EQIIR1.0 freed
[    7.392414] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30210000
[    7.392491] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30210000
[    7.392494] sof-audio-pci-intel-cnl 0000:00:1f.3: widget BUF1.1 freed
[    7.392496] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30110000
[    7.393071] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30110000
[    7.393074] sof-audio-pci-intel-cnl 0000:00:1f.3: widget PIPELINE.1.HDA0.OUT freed
[    7.393076] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed widget list set up for pcm 0 dir 0
[    7.393080] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[    7.393085]  HDA Analog: ASoC: soc_pcm_hw_params() failed (-22)
```

Here's the contents of my m4 pipeline file that works

```
# Low Latency Passthrough with My SDK codec Pipeline and PCM
#
# Pipeline Endpoints for connection are :-
#
# host PCM_P --> B0 --> EQ 0 --> B1 --> My SDK codec --> B2 --> sink DAI0

# My SDK codec setup config
define(`CA_SETUP_CONTROLBYTES',
``      bytes "0x53,0x4f,0x46,0x00,0x00,0x00,0x00,0x00,'
`       0x20,0x00,0x00,0x00,0x01,0x20,0x01,0x03,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       0x00,0x01,0x41,0x57,0x00,0x00,0x00,0x00,'
`       0x80,0xbb,0x00,0x00,0x18,0x00,0x00,0x00,'
`       0x02,0x00,0x00,0x00,0x01,0x00,0x00,0x00,'
`       0x0c,0x00,0x00,0x00,0x00,0x00,0x00,0x00"''
)
define(`CA_SETUP_CONTROLBYTES_MAX', 8192)
define(`CA_SETUP_CONTROLBYTES_NAME', `MyCompany' `ENDPOINT_NAME' `Setup ')

define(`CA_RUNTIME_CONTROLBYTES',
``  bytes "0x53,0x4f,0x46,0x00,0x01,0x00,0x00,0x00,'
`   0x00,0x00,0x00,0x00,0x01,0x20,0x01,0x03,'
`   0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`   0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00"''
)
define(`CA_RUNTIME_CONTROLBYTES_MAX', 8192)
define(`CA_RUNTIME_CONTROLBYTES_NAME', `MyCompany' `ENDPOINT_NAME' `Runtime ')

define(`CA_SCHEDULE_CORE', 0)

DECLARE_SOF_RT_UUID("My codec", my_sdk_uuid, 0xd95fc34f, 0x370f, 0x4ac7, 0xbc, 0x86, 0xbf, 0xdc, 0x5b, 0xe2, 0x41, 0xe6);
define(`CA_UUID', my_sdk_uuid)

# Include codec adapter playback topology
include(`sof/pipe-eq-iir-codec-adapter-playback.m4')
```

The contents of CA_SETUP_CONTROLBYTES that I want to specify in the .m4 is this:

```
# My SDK codec setup config
define(`CA_SETUP_CONTROLBYTES',
``      bytes "0x53,0x4f,0x46,0x00,0x00,0x00,0x00,0x00,'
`       0xd4,0x02,0x00,0x00,0x01,0x20,0x01,0x03,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       0x00,0x01,0x41,0x57,0x00,0x00,0x00,0x00,'
`       0x80,0xbb,0x00,0x00,0x10,0x00,0x00,0x00,'
`       0x02,0x00,0x00,0x00,0x01,0x00,0x00,0x00,'
`       0xc0,0x02,0x00,0x00,0x20,0x00,0x00,0x00,'
`       0x64,0x02,0x00,0x00,0x0a,0x6c,0x4e,0x00,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       0x00,0x00,0x00,0x00,0x0a,0x6c,0x4e,0x00,'
`       0x00,0x00,0x20,0x00,0x00,0x00,0x20,0x00,'
`       0x00,0x00,0x00,0x00,0x40,0x00,0x00,0x00,'

[SNIP] Omitting ~800 bytes here for brevity

`       0x00,0x00,0x00,0x20,0x08,0x00,0x00,0x00,'
`       0x00,0x00,0x00,0x40,0x1e,0xc7,0xfe,0x12,'
`       0xbc,0x38,0x21,0x2f,0x00,0x00,0x00,0x40,'
`       0x57,0x24,0xfd,0x07,0xe2,0xfa,0xd8,0x00,'
`       0xe0,0x42,0x0a,0x39,0x53,0xd3,0x15,0x00,'
`       0x53,0xd3,0x15,0x00,0x00,0x00,0x00,0x00,'
`       0x33,0xbc,0x06,0x11,0x00,0x00,0x00,0x00,'
`       0x14,0x9e,0xca,0x7e"''
)
```

Thanks in advance,
Mark

dbaluta, 2021-10-04:
@markbartonxperi what is the output you see with sof-logger?
https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

markbartonxperi, 2021-10-04:
Nothing very useful, unfortunately.

```
gary@up:~/incoming$ sudo sof-logger -l sof-cnl.ldc -n
[sudo] password for gary: 
         TIMESTAMP              DELTA C# COMPONENT          LOCATION                      CONTENT
error: in logger_read(), fread(..., /sys/kernel/debug/sof/etrace) failed: Operation not permitted(-1)
```

When I run with the 'working' topology from my original message, and play audio content then I see all the log messages that I would expect to see, including those coming from my own codec adapter.

markbartonxperi, 2021-10-04:
Sorry - I should probably add some other more useful diagnostics about my current target system:

SOF Firmware: I'm currently building commit a938cec8b, plus my own changes to integrate my company's technology
Up! Xtreme linux kernel: I rebuilt the kernel on Friday from commit 138c1c5da4b5 in the sof/linux repository just in case there had been any upstream changes that addressed my problem (in particular I saw that there was some activity around IPC4, which felt like it might be relevant...)

On my Up! board, I get:
```
gary@up:~/incoming$ uname -r
5.15.0-rc2+
```

markbartonxperi, 2021-10-04:
@dbaluta I rebuilt the kernel for my Up board with some of the helpful sounding extra debug options enabled. I think setting `SND_SOC_SOF_DEBUG_RETAIN_DSP_CONTEXT` means that the log data is now persisting. Attaching the output from sof-logger after a restart.

Looks like the config has zero size when it reaches the firmware. I will start triple checking my data packing - I've checked this many times already, of course, but maybe I missed something and there is an error there....!

[sof-logger-output.txt](https://github.com/thesofproject/sof/files/7279009/sof-logger-output.txt)

juimonen, 2021-10-05:
@markbartonxperi can you compare the .conf topology files with the working/non-working case. So you have the m4 -> which is converted to .conf -> which alsatplg converts to .tplg binary. Or can you perhaps attach here both? Just to first check that there's nothing strange...

dbaluta, 2021-10-05:
@markbartonxperi there is definitely something wrong with the tplg file. From the log:

```
[      243892.229892] (      243892.234375) c0 dts_codec    1.2   ......../codec_adapter.c:131  ERROR load_setup_config(): no config available cfg: 9e16ede4, size: 0
[      243902.177808] (           9.947916) c0 dts_codec    1.2   ......../codec_adapter.c:74   ERROR codec_adapter_new() error -22: config loading has failed.
[      243918.792391] (          16.614582) c0 dts_codec             src/ipc/ipc3/helper.c:303  ERROR comp_new(): unable to create the new component
```

Maybe start adding some prints in comp_new() (ipc/ipc3/helper.c) and understand why ipc_config_specific.size = 0.

markbartonxperi, 2021-10-05:
@juimonen 
Hi - I've attached the .conf files for you to have a look at
Thanks!!

[sof-hda-generic-non-working.conf.txt](https://github.com/thesofproject/sof/files/7283982/sof-hda-generic-non-working.conf.txt)
[sof-hda-generic-working.conf.txt](https://github.com/thesofproject/sof/files/7283983/sof-hda-generic-working.conf.txt)

juimonen, 2021-10-05:
@markbartonxperi I could not immediately spot anything wrong in conf files... 

@dbaluta am I reading it correctly that size is never set in codec_adapter_set_params, if msg_index is something else than 0?

lgirdwood, 2021-10-05:
@singalsu can you advise how we are doing this with other components. I dont think there are any size limitations.
@markbartonxperi could you add more debug around
```
[      243892.229892] (      243892.234375) c0 dts_codec    1.2   ......../codec_adapter.c:131  ERROR load_setup_config(): no config available cfg: 9e16ede4, size: 0
[      243902.177808] (           9.947916) c0 dts_codec    1.2   ......../codec_adapter.c:74   ERROR codec_adapter_new() error -22: config loading has failed.
[      243918.792391] (          16.614582) c0 dts_codec             src/ipc/ipc3/helper.c:303  ERROR comp_new(): unable to create the new component
```
The size will be extracted from an IPC structure or IPC tuple. Btw, M4 does no macro argument validation or verification (this is fixed in topology 2) but please also check your macro argument ordering and count agianst a working version that does the same thing.

dbaluta, 2021-10-05:
> @dbaluta am I reading it correctly that size is never set in codec_adapter_set_params, if msg_index is something else than 0?

Looking at the code this looks correct.

juimonen, 2021-10-05:
@dbaluta ok I missed the "static" in the declaration of "size".

If I interpret the dmesg correctly we should be at the 1st large ipc tx. If I recall correctly for large messages we should see tx in dmesg for all parts of the chopped message -> so if the size would be 0, and we would be in first ipc, we should bail out earlier in the codec_adapter_set_params... which makes me think something overwrites the static variable in the middle?

@lgirdwood could we possibly be overwriting the "static size" with mem alloc and memset? or I'm just totally lost here..

markbartonxperi, 2021-10-05:
@dbaluta @juimonen @lgirdwood 
Evening all

I had some time this afternoon to try adding some extra logging on the firmware side.... I'll be honest and say my log message locations felt a bit random given that I don't know the codebase so well. And also don't completely understand the expected behaviour.

Here's a fragment from sof-logger with a few additional prints (note I'm using the error trace macro in favour of working out how to enable the debug trace macro!)

```
[        1781.354096] (         323.593750) c0 component             src/ipc/ipc3/helper.c:100  ERROR get_drv comp->hdr.size 116 comp->ext_data_length 16
[        1794.479095] (          13.124999) c0 component             src/ipc/ipc3/helper.c:139  ERROR get_drv(), found driver type 22, uuid dts_codec <d95fc34f-370f-4ac7-bc86-bfdc5be241e6>
[        1804.218678] (           9.739583) c0 component             src/ipc/ipc3/helper.c:296  ERROR comp new dts_codec <d95fc34f-370f-4ac7-bc86-bfdc5be241e6> type 0 id 1.2
[        1824.374928] (          20.156250) c0 dts_codec    1.2   ......../codec_adapter.c:131  ERROR load_setup_config(): no config available cfg: 9e16ede4, size: 0
[        1834.374927] (          10.000000) c0 dts_codec    1.2   ......../codec_adapter.c:74   ERROR codec_adapter_new() error -22: config loading has failed.
[        1851.562426] (          17.187500) c0 dts_codec             src/ipc/ipc3/helper.c:304  ERROR comp_new(): unable to create the new component
[        1861.406176] (           9.843750) c0 ipc                   src/ipc/ipc3/helper.c:638  ERROR ipc_comp_new(): component cd = NULL
[        1871.510342] (          10.104166) c0 ipc                  src/ipc/ipc3/handler.c:1198 ERROR ipc: pipe 1 comp 2 creation failed -22
```

Once I got to this point I noticed that the size logged from src/ipc/ipc3/helper.c:100 (116 bytes) matched the log message from the kernel that the failed IPC tx size was also 116 bytes, which made me wonder if actually my data is being read from the topology incorrectly.... so I switched my efforts over to the kernel side of things.

In my latest run I see this in dmesg:

```
[    3.701261] sof-audio-pci-intel-cnl 0000:00:1f.3: sof_widget_load_pcm
[    3.701269] sof-audio-pci-intel-cnl 0000:00:1f.3: loaded host PCM0P
[    3.701271] sof-audio-pci-intel-cnl 0000:00:1f.3:  config: periods snk 2 src 0 fmt 0
[    3.701275] sof-audio-pci-intel-cnl 0000:00:1f.3: tplg: load control type 2 name : DTS Headphones Setup 1
[    3.701278] sof-audio-pci-intel-cnl 0000:00:1f.3: tplg: load kcontrol index 2 chans 0
[    3.701280] sof-audio-pci-intel-cnl 0000:00:1f.3: tplg: load control type 2 name : DTS Headphones Runtime 1
[    3.701282] sof-audio-pci-intel-cnl 0000:00:1f.3: tplg: load kcontrol index 2 chans 0
[    3.701284] sof-audio-pci-intel-cnl 0000:00:1f.3: tplg: ready widget id 2 pipe 1 type 33 name : CODEC_ADAPTER1.0 stream none
[    3.701287] sof-audio-pci-intel-cnl 0000:00:1f.3: sof_process_load ipc_data_size 724
```

..which suggests my configuration is being loaded (total size is 756 including ABI header, so 724 bytes when that's subtracted)

Later on in dmesg I see the same as before:

```
[    7.273287] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30010000
[    7.273419] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 116/20): -22
[    7.273427] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to load widget CODEC_ADAPTER1.0
```

I will keep digging, but would definitely welcome any pointers or corrections :) In particular, is it expected that my 724 bytes should later be logged with a message size of 116 bytes? I feel like the 116 bytes is just the size of a header, and that there's probably some other struct member I should query for the payload size....

Thanks
Mark

juimonen, 2021-10-06:
@markbartonxperi the codec_adapter doesn't behave nicely if the control data size is over MAX_IPC_MSG_SIZE=384...

long story: driver only supports "large" messages with control data. When the widget is loaded -> data is dug out from the related config and added to the widget -> but if the total size is > 384 the control data (initial config) is discarded, as it can't be partly added.

Anyway, you should get the data little bit later when the control is loaded (for e.g. when streaming starts). The component just needs to handle the situation, that is doesn't get the init data in comp_new, but later. This is how for e.g. the eq works.
comp_init_data_blob function is just happy with 0 data and doesn't error out.

So, codec_adapter should not bail out with "0" init data, it needs to be changed on the firmware code. @mrajwa FYI.

juimonen, 2021-10-06:
@markbartonxperi so dsp is started in 2 steps. When the device boots -> we load the topology, but do basically nothing. If there's any gap before actual playback, dsp goes to power save/suspend, and basically shuts down. When you start actual streaming -> all existing components from kernel side are "reloaded". So what is in controls gets just pushed after widgets and before streaming starts. So in a way the "initial" initialization in the topology loading is almost unnecessary. Maybe you could do some checking there, but in the usual case the whole thing is anyway reloaded when streaming starts.

The runtime/config 2 control setting is a little bit unknown to me, but you could just try to add the line:

 diff --git a/src/audio/codec_adapter/codec_adapter.c b/src/audio/codec_adapter/codec_adapter.c
index 9891eb9e2..5eb031f5d 100644
--- a/src/audio/codec_adapter/codec_adapter.c
+++ b/src/audio/codec_adapter/codec_adapter.c
@@ -127,7 +127,10 @@ int load_setup_config(struct comp_dev *dev, void *cfg, uint32_t size)
                comp_err(dev, "load_setup_config(): no component device.");
                ret = -EINVAL;
                goto end;
-       } else if (!cfg || !size) {
+       } else if (!size) {
+               ret = 0;
+               goto end;
+       } else if (!cfg) {
                comp_err(dev, "load_setup_config(): no config available cfg: %x, size: %d",
                         (uintptr_t)cfg, size);
                ret = -EINVAL;

and see what happens...

juimonen, 2021-10-06:
@markbartonxperi nice! I'll have a meeting tomorrow with the original author of the codec adapter @mrajwa and we'll discuss best way to add this.

markbartonxperi, 2021-10-07:
@juimonen Thanks for the updates. Thinking about this selfishly(!)... If the decision ends up being that the codec_adapter should continue to fail when config size == 0, what other options are available to me for passing my initial set-up data to my code?

Thanks
Mark

juimonen, 2021-10-07:
@markbartonxperi somehow we need to fix in codec adapter, if you want to set these "big" parameters in the topology. It doesn't help here to move them around between runtime and setup controls. 1 way to set these is put some smaller parameters in these controls in topology and send the big parameters through alsa control interface in user space.

I think the google hotword works like that. So the big params are in alsa UCM user space config -> when you start streaming, particular use case in UCM is loaded -> it loads the big blob through alsa interface before streaming.

So parameters are not used in the dsp loading, but set separately. so not defined in topology. OTH this sets up dependency for user space to load those -> but in chrome that is existing mechanism (alsa UCM).

But let me still have couple of discussions with the folks how to solve this. We'll get the params there one way or another :)

juimonen, 2021-10-07:
@markbartonxperi so to mimic the "UCM way" in testing is to setup the component with some smaller parameters. In your case they could be just bogus or whatever. Then before you start to aplay you set the big params with sof-ctl (found in sof codebase). If you set them once they will be there "forever". We use this method with eq all the time.

The unfortunate thing here is that we support big params currently only with kcontrols -> and in initial topology loading the kcontrol values are not set. When component is initially loaded we use a hack where the config is dug out of the kcontrol associated with it -> but this has an issue that the component loading doesn't support big params. 

Why this 0-size change above works is because when the dsp wakes up from power save (when you start to stream) it will restore all components and all kcontrols (and send the values to match the state before power save). These kcontrol values are not send in initial loading as the dsp is not in a state to receive these controls (it is still loading the topology).

If we would not go to power save, this thing would break for sure. 99% we are going for power save, but to really make sure your parameters are set -> it needs to be set explicitly from user space.

markbartonxperi, 2021-10-07:
@juimonen Thanks for the reply - all understood!

I'll leave things in your hands, then. As before, feel free to let me know here if there's anything you want me to test.

Thanks again!

markbartonxperi, 2021-10-13:
@lgirdwood Hi - welcome back - hope you had a good break!

I made a little bit of progress with UCM - mostly getting to grips with the `alsaucm` command as well as the format of the `.conf` files. There was also a little work required on my tooling-side to generate suitable binary files. At that point I hit a bit of roadblock as I managed to run `alsaucm` in some way that killed the audio output of my Up Xtreme....

So along with recovering the UP board, and a pre-PR refactor of our code, I haven't quite got to the point where I've successfully applied settings via UCM.... however, I feel confident that this approach will work for us in the shorter term. Longer term, it sounds like topology2 will be the way to go.

lgirdwood, 2021-12-01:
@markbartonxperi all good now, can we close ?

markbartonxperi, 2021-12-02:
@lgirdwood Afternoon. Yes - all good now! Thanks for your help!

closed 2021-12-02 by markbartonxperi