thesofproject/sof #4859: [BUG] IPC timed out when multiple-pause-resume on  BDW_WSB_RT286

XiaoyunWu6666, 2021-10-09:
**Describe the bug**
in inner daily 7241 testcase=multiple-pause-resume-25
ipc timed out when multiple-pause-resume on BDW_WSB_RT286

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg ~/sof-test/test-case/multiple-pause-resume.sh -r 25

**Environment**
Start Time: 2021-10-08 21:20:17 UTC
End Time: 2021-10-08 23:58:28 UTC
Kernel Branch: topic/sof-dev
Kernel Commit: bd1219a2
SOF Branch: main
SOF Commit: b039018d0adc
**Screenshots or console output**

[dmesg]
```
[ 1375.015212] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: stream dir 1, posn mailbox offset is 649804
[ 1375.030744] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: open stream 1 dir 0
[ 1375.030751] kernel: sof-audio-acpi-intel-bdw INT3438:00: period min 192 max 16384 bytes
[ 1375.030755] kernel: sof-audio-acpi-intel-bdw INT3438:00: period count 2 max 16
[ 1375.030758] kernel: sof-audio-acpi-intel-bdw INT3438:00: buffer max 65536 bytes
[ 1375.031120] kernel: sof-audio-acpi-intel-bdw INT3438:00: rate_min: 48000 rate_max: 48000
[ 1375.031120] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: trigger stream 0 dir 1 cmd 1
[ 1375.031129] kernel: sof-audio-acpi-intel-bdw INT3438:00: channels_min: 2 channels_max: 2
[ 1375.031142] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1375.031329] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1375.033325] kernel: sof-audio-acpi-intel-bdw INT3438:00: DAI config 0 matches pcm hw params
[ 1375.033332] kernel: sof-audio-acpi-intel-bdw INT3438:00: DAI config 0 matches pcm hw params
[ 1375.033337] kernel: sof-audio-acpi-intel-bdw INT3438:00: rate_min: 48000 rate_max: 48000
[ 1375.033343] kernel: sof-audio-acpi-intel-bdw INT3438:00: channels_min: 2 channels_max: 2
[ 1375.033840] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: hw params stream 1 dir 0
[ 1375.033847] kernel: sof-audio-acpi-intel-bdw INT3438:00: generating page table for 00000000cdf0e9a7 size 0x3e80 pages 4
[ 1375.033865] kernel: sof-audio-acpi-intel-bdw INT3438:00: stream_tag 0
[ 1375.033911] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1375.034448] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1375.034456] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: stream dir 0, posn mailbox offset is 649728
[ 1375.035725] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: trigger stream 1 dir 0 cmd 1
[ 1375.035736] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1375.537460] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx timed out for 0x60040000 (msg/reply size: 12/12)
[ 1375.537491] kernel: sof-audio-acpi-intel-bdw INT3438:00: preventing DSP entering D3 state to preserve context
[ 1375.537501] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump start ]------------
[ 1375.537511] kernel: sof-audio-acpi-intel-bdw INT3438:00: fw_state: SOF_FW_BOOT_COMPLETE (5)
[ 1375.537604] kernel: sof-audio-acpi-intel-bdw INT3438:00: unexpected fault 0x00000000 trace 0x80000000
[ 1375.537626] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc host -> DSP: pending yes complete no raw 0x80000000
[ 1375.537645] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask host: pending no complete no raw 0x7fff0000
[ 1375.537663] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc DSP -> host: pending no complete no raw 0x00000000
[ 1375.537682] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask DSP: pending no complete no raw 0x7f9e0000
[ 1375.537701] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump end ]------------
[ 1375.537733] kernel: sof-audio-acpi-intel-bdw INT3438:00: ASoC: error at soc_component_trigger on INT3438:00: -110
[ 1375.537754] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: trace IO error
[ 1375.537758] kernel:  Media Playback 1: ASoC: trigger FE cmd: 1 failed: -110
[ 1375.537777] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: trigger stream 1 dir 0 cmd 0
[ 1375.537798] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1375.538809] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: trigger stream 0 dir 1 cmd 0
[ 1376.042350] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[ 1376.042362] kernel: sof-audio-acpi-intel-bdw INT3438:00: preventing DSP entering D3 state to preserve context
[ 1376.042372] kernel: sof-audio-acpi-intel-bdw INT3438:00: ASoC: error at soc_component_trigger on INT3438:00: -110
[ 1376.042378] kernel:  Media Playback 1: ASoC: trigger FE cmd: 0 failed: -110
[ 1376.042399] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1376.545406] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[ 1376.545448] kernel: sof-audio-acpi-intel-bdw INT3438:00: preventing DSP entering D3 state to preserve context
[ 1376.545462] kernel: sof-audio-acpi-intel-bdw INT3438:00: ASoC: error at soc_component_trigger on INT3438:00: -110
[ 1376.545478] kernel:  Low Latency: ASoC: trigger FE cmd: 0 failed: -110
```

XiaoyunWu6666, 2021-10-12:
will close it if not happened again by Oct 15

XiaoyunWu6666, 2021-10-18:
Close it now since it hasn't happened since its first appearance

> I think you meant `TPLG=/lib/firmware/intel/sof-tplg/sof-bdw-rt286.tplg `
 @marc-hb  Yes , sorry for mistakes.

> @XiaoyunWu6666 can you tag any new issue that uses a mixer, mux or demux in the test pipeline (feel free to create a new label for this).
@lgirdwood  I think yes, I will create a 'mixer' tag for it since it happened on -Dhw:0,1
[image]

marc-hb, 2022-02-11:
Happened in https://sof-ci.sh.intel.com/#/result/planresultdetail/10141?model=BDW_WSB_RT286&testcase=multiple-pipeline-capture

Looks like bugs don't fix themselves...

keqiaozhang, 2022-02-15:
From the history, we only caught this issue once 4 months ago. So I think we can close this issue. will reopen it if we run to this issue again.

marc-hb, 2022-02-15:
> From the history, we only caught this issue once 4 months ago.

You seem to have missed my previous comment.

I just searched the test results and multiple pipelines tests are failing on BDW_WSB_RT286 _on a regular basis_.

Please either reduce the priority or close as "wontfix" but don't close as "this almost never happens" because it does happen.

Recent test failures:

```
...
9950/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
9961/BDW_WSB_RT286/multiple-pipeline-capture/multiple-pipeline-capture.sh.html
9980/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
9986/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
9992/BDW_WSB_RT286/multiple-pipeline-capture-100/multiple-pipeline-capture-100.sh.html
9996/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
10012/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
10046/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
10047/BDW_WSB_RT286/multiple-pipeline-capture/multiple-pipeline-capture.sh.html
10053/BDW_WSB_RT286/multiple-pipeline-capture/multiple-pipeline-capture.sh.html
10064/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
10079/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
10086/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
10105/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
10141/BDW_WSB_RT286/multiple-pipeline-all/multiple-pipeline-all.sh.html
10141/BDW_WSB_RT286/multiple-pipeline-capture/multiple-pipeline-capture.sh.html
10146/BDW_WSB_RT286/multiple-pipeline-capture-50/multiple-pipeline-capture-50.sh.html
10151/BDW_WSB_RT286/multiple-pipeline-capture-100/multiple-pipeline-capture-100.sh.html
10186/BDW_WSB_RT286/multiple-pipeline-capture/multiple-pipeline-capture.sh.html
10188/BDW_WSB_RT286/multiple-pipeline-capture/multiple-pipeline-capture.sh.html
```

mengdonglin, 2022-02-22:
Lower the priority to P2 as it's on BDW, but not P3 as it seems to be a generic issue of mixer.

@XiaoyunWu6666 @marc-hb Do you still observe this issue after PR #5374 was merged?

XiaoyunWu6666, 2022-02-22:
Actually it was observed in CI daily tests only once ....
@marc-hb the recent test failures you post look actually #3170 ?

marc-hb, 2022-02-22:
Sorry, I may have confused the two. I've been staring at too many logs lately.

XiaoyunWu6666, 2022-03-14:
this also happen on BYT_MB_NOCODECwhen multiple-pipeline-capture
 in inner daily test result 10992

IO error happened when start hw0:0 PCM to capture first and start hw0:0 PCM and hw0:1 PCM Deep Buffer to aplay
[image]

[dmesg]
```
[ 2298.486099] kernel: sof-audio-acpi-intel-byt 80860F28:00: DAI config 0 matches pcm hw params
[ 2298.486107] kernel: sof-audio-acpi-intel-byt 80860F28:00: rate_min: 48000 rate_max: 48000
[ 2298.486116] kernel: sof-audio-acpi-intel-byt 80860F28:00: channels_min: 2 channels_max: 2
[ 2298.486137] kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: hw params stream 0 dir 0
[ 2298.486170] kernel: sof-audio-acpi-intel-byt 80860F28:00: stream_tag 0
[ 2298.486191] kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 2298.486700] kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 2298.486713] kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: stream dir 0, posn mailbox offset is 1329740
[ 2298.487065] kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: trigger stream 0 dir 0 cmd 1
[ 2298.487088] kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 2298.519223] kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: open stream 1 dir 0
[ 2298.519241] kernel: sof-audio-acpi-intel-byt 80860F28:00: period min 192 max 16384 bytes
[ 2298.519251] kernel: sof-audio-acpi-intel-byt 80860F28:00: period count 2 max 16
[ 2298.519259] kernel: sof-audio-acpi-intel-byt 80860F28:00: buffer max 65536 bytes
[ 2298.519751] kernel: sof-audio-acpi-intel-byt 80860F28:00: DAI config 0 matches pcm hw params
[ 2298.519765] kernel: sof-audio-acpi-intel-byt 80860F28:00: DAI config 0 matches pcm hw params
[ 2298.519772] kernel: sof-audio-acpi-intel-byt 80860F28:00: rate_min: 48000 rate_max: 48000
[ 2298.519781] kernel: sof-audio-acpi-intel-byt 80860F28:00: channels_min: 2 channels_max: 2
[ 2298.519797] kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: hw params stream 1 dir 0
[ 2298.519828] kernel: sof-audio-acpi-intel-byt 80860F28:00: stream_tag 0
[ 2298.987825] kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: trigger stream 0 dir 1 cmd 0
[ 2298.995030] kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx timed out for 0x60040000 (msg/reply size: 12/12)
[ 2298.995057] kernel: sof-audio-acpi-intel-byt 80860F28:00: preventing DSP entering D3 state to preserve context
[ 2298.995066] kernel: sof-audio-acpi-intel-byt 80860F28:00: ------------[ DSP dump start ]------------
[ 2298.995077] kernel: sof-audio-acpi-intel-byt 80860F28:00: Firmware exception
[ 2298.995088] kernel: sof-audio-acpi-intel-byt 80860F28:00: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 2298.995158] kernel: sof-audio-acpi-intel-byt 80860F28:00: unexpected fault 0x600a0008 trace 0x00000000
[ 2298.995173] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: ipc host -> DSP: pending no complete no raw 0x8000000000000000
[ 2298.995185] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: mask host: pending no complete no raw 0x0
[ 2298.995197] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: ipc DSP -> host: pending no complete yes raw 0x600a0008
[ 2298.995209] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: mask DSP: pending no complete no raw 0x0
[ 2298.995221] kernel: sof-audio-acpi-intel-byt 80860F28:00: ------------[ DSP dump end ]------------
[ 2298.995245] kernel: sof-audio-acpi-intel-byt 80860F28:00: ASoC: error at soc_component_trigger on 80860F28:00: -110
[ 2298.995263] kernel:  PCM: ASoC: trigger FE cmd: 1 failed: -110
[ 2298.995284] kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: trigger stream 0 dir 0 cmd 0
[ 2298.995301] kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 2298.995763] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: trace IO error
[ 2299.499540] kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[ 2299.499601] kernel: sof-audio-acpi-intel-byt 80860F28:00: preventing DSP entering D3 state to preserve context
[ 2299.499623] kernel: sof-audio-acpi-intel-byt 80860F28:00: ------------[ DSP dump start ]------------
[ 2299.499654] kernel: sof-audio-acpi-intel-byt 80860F28:00: Firmware exception
[ 2299.499683] kernel: sof-audio-acpi-intel-byt 80860F28:00: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 2299.499778] kernel: sof-audio-acpi-intel-byt 80860F28:00: unexpected fault 0x600a0008 trace 0x00000000
[ 2299.499814] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: ipc host -> DSP: pending no complete no raw 0x8000000000000000
[ 2299.499849] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: mask host: pending no complete no raw 0x0
[ 2299.499881] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: ipc DSP -> host: pending no complete yes raw 0x600a0008
[ 2299.499913] kernel: sof-audio-acpi-intel-byt 80860F28:00: error: mask DSP: pending no complete no raw 0x0
[ 2299.500064] kernel: sof-audio-acpi-intel-byt 80860F28:00: ------------[ DSP dump end ]------------
```

mengdonglin, 2022-06-21:
Move BDW platform to maintenance mode and SOF v2.2 will be the last upstream release with support for this platform. There will be no more bug fixing effort on this platform.

closed 2022-06-21 by mengdonglin