thesofproject/sof #4901: audio: codec_adapter: DTS codec problem when running with sof-mt8195-mt6359-rt1019-rt5682 topology

markbartonxperi, 2021-10-20:
As a result of some internal testing on a machine running the sof-mt8195-mt6359-rt1019-rt5682 topology (but modified to use our DTS Codec pipeline on the speaker path in place of the passthrough pipeline) I received a bug report that audio output was not correct. In the recording I was sent I hear a small segment of buffer repeating ~10 times, then skipping ahead to another segment of buffer, which then repeated.

Looking at the sof log for the machine under test, a buffer under-run is being reported:

eg:

```
[   205547880.373928] (       54166.871094) c0 dai          1.6            src/audio/dai.c:954  WARN dai_copy(): Copy_bytes 192 + avail bytes 0 < period bytes 1920, possible glitch
[   205561422.248390] (       13541.874023) c0 dai          1.6            src/audio/dai.c:963  WARN dai_copy(): nothing to copy

... repeat many times...
```

Unfortunately I don't have access to the machine under test - my development system is an UP! Xtreme board, on which everything works. However, I compared the sof-hda-generic topology I use with the sof-mt8195-mt6359-rt1019-rt5682 topology and noticed the pipeline period in the sof-mt8195-mt6359-rt1019-rt5682 case was 10000, as opposed to 1000. After changing the PIPELINE_PCM_ADD and DAI_ADD declarations in sof-mt8195-mt6359-rt1019-rt5682.m4 and redeploying the topology to the machine under test, everything starting working.

Conversely, when I change the PIPELINE_PCM_ADD and DAI_ADD declarations in my version sof-hda-generic.m4 to use a period of 10000, I am able to replicate the problem heard on the machine under test in my environment.

Repeating the test with my codec_adapter-based DTS codec removed from the pipeline and only EQ in place, everything works on the machine under test with a period of either 1000 or 10000, which made me suspicious of my implementation, of course. I felt like  I should be detecting a change in period (from 192 to 1920) somewhere in the life-cycle of my codec, which I should interpret as a signal to reallocate my input and output buffers. I've added logging in a fair-few locations now, and the period is always reported to my code as 192 bytes.

Is there something other than the period that I'm meant to check in order to understand that I should allocate larger buffers?

Thanks in advance
Mark

lgirdwood, 2021-11-12:
@markbartonxperi any update ?

markbartonxperi, 2021-11-16:
@lgirdwood Apologies for the delayed reply.
As we were able to change the pipeline period for the topology of interest (mt8195) to 1000 we've been able to work around this issue in the short term.
However, it would be great to work out what changes I need to make to allow our codec to work with an optimal buffer size for any pipeline period.
I think you mentioned that I should try following up with @singalsu about this to see if they had any thoughts.
@singalsu: hopefully there's already enough information here to follow where I'm currently with this issue and what I've tried so far, but if there's any more information you need, let me know.

singalsu, 2021-11-18:
@markbartonxperi Does dev->frames work for you? It's set by framework to nominal frames count per copy() schedule. I'm using it in many components to allocate resources. I'm assuming e.g. max 1.1 times of that to be max. samples to process. Maybe that's today a bit excessive so could be closer to 1.0. Due to timer based scheduling there's quite often +/- 1 frames jitter in frames avail/free in buffers.

markbartonxperi, 2021-11-19:
@singalsu Thanks for the input.
I took a look at this today.... I'll try and summarise some of the above and describe what I've looked into today, as I feel like I'm missing something very obvious here :)

My test system is an Up! Xtreme, so I'm working with the `sof-hda-generic.m4` topology. I insert my codec on pipeline 1, which corresponds to the headphone output on that board.

In the first test I log the value of `dev->frames` using the default pipeline configuration in the m4 file:

```
# Low Latency playback pipeline 1 on PCM 0 using max 2 channels of s24le.
# 1000us deadline with priority 0 on core 0
PIPELINE_PCM_ADD(PIPE_HEADSET_PLAYBACK,
	1, 0, 2, s24le,
	1000, 0, 0,
	48000, 48000, 48000)
```
My logging reports the value of `dev->frames` to be `48`, which is expected.

In the second test I change the pipeline configuration to reproduce the conditions for the mt8195 system that I'm ultimately targeting (I don't have physical access to this system, unfortunately) my .m4 file looks like this:

```
# Low Latency playback pipeline 1 on PCM 0 using max 2 channels of s24le.
# 1000us deadline with priority 0 on core 0
PIPELINE_PCM_ADD(PIPE_HEADSET_PLAYBACK,
	1, 0, 2, s24le,
	10000, 0, 0,
	48000, 48000, 48000)
```
In this case, my logging still reports the value of `dev->frames` to be `48`, which I'm not expecting. I would expect this to be `480` so that my code knows it needs to allocate a larger buffer to 'cope' with the larger pipeline period and avoid buffer underruns.

singalsu, 2021-11-19:
@markbartonxperi Ouch, that looks like a bug in the framework or topology macros. Need to find out where.

lgirdwood, 2021-11-19:
Thanks @singalsu - lets take this fix for rc2.

markbartonxperi, 2021-11-19:
@singalsu @lgirdwood Let me know if you want me to test anything, or debug any code paths with logging on my system.
Thanks

lgirdwood, 2021-11-23:
Will get this fixed for v2.0 rc2

singalsu, 2021-11-26:
@markbartonxperi Thanks, I think I have everything needed to debug this. I'll start working with this on Monday.

singalsu, 2021-11-29:
@markbartonxperi I was not able to repeat this issue though I don't have similar device. I did this change to topology for the speaker path of the Chromebook device:

```
diff --git a/tools/topology/topology1/development/sof-cml-rt1011-rt5682-nokwd.m4 b/tools/topology/topology1/development/sof-cml-rt1011-rt5682-nokwd.m4
index 16aaf5b82..33529e72f 100644
--- a/tools/topology/topology1/development/sof-cml-rt1011-rt5682-nokwd.m4
+++ b/tools/topology/topology1/development/sof-cml-rt1011-rt5682-nokwd.m4
@@ -22,7 +22,7 @@ include(`platform/intel/cml.m4')
 # if XPROC is not defined, define with default pipe
 ifdef(`HSMICPROC', , `define(HSMICPROC, volume)')
 ifdef(`HSEARPROC', , `define(HSEARPROC, volume)')
-ifdef(`SPKPROC', , `define(SPKPROC, volume)')
+ifdef(`SPKPROC', , `define(SPKPROC, eq-iir-volume)')
 ifdef(`DMICPROC', , `define(DMICPROC, eq-iir-volume)')
 ifdef(`DMIC16KPROC', , `define(DMIC16KPROC, eq-iir-volume)')
 
@@ -104,12 +104,12 @@ PIPELINE_PCM_ADD(sof/pipe-volume-playback.m4,
        48000, 48000, 48000)
 
 # Low Latency playback pipeline 7 on PCM 5 using max 2 channels of s32le.
-# Schedule 48 frames per 1000us deadline with priority 0 on core 0
+# Schedule 480 frames per 10000us deadline with priority 0 on core 0
 ifdef(`SPKPROC_FILTER1', `define(PIPELINE_FILTER1, SPKPROC_FILTER1)', `undefine(`PIPELINE_FILTER1')')
 ifdef(`SPKPROC_FILTER2', `define(PIPELINE_FILTER2, SPKPROC_FILTER2)', `undefine(`PIPELINE_FILTER2')')
 PIPELINE_PCM_ADD(sof/pipe-SPKPROC-playback.m4,
        7, 5, 2, s32le,
-       1000, 0, 0,
+       10000, 0, 0,
        48000, 48000, 48000)
 undefine(`PIPELINE_FILTER1')
 undefine(`PIPELINE_FILTER2')
@@ -138,11 +138,11 @@ DAI_ADD(sof/pipe-dai-capture.m4,
        1000, 0, 0, SCHEDULE_TIME_DOMAIN_TIMER)
 
 # playback DAI is SSP1 using 2 periods
-# Buffers use s16le format, with 48 frame per 1000us on core 0 with priority 0
+# Buffers use s16le format, with 480 frame per 10000us on core 0 with priority 0
 DAI_ADD(sof/pipe-dai-playback.m4,
        7, SSP, 1, SSP1-Codec,
        PIPELINE_SOURCE_7, 2, s24le,
-       1000, 0, 0, SCHEDULE_TIME_DOMAIN_TIMER)
+       10000, 0, 0, SCHEDULE_TIME_DOMAIN_DMA)
 
 # playback DAI is iDisp1 using 2 periods
 # Buffers use s32le format, with 48 frame per 1000us on core 0 with priority 0

```
Both params() and prepare() of IIR with trace patches report that dev->frames is 480. The IIR copy() also got scheduled every 10 ms. Did you change for HDA generic the DAI schedule as well? If this works (get 480) then the issue is likely in the codec adapter.

lgirdwood, 2021-12-01:
@singalsu it could be a documentation issue as we have to set the 10000 in two places ? i.e. we are setting this at the pipeline endpoints for host and DAI (and there is nor SRC in the pipeline to change the rates or data sizes).

markbartonxperi, 2021-12-01:
@singalsu @lgirdwood Thanks so much for taking a look at this.

I revisited things again in the light of your comments - I'm currently building from commit dda9d0f9f

Now, I have to say that I'm convinced I've reproduced this issue at least 4 or 5 times over since opening this issue, and I'm sure that I've always been changing the period in both the `PIPELINE` and `DAI` declarations. However, yesterday when I changed the hda generic topology to a period of 10000 and traced the value of `dev->frames` in my codec's `prepare` and `process` functions, I'm now seeing a value of `480`! In the event there _hasn't_ been a code change affecting this area, then I have to apologise as the only explanation here would be that I wasn't changing both `PIPELINE` and `DAI` declarations.... I'm _sure_ I was, though....

Anyway, with my codec now updated to allocate its buffers based on `dev->frames` rather than `comp_get_drvdata(dev)->period_bytes` and processing 480 samples per call to process, I'm still seeing (and hearing) buffer underruns. I haven't fully had chance to dig into that some more yet, but thought I'd post an update in the meantime.

Here's a log excerpt of what I'm currently seeing. I will investigate some more soon - I note in the above that the DAI scheduling is changed to `SCHEDULE_TIME_DOMAIN_DMA` which I haven't experimented with yet

```
[     5575014.101386] (          15.156249) c0 hda-dma            ..../intel/hda/hda-dma.c:953  ERROR hda_dma_link_check_xrun(): underrun detected
[     5575025.663885] (          11.562500) c0 dai          1.7            src/audio/dai.c:942  WARN dai_copy(): Copy_bytes 960 + avail bytes 0 < period bytes 3840, possible glitch
[     5587169.621736] (       12143.958008) c0 eq-iir       1.3   ../audio/eq_iir/eq_iir.c:802  INFO eq_iir_copy() dev->frames 480
[     5587424.830059] (         255.208328) c0 dts_codec    1.2   ......../codec_adapter.c:378  INFO codec_adapter_copy() start: codec_buff_size: 480, local_buff free: 1920 source avail 7680
[     5587479.413390] (          54.583332) c0 dts_codec    1.2   ............/codec/dts.c:236  INFO dts_codec_process() dev->frames 480 codec->cpd.consumed 480 codec->cpd.produced 480
[     5587498.215473] (          18.802082) c0 dts_codec    1.2   ......../codec_adapter.c:473  INFO codec_adapter_copy(): processed 480 in this call 1440 bytes left for next period
[     5587513.007139] (          14.791666) c0 hda-dma            ..../intel/hda/hda-dma.c:953  ERROR hda_dma_link_check_xrun(): underrun detected
[     5587524.413388] (          11.406250) c0 dai          1.7            src/audio/dai.c:942  WARN dai_copy(): Copy_bytes 960 + avail bytes 0 < period bytes 3840, possible glitch
[     5599670.037906] (       12145.625000) c0 eq-iir       1.3   ../audio/eq_iir/eq_iir.c:802  INFO eq_iir_copy() dev->frames 480
[     5599925.767062] (         255.729156) c0 dts_codec    1.2   ......../codec_adapter.c:378  INFO codec_adapter_copy() start: codec_buff_size: 480, local_buff free: 1920 source avail 7680
[     5599980.246227] (          54.479164) c0 dts_codec    1.2   ............/codec/dts.c:236  INFO dts_codec_process() dev->frames 480 codec->cpd.consumed 480 codec->cpd.produced 480
[     5599998.944143] (          18.697916) c0 dts_codec    1.2   ......../codec_adapter.c:473  INFO codec_adapter_copy(): processed 480 in this call 1440 bytes left for next period
[     5600013.631642] (          14.687499) c0 hda-dma            ..../intel/hda/hda-dma.c:953  ERROR hda_dma_link_check_xrun(): underrun detected
[     5600025.298308] (          11.666666) c0 dai          1.7            src/audio/dai.c:942  WARN dai_copy(): Copy_bytes 960 + avail bytes 0 < period bytes 3840, possible glitch
[     5612169.985326] (       12144.687500) c0 eq-iir       1.3   ../audio/eq_iir/eq_iir.c:802  INFO eq_iir_copy() dev->frames 480
[     5612425.454066] (         255.468735) c0 dts_codec    1.2   ......../codec_adapter.c:378  INFO codec_adapter_copy() start: codec_buff_size: 480, local_buff free: 1920 source avail 7680
[     5612480.037397] (          54.583332) c0 dts_codec    1.2   ............/codec/dts.c:236  INFO dts_codec_process() dev->frames 480 codec->cpd.consumed 480 codec->cpd.produced 480
[     5612499.047813] (          19.010416) c0 dts_codec    1.2   ......../codec_adapter.c:473  INFO codec_adapter_copy(): processed 480 in this call 1440 bytes left for next period
```

singalsu, 2021-12-02:
@markbartonxperi Great that dev->frames looks correct now! With timer based scheduling the component should be capable to process a bit less or a bit more than nominal frames count (480). Have you tried to let it process more? I don't have a rule but something like few percent more could be safe if RAM and MCPS constraints allow. With DMA events based scheduling the exact 480 frames/10 ms processing should be possible since there is no "jitter" in consumed/produced frames in the pipeline.

singalsu, 2021-12-02:
@markbartonxperi The observations with frames count could maybe due to incomplete topologies build. It's good idea to build topology1 from scratch always. The makefiles maybe do not scan all dependencies. So your topology build may not be what you edited to m4 after run "make" in tools build directory. I've had a similar issue but I could spot where it's in the cmake system unfortunately.

lgirdwood, 2021-12-21:
@markbartonxperi can we close ?

markbartonxperi, 2021-12-24:
@lgirdwood @singalsu Yes, this is all sorted now. Our codec is now working with topologies configured for periods of 1000 or 10000.
Thanks very much for all the help!

closed 2021-12-24 by markbartonxperi