thesofproject/sof #4967: [FEATURE] Optimize audio processing components' source/sink buffer access, deprecate read/write frag

singalsu, 2021-11-05:
**Is your feature request related to a problem? Please describe.**
Yes, audio_stream_read_frag_sXX() and audio_stream_write_frag_sXX()  cause high processing load due to their check for circular buffer wrap for every PCM samples.

**Describe the solution you'd like**
Access buffers in max. size blocks before buffers circular wrap, check wrap, and continue with next blob. The function audio_stream_bytes_without_wrap() has only a small overhead and typically is needed only a couple of times per copy() period.

**Describe alternatives you've considered**
Implement processing access to buffers with Xtensa circular buffers.

**Additional context**
Components those use read/write frag are and initial assumed optimize priority. I assumed priority is high if topologies contain several instances of the component. Medium priority for single instance, low priority for test or demo components. All components in key phrase detection are high priority.

- Crossover (medium)
- DC blocker (medium)
- DRC (medium)
- EQ FIR (medium)
- EQ IIR (high, fix in PR #4940)
- IGO NR (medium)
- KBP (high)
- Mixer (high, fix in PR #4944)
- Multi-band DRC (medium)
- PCM converter generic (high)
- Selector (high)
- Smart amp generic (medium)
- Smart amp Maxim DSM (medium)
- TDFB (medium)
- Volume zero-crossings detect mode (medium)
- Volume generic (low)
- Components in samples, detect test, smart amp test, kwd nn test (low priority)

singalsu, 2021-11-09:
Changed IGO optimize priority to high.

lgirdwood, 2021-11-18:
PCM converter generic (high) - pls do this next.

singalsu, 2021-12-01:
Next ones to optimize are KBP and Selector, unless other requested.

singalsu, 2024-04-25:
This issue can be closed.  All components are now updated.

closed 2024-04-25 by singalsu