thesofproject/sof #4995: What does "Initial infrastructure support for MS Windows" mean to release features in v1.9?

hongshui3000, 2021-11-16:
Can you build code on windows? Or can Windows be used as a host to communicate with dsp like linux?

dbaluta, 2021-11-16:
@hongshui3000 i suppose it is the IPC4 commands protocol used by windows to communicate with the SOF firmware. @lgirdwood could tell you more about this.

lgirdwood, 2021-11-16:
It is indeed the IPC4 and Extended manifest support being added that is used by the Intel Windows audio DSP driver. This will also extend to support the Microsoft APO APIs (for audio processing too) and many other features used by Windows.

hongshui3000, 2021-11-17:
good news.

lgirdwood, 2021-11-17:
> good news.Hope there are some examples of windows and sof communication

There is, please enable the IPC4 MAJOR ABI in the Kconfig to build the Windows based IPC used by the Intel Windows driver. Please note this cannot be tested outside Intel atm as it require access Windows driver code.

marc-hb, 2021-12-10:
> good news.

Closing.

closed 2021-12-10 by marc-hb