thesofproject/sof #4996: [BUG] xrun on multiple NOCODEC platforms when multiple pause release

XiaoyunWu6666, 2021-11-17:
**Describe the bug**
When running multiple pause release ,**dai_report_xrun(): underrun due to no data available**  happened with xrun recovery failed and pipeline stopped.
This happened on multiple platforms including CML_RVP_NOCODEC ,  JSL_RVP_NOCODEC  , TGLU_RVP_NOCODEC ,APL_UP2_NOCODEC  , ADLP_RVP_NOCODEC ,etc , since  2021-11-13 (inner daily result 8155)

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-cnl-nocodec.tplg MODEL=CML_RVP_NOCODEC ~/sof-test/test-case/multiple-pause-resume.sh -r 25

**Reproduction Rate**
Not 100% but can be seen on more than 2 platforms in every daily result after Nov 13

**Environment**
Start Time: 2021-11-12 21:26:21 UTC
End Time: 2021-11-13 00:01:58 UTC
Kernel Branch: topic/sof-dev
Kernel Commit: 66bd4e2a
SOF Branch: main
SOF Commit: 94517163580d
Zephyr Branch: main
Zephyr Commit: 6aa783ed6ac2

**Screenshots or console output**
Take  CML_RVP_NOCODEC as an example 
[console]
```
(22/25) pcm'Port0' cmd'aplay' id'2': Wait for 41 ms before resume                                                                           
                                                                          
 aplay: pcm_write:2061: aplay: pcm_write:2061: write error: Input/output error
write error: Input/output error

2021-11-12 22:32:24 UTC [REMOTE_INFO] pipeline: Port0 with aplay
2021-11-12 22:32:24 UTC [REMOTE_INFO] pipeline: Port0 Deep Buffer with aplay
2021-11-12 22:32:24 UTC [REMOTE_INFO] Check expect exit status
```
[etrace]
```
[   374022631.908509] (     3783752.750000) c0 dma-trace                    src/lib/wait.c:51   ERROR poll timeout reg 487496 mask 63 val 0 us 937
[   377827749.413557] (     3805117.500000) c0 dma-trace                    src/lib/wait.c:51   ERROR poll timeout reg 487496 mask 63 val 0 us 937
[   377866378.734939] (       38629.320312) c0 dma-trace                    src/lib/wait.c:51   ERROR poll timeout reg 488520 mask 63 val 0 us 937
[   381656506.969750] (     3790128.250000) c0 dma-trace                    src/lib/wait.c:51   ERROR poll timeout reg 487496 mask 63 val 0 us 937
[   385007707.722002] (     3351200.750000) c0 dma-trace              src/drivers/dw/dma.c:1114 ERROR dw_dma_get_data_size(): xrun detected
[   385007726.367834] (          18.645832) c0 dma-trace                   src/audio/dai.c:879  ERROR dai_report_xrun(): underrun due to no data available
[   385007739.180334] (          12.812499) c0 dma-trace                   src/audio/dai.c:727  ERROR comp_underrun(): dev->comp.id = 2, source->avail = 768, copy_bytes = 0
[   385007785.482415] (          46.302082) c0 dma-trace          ....../pipeline-stream.c:134  ERROR pipeline_copy(): ret = -61, start->comp.id = 2, dir = 1
[   385007798.659498] (          13.177083) c0 dma-trace          ..../pipeline-schedule.c:160  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.
[   385770207.327119] (      762408.687500) c0 dma-trace              src/drivers/dw/dma.c:1114 ERROR dw_dma_get_data_size(): xrun detected
[   385770226.233368] (          18.906250) c0 dma-trace                   src/audio/dai.c:879  ERROR dai_report_xrun(): underrun due to no data available
[   385770239.150035] (          12.916666) c0 dma-trace                   src/audio/dai.c:727  ERROR comp_underrun(): dev->comp.id = 2, source->avail = 768, copy_bytes = 0
[   385770273.629200] (          34.479164) c0 dma-trace          ....../pipeline-stream.c:134  ERROR pipeline_copy(): ret = -61, start->comp.id = 2, dir = 1
[   385770286.962533] (          13.333333) c0 dma-trace          ..../pipeline-schedule.c:160  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.
```

XiaoyunWu6666, 2021-11-17:
@lyakh 
FYI , @keyonjie run with the PR 4893 that reverted ]PR 4960, result : inner result 8207 The only one failure here is because of trigger release, no xrun happen.

For double confirm , I run another multiple pause release 100 test for multiple nocodec platform with PR 4893 , inner result 8251 (result pending)

lyakh, 2021-11-17:
I've looked at the latest daily result on APL nocodec, and I think I have an idea why this can be happening. The reason might be, that now that we handle STOP and PAUSE in pipeline tasks, we often have to stop several pipelines in cases, like a topology with a mixer, where the host and the DAI belong to different pipelines, and each of them has its own task. So the order might be important and where to send the reply from too... I'll investigate more.

lgirdwood, 2021-11-17:
I think the general robustness fix here is that when we have connected pipeline they are allowed to overrun and underrun without it being fatal. i.e. 
1) pipeline source underruns - we copy 0s
2) pipeline sink overruns - we overwrite oldest data.

lgirdwood, 2021-11-17:
Btw, this wont stop rc1, we can fix for v2.0 final though.

plbossart, 2021-11-17:
> I think the general robustness fix here is that when we have connected pipeline they are allowed to overrun and underrun without it being fatal. i.e.
> 
> 1. pipeline source underruns - we copy 0s
> 2. pipeline sink overruns - we overwrite oldest data.

ALL the default topologies moving forward will have mixers and connected pipelines, not sure there's any merit in defining what we should do in a xrun. What we need to do is avoid the xrun in the first place. I've honestly NEVER seen this sort of concealment provide any benefit, in fact more often than not you end-up with worse problems such as channel swaps, etc.

keyonjie, 2021-11-18:
If there is no simple/quick fix for it, I would suggest to revert PR 4960 for the release, and add back new version which verified pass the multi-core stressful CI later.

lgirdwood, 2021-11-18:
> ALL the default topologies moving forward will have mixers and connected pipelines, not sure there's any merit in defining what we should do in a xrun. What we need to do is avoid the xrun in the [...]

I agree, that the xrun term is being used in too many places, it's really better suited to any external endpoint not consuming or providing data.

The connected pipelines do need to better tolerate (and avoid) underrun and overrun conditions as they can all be independently triggered by the host.  This is why we need the rules above to support connected pipelines to avoid channels swapping (as we work on a frame basis instead of bytes).

lyakh, 2021-11-19:
#5000 now fixes this. See internal test result number 8332.

miRoox, 2021-11-30:
After #5000 merged, this issue seems to happen only on Zephyr nocodec platforms.

lgirdwood, 2021-12-01:
@lyakh the pipeline simplification and ref counting updates after your vacation will really help this.

XiaoyunWu6666, 2022-02-14:
haven't seen this for long time in daily tests , close it

marc-hb, 2022-02-16:
https://sof-ci.01.org/sofpr/PR5192/build12056/devicetest/?model=APL_UP2_NOCODEC_ZEPHYR&testcase=multiple-pause-resume-5

```
[     3342105.075530] (          99.114578) c0 host         9.46  ....../pipeline-stream.c:177  INFO perf comp_copy peak plat 350 cpu 348
[     3342169.450527] (          64.375000) c0 pga          9.47  ....../pipeline-stream.c:177  INFO perf comp_copy peak plat 979 cpu 979
[     3342192.002610] (          22.552082) c0 mixer        5.24     src/audio/component.c:64   INFO comp_set_state(), state already set to 6
[     3342209.138026] (          17.135416) c0 mixer        5.24     src/audio/component.c:64   INFO comp_set_state(), state already set to 5
[     3343170.335904] (         961.197876) c0 host         11.51 ....../pipeline-stream.c:177  INFO perf comp_copy peak plat 282 cpu 279
[     3343233.773402] (          63.437496) c0 pga          11.52 ....../pipeline-stream.c:177  INFO perf comp_copy peak plat 957 cpu 957
[     3343280.544233] (          46.770832) c0 mixer        5.24  ....../pipeline-stream.c:177  INFO perf comp_copy peak plat 480 cpu 481

[     3343656.481718] (         140.208328) c0 pipe         5.27  ....../pipeline-stream.c:261  INFO pipe trigger cmd 6
[     3343681.794217] (          25.312498) c0 dai          5.26  ....../pipeline-stream.c:177  INFO perf comp_copy peak plat 7034 cpu 7035

[     3343931.481707] (         118.072914) c0 zll-schedule       src/schedule/zephyr_ll.c:67   INFO task complete 0xbe066740 pipe-task
[     3343318.096315] (          37.552082) c0 dw-dma                 src/drivers/dw/dma.c:1128 ERROR dw_dma_get_data_size(): xrun detected[     3343408.721312] (          90.625000) c0 dai          5.26           src/audio/dai.c:881  ERROR dai_report_xrun(): underrun due to no data available
[     3343516.273391] (         107.552078) c0 dai          5.26           src/audio/dai.c:748  ERROR comp_underrun(): dev->comp.id = 26, source->avail = 768, copy_bytes = 0[     3343692.835884] (          11.041666) c0 pipe         5.27  ....../pipeline-stream.c:134  ERROR pipeline_copy(): ret = -61, start->comp.id = 26, dir = 1
[     3343813.408795] (         120.572914) c0 pipe         5.27  ..../pipeline-schedule.c:183  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.
[     3343942.731707] (          11.250000) c0 zll-schedule       src/schedule/zephyr_ll.c:69   INFO num_tasks 5 total_num_tasks 5
[     3344155.960865] (         213.229156) c0 zll-schedule       src/schedule/zephyr_ll.c:67   INFO task complete 0xbe067e40 pipe-task 
[     3344168.044198] (          12.083333) c0 zll-schedule       src/schedule/zephyr_ll.c:69   INFO num_tasks 4 total_num_tasks 4
[     3351080.127257] (        6912.083008) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3351094.814756] (          14.687499) c0 host         9.46  ....../pipeline-stream.c:177  INFO perf comp_copy peak plat 395 cpu 393
[     3352080.856384] (         986.041626) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3353080.387594] (         999.531189) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3354080.283387] (         999.895813) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3355080.335431] (        1000.052063) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3356080.283308] (         999.947876) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3357080.335352] (        1000.052063) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3358080.126979] (         999.791626) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3359080.283189] (        1000.156189) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3360080.231066] (         999.947876) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3361080.178943] (         999.947876) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3362080.230986] (        1000.052063) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
[     3363080.335113] (        1000.104126) c0 host         9.46          src/audio/host.c:464  INFO no bytes to copy, 0 free in buffer, 384 available in DMA
```

lyakh, 2022-02-17:
@marc-hb are you sure it's the correct place to put this? Isn't your log showing towards #4414 ?

marc-hb, 2022-02-17:
I don't remember seeing any xrun in #4414, is there?

miRoox, 2022-02-25:
In today's daily test (10509), a very similar xrun error happened on TGLU_RVP_NOCODEC and ADLP_RVP_NOCODEC when running multiple pipelines in parallel, but I cannot reproduce it. I guess it's the same as this issue, but just keep an eye on it.

miRoox, 2022-03-17:
This issue is out-of-date, now #5469 is used to track such xrun on TGLU_RVP_NOCODEC and ADLP_RVP_NOCODEC.

closed 2022-03-17 by miRoox