thesofproject/sof #5018: [BUG] [ADL][ZEPHYR] failed to enable target core for widget SSP0.OUT on ADLP ZEPHYR platform

keqiaozhang, 2021-11-22:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
Please also read https://thesofproject.github.io/latest/contribute/process/bug-tracking.html for further information on submitting bugs.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

lyakh, 2021-11-23:
Zephyr is currently limited to using 2 cores on Intel ADSP, this should be eliminated when https://github.com/zephyrproject-rtos/zephyr/pull/39603 is merged

XiaoyunWu6666, 2021-11-29:
FYI 
in inner daily test 8518 
on  model ADLP_RVP_NOCODEC_ZEPHYR
in testcase check-alsabat-headset-playback
still failed to enable target core for widget SSP0.OUT

lgirdwood, 2021-12-01:
@kv2019i I assume we are bringing up the target core after core 0 is fully booted. @andyross is asynchronous secondary core booting/power off supported today or still WIP for some platforms ?

kv2019i, 2021-12-01:
There's first a simple problem that the core count default is 2 on Zephyr side. That is easy to fix. But with that done, we are seeing some weird errors. With a topology with four PCM pipelines assigned to cores 0, 1, 2 and 3, tests works ok on cores 0 and 3, but the secondary boot fails on cores 1 and 2. Debugging continues. This should be supported in both Zephyr and SOF.

kv2019i, 2021-12-02:
Not a complete fix, but gets us further along: https://github.com/kv2019i/zephyr/pull/new/topic/cavs25-4cores-default

kv2019i, 2021-12-02:
@lgirdwood More issues found, maybe we need to move this to v2.1? New issue https://github.com/thesofproject/sof/issues/5050

kv2019i, 2022-02-23:
Depends on Zephyr https://github.com/zephyrproject-rtos/zephyr/pull/41764

lgirdwood, 2022-02-25:
Duplicate with #5432

keqiaozhang, 2022-02-28:
This issue is blocked by https://github.com/zephyrproject-rtos/zephyr/issues/43087.

kv2019i, 2022-02-28:
To fix this issue, https://github.com/zephyrproject-rtos/zephyr/pull/41764 is mandatory but not enough on its own. Steps to fix stil needed:
- get above Zephyr fix to SOF (gated by https://github.com/zephyrproject-rtos/zephyr/issues/43087 )
- change SOF to use a different target for TGLLP/TGLH builds (cannot be done before above or otherwise we break all TGLH targets)
- change core count for TGL-LP Zephyr in mainline
- verify above fix with SOF (TGL-LP nocodec tests should now pass in SOF CI)

kv2019i, 2022-03-07:
@keqiaozhang  We can close this now, right?

closed 2022-03-08 by kfrydryx