thesofproject/sof #5032: [BUG][Zephyr] IPC error for FILTER_UPDATE when kmod unload after playback on Zephyr platforms

miRoox, 2021-11-26:
**Describe the bug**
IPC error when SOF kernel modules unload after playback on most Zephyr platforms.
This issue has happened in recent daily test results since 11/24/2021.

**To Reproduce**
play the audio and then unload SOF modules, e.g.
```
TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg MODEL=WHL_UPEXT_HDA_ZEPHYR ~/sof-test/test-case/check-kmod-load-unload-after-playback.sh
```

**Reproduction Rate**
Not 100%, but can be reproduced in the first iteration on most Zephyr platforms.

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: https://github.com/thesofproject/linux/commit/d5c4c30a
SOF Branch: main
SOF Commit: https://github.com/thesofproject/sof/commit/f41010d3868d
Zephyr Branch: main
Zephyr Commit: https://github.com/zephyrproject-rtos/zephyr/commit/f8f2936dbac3
platforms: Zephyr-based platforms, like `APL_UP2_NOCODEC_ZEPHYR`, `TGLH_RVP_NOCODEC_ZEPHYR`, `WHL_UPEXT_HDA_ZEPHYR`...

**Screenshots or console output**
Take `WHL_UPEXT_HDA_ZEPHYR` as an example:

[dmesg]
```
[ 6126.749284] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[ 6126.749379] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6126.749399] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6126.750077] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[ 6126.750091] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: widget PIPELINE.9.HDA6.OUT freed
[ 6126.750261] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: close stream 5 dir 0
[ 6126.750402] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6126.750448] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6126.751423] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6126.751462] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6127.093836] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x90040000: GLB_TRACE_MSG: FILTER_UPDATE
[ 6127.094070] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x90040000 (msg/reply size: 60/12): -22
[ 6127.094079] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: fail in sof_ipc_trace_update_filter -22
[ 6127.098105] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6127.098139] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6127.098402] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6127.098454] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6127.260544] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x90040000: GLB_TRACE_MSG: FILTER_UPDATE
[ 6127.260771] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x90040000 (msg/reply size: 60/12): -22
[ 6127.260781] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: fail in sof_ipc_trace_update_filter -22
[ 6127.264810] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 6127.264843] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
```

<details>
<summary>[console]</summary>

```
Hardware PCM card 0 'sof-hda-dsp' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16384
  period_size  : 4096
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
[... 211 lines cut ...]
RMMOD	soundwire_cadence
RMMOD	soundwire_generic_allocation
RMMOD	regmap_sdw
SKIP	regmap_sdw_mbq  	not loaded
RMMOD	soundwire_bus
RMMOD	snd_intel_sdw_acpi
RMMOD	snd_soc_core
RMMOD	snd_hda_codec
RMMOD	snd_hda_core
RMMOD	snd_hwdep
RMMOD	snd_compress
SKIP	snd_pcm_dmaengine  	not loaded
RMMOD	snd_pcm
2021-11-23 23:09:02 UTC Sub-Test: [REMOTE_INFO] checking for general errors after kmod unload with sof-kernel-log-check tool
declare -- cmd="journalctl_cmd --since=@1637708934"
2021-11-23 23:09:02 UTC [ERROR] Caught kernel log error
===========================>>
[ 6127.094070] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x90040000 (msg/reply size: 60/12): -22
[ 6127.094079] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: fail in sof_ipc_trace_update_filter -22
[ 6127.260771] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x90040000 (msg/reply size: 60/12): -22
[ 6127.260781] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: fail in sof_ipc_trace_update_filter -22
<<===========================
```

</details>

lgirdwood, 2021-11-26:
@kv2019i @lyakh @marc-hb fyi

kv2019i, 2021-11-29:
@marc-hb can you take a look at this?

marc-hb, 2021-12-02:
Will have a look this week bar any usual "top priority" request / major regression.

marc-hb, 2021-12-03:
The problem is May 2020 commit 0e1176d9434a413 `trace: Update trace level after ipc message` has never been compatible with Zephyr. So filter updates have been simply disabled by "quickfix" commit b31d19b6c2ad3

I'm still looking at what can be done, in the mean time the error is minor and can be ignored (unless of course you really want to change the filters at run-time - you can't)

marc-hb, 2021-12-07:
https://github.com/thesofproject/sof-test/pull/820 (temporarily?) disables the sof-test code that triggers this error message, please help review.

PR #5066 (2021-12-25): Zephyr: fix log filtering
Fixes #5032.

Hard-dependency on zephyrproject-rtos/zephyr/pull/40975, Zephyr will fail to build until then.
closed 2021-12-25 by lgirdwood