thesofproject/sof #5038: [BUG]

XiaoyunWu6666, 2021-11-29:
**Describe the bug**
xrun and IO error happened when playing with Speaker 
on TGLH_0A5E_SDW and TGLU_SKU0A32_SDCA

This issue has happened in recent daily test results since 11/26/2021 [ inner daily 8502 ]
The last safe daily testresult id is 8479

**To Reproduce**
Run 'aplay   -Dhw:0,2 -r 48000 -c 2 -f S24_LE -d 3 /dev/zero -v -q ' twice
and it should happen at the second time

For your convenience, you can run below commands to observe the issue
```
for i in {1..3}; do ./sof-test/tools/kmod/sof_remove.sh ; sudo ./sof-test/tools/kmod/sof_insert.sh ; sleep 1 ; aplay   -Dhw:0,2 -r 48000 -c 2 -f S24_LE -d 3 /dev/zero -v -q; aplay   -Dhw:0,2 -r 48000 -c 2 -f S24_LE -d 3 /dev/zero -v -q; done
```

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 76583781
SOF Branch: main
SOF Commit: 3ffeb8712354
Zephyr Commit: f1f0dadad33c
platforms: TGLU_SKU0A32_SDCA and TGLH_0A5E_SDW

**Screenshots or console output**
Take TGLU_SKU0A32_SDCA as an example:

<details>
<summary>[dmesg] </summary>
```
[  324.942186] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 2 dir 0 cmd 0
[  324.942203] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  325.169063] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[  325.169115] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[  325.443097] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[  325.443114] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[  325.443119] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  325.443133] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
[  325.443139] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[  325.443155] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  325.443160] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  325.443164] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  325.443168] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware exception
[  325.443173] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  325.443184] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[  325.443551] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
[  325.443556] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  325.443577] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x20140000 successful
[  325.443584] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  325.443593] kernel:  Speaker: ASoC: trigger FE cmd: 0 failed: -110
[  325.443629] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace IO error
[  325.443631] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 2 dir 0
[  325.443642] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  325.669066] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[  325.669104] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[  325.947106] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60030000 (msg/reply size: 12/12)
[  325.947122] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware exception
[  325.947156] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -110
[  325.950118] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  326.451146] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x80010000 (msg/reply size: 216/12)
[  326.451163] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware exception
[  326.451171] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed resetting DAI config for ALH258.OUT
[  326.451207] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  326.955164] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x80010000 (msg/reply size: 216/12)
[  326.955182] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware exception
[  326.955190] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed resetting DAI config for ALH514.OUT
```
</details>

XiaoyunWu6666, 2021-11-29:
The good commit is f41010d3868d
The bad commit is  3ffeb8712354
Will try to do a bisect

kv2019i, 2021-11-29:
Updated priority as this affects many test plans.

lgirdwood, 2021-11-29:
@XiaoyunWu6666 which topology file is used on this failing case ? I can only find TGL topologies with Smart AMP and demux. Looking for the topology above. Thanks.

miRoox, 2021-11-29:
@lgirdwood it's `sof-tgl-rt711-rt1308-rt715.tplg`

lgirdwood, 2021-11-29:
@lyakh I've pushed ```lrg/topic/demux-test``` branch with example DEMUX topology. This should allow testing with SSP on other platforms.

plbossart, 2021-11-29:
@lgirdwood this topology ONLY uses SoundWire. No SSP or DMIC, HDAudio for HDMI only. The demux is used to split the speaker content and render it on two separate SoundWire links, with hardware synchronization between links.

plbossart, 2021-11-30:
@lyakh is PR #5041 trying to fix this issue?

XiaoyunWu6666, 2021-12-01:
in today's inner daily test TGLH_0A5E_SDW and TGLU_SKU0A32_SDCA are good 
Will see how they perform in following workdays

lgirdwood, 2021-12-01:
Thank you @lyakh for your timely fix and @XiaoyunWu6666 for validating.

XiaoyunWu6666, 2021-12-03:
Are we good to close this ?
No such issue on TGLH_0A5E_SDW and TGLU_SKU0A32_SDCA in inner daily tests for 3 workdays

plbossart, 2021-12-03:
let's close. thanks everyone for reporting, fixing and re-testing this issue.

closed 2021-12-03 by plbossart