thesofproject/sof #5043: [FEATURE] SoundWire topologies need to use 3 pipelines for speaker path

plbossart, 2021-11-30:
As pointed out by @lyakh, the SoundWire topologies use 2 pipelines for the aggregated amplifier/speaker path.

[image]

The branch with "Buffer3.2" should really be a different pipeline.

The rule is that each input and output to a mixer/mux/demux shall be handled by a different pipeline.

juimonen, 2021-11-30:
@plbossart isn't this also a driver feature? not sure can asoc handle the 3 pipelines type connection you describe above...
I'm referring to this "virtual dai" thing we have in some topologies to terminate the pipe, would be nice to get rid of that at the same time.

plbossart, 2021-11-30:
No I don't think it's a driver feature. We've already moved the nocodec and some HDaudio topologies to use the split pipelines with a mixer in the middle. Works fine.

The demux with the virtual dai is a different issue IMHO, it's the coupling between playback and capture that's not well handled.

mengdonglin, 2021-12-07:
@plbossart @lyakh Can you please share more info why "The rule is that each input and output to a mixer/mux/demux shall be handled by a different pipeline"?

plbossart, 2021-12-07:
@mengdonglin this is required to allow multiple streams to be handled by the same capture or playback endpoint.

We've started with the 'single pipeline' approach and it's just awful to maintain with a mixer in the middle of a pipeline. It's way simpler to use a 3 pipeline topology as we've done for cavs-nocodec, and are currently extending to HDaudio/SoundWire.

lyakh, 2021-12-13:
The sof-tgl-sdw-max98373-rt5682.m4 topology has a similar and related issue, arguably even more severe: one capture and one playback PCMs are connected there via a demux and a smart amp. If I count correctly such a topology should be using 5 pipelines, currently that part of the topology is using 2 pipelines. That is also currently causing intermittent multiple-pause-resume failures in daily tests, e.g. in test 8806 from Dec 09-10. It seems like we need a demux-DAI pipeline similar to the pipe-mixer-dai-playback.m4

plbossart, 2021-12-13:
@lyakh the demux should be at the end of a host pipeline, not the start of a new dai pipeline one as done for the mixer.

plbossart, 2021-12-13:
> @lyakh the demux should be at the end of a host pipeline, not the start of a new dai pipeline one as done for the mixer.

actually it depends. My comment applies for the case where we use the demux to split the playback stream to go to multiple amps. The case of sof-tgl-sdw-max98373-rt5682.m4 is different, IIRC the demux is on a capture path, isn't it?

lyakh, 2021-12-14:
> actually it depends. My comment applies for the case where we use the demux to split the playback stream to go to multiple amps. The case of sof-tgl-sdw-max98373-rt5682.m4 is different, IIRC the [...]

It is in capture. yes

plbossart, 2024-05-28:
closing, legacy change that we're not going to improve

closed 2024-05-28 by plbossart