thesofproject/sof #5050: [BUG][Zephyr] HDMI on secondary core reports many xruns

kv2019i, 2021-12-02:
**Describe the bug**
Flood of hda link xrun erros in sof-loggers when running HDMI pipe+dai on non-primary core.

**To Reproduce**

**Reproduction Rate**
100%

**Expected behavior**
Similar performance on secondary cores as on primary core.

**Impact**
Showstopper to use additional cores.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev
    * SOF: cb280c4c5968800e56e5c5519088c6138a30ec2f
    * Zephyr: abeaf948554038b1952e07b5a81c3e1b8674e9e9
2) Name of the topology file
    * Topology: modified sof-tgl-rt711-rt1308 to have HDMI1+2+3+4 on cores1+2+3+4 (but any topology would do, using this as I have the hardware at hand)
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL

**Screenshots or console output**
On core3:
```
[   177294676.392443] (        1775.624878) c3 zll-schedule       src/schedule/zephyr_ll.c:132  INFO ll task 0xbe07c040 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 80554, max 81046
[   183438680.419135] (        1779.374878) c3 zll-schedule       src/schedule/zephyr_ll.c:132  INFO ll task 0xbe07c040 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 80644, max 81066
[   189582664.966662] (        1776.249878) c3 zll-schedule       src/schedule/zephyr_ll.c:132  INFO ll task 0xbe07c040 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 80568, max 81076
```

On core0:
```
[    20720566.780805] (      453993.781250) c0 zll-schedule       src/schedule/zephyr_ll.c:132  INFO ll task 0xbe07c040 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 7379, max 9799
[    22768567.324425] (      453994.781250) c0 zll-schedule       src/schedule/zephyr_ll.c:132  INFO ll task 0xbe07c040 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 7336, max 7356
[    24816566.305545] (      453993.468750) c0 zll-schedule       src/schedule/zephyr_ll.c:132  INFO ll task 0xbe07c040 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 7335, max 7358

```

lgirdwood, 2021-12-03:
@kv2019i different clock config on each core ? @andyross ?

kv2019i, 2021-12-13:
With the icache fix of 
https://github.com/zephyrproject-rtos/zephyr/pull/41045

Without icache fix:
[   101175197.490076] (        1121.250000) c1 zll-schedule       src/schedule/zephyr_ll.c:132  INFO ll task 0xbe079840 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 40160, max 40256

With the fix:
[    25074202.336974] (        1098.854126) c1 zll-schedule       src/schedule/zephyr_ll.c:132  INFO ll task 0xbe079840 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 39682, max 39785

Seems rootcause is elsewhere, but at least a data point for impact of the icache fix.

lgirdwood, 2021-12-14:
@kv2019i @andyross some things to check.
1) do we know if the core1 tasks are starting when expected i.e. there is no delay running the c1 task for each tick
2) Is there anyway to dump all secondary core DSP special registers (i.e. regs we read/write with rsr/wsr and compare to core0)
It just seems to me we have a correct config for core0, but something is different with core1..3 and this could be related to slightly different boot paths ?

kv2019i, 2021-12-21:
@lgirdwood  @andyross 
It seems the non-primary DSP cores are not running at correct clocks. I could verify all logic is working as expected, but a simple synchronous processing loop (with no access to HW peripherals, just DSP processing) runs ~8X slower on core1 than core0. The observed speed doesn't match LP/HP cro clocks, but this is pretty close to XTAL output rate, so that's one possibility. I'll go through core config registers next.

kv2019i, 2021-12-28:
@lgirdwood @andyross I now cross-checked the numbers with CCOUNT and bumping irq level to 5 to ensure nothing else is running, and results still match, we are in fact running slower. Today I found one difference. Zephyr is setting local clock gating (TCPLCK to CLKCTL) for core1 while XTOS builds are not. Unfortunately changing this to prevet clock gating also Zephyr doesn't change the results, we are still running slower. Also checked core cache attributes and they are identical.

kv2019i, 2021-12-29:
Created a standalone Zephyr PR to reproduce the problem: https://github.com/zephyrproject-rtos/zephyr/pull/41496

closed 2022-01-05 by nashif