thesofproject/sof #5055: [BUG][Zephyr] fimware boot failure on TGLH_RVP_NOCODEC Zephyr platform

miRoox, 2021-12-06:
**Describe the bug**
SOF-Zephyr boot fails on TGLH_RVP_NOCODEC devices.
This is a Zephyr regression, according to the daily test result, the good point is [fd089b361d8a](https://github.com/zephyrproject-rtos/zephyr/commit/fd089b361d8a) and the bad point is [4b822395ed36](https://github.com/zephyrproject-rtos/zephyr/commit/4b822395ed36).

**To Reproduce**
Build SOF-Zephyr with tgl-h target, boot on DUT.

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: https://github.com/thesofproject/linux/commit/59a72819
SOF Branch: main
SOF Commit: https://github.com/thesofproject/sof/commit/47b159fca336
Zephyr Branch: main
Zephyr commit: https://github.com/zephyrproject-rtos/zephyr/commit/4b822395ed36
topology: sof-tgl-h-nocodec.tplg

**Screenshots or console output**
**dmesg**
```
[    5.058338] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[    5.058342] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware boot failure due to timeout
[    5.058344] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
[    5.058378] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[    5.058472] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: invalid header size 0x3010202. FW oops is bogus
[    5.058474] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00000000
[    5.058476] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[    5.058478] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware -5
[    5.058480] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 2 -> 3
[    5.058666] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x3010303 successful
[    5.059214] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x303 successful
[    5.059219] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    5.059514] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 3 -> 0
[    5.059546] kernel: sof-audio-pci-intel-tgl: probe of 0000:00:1f.3 failed with error -5
```

[full_dmesg.txt](https://github.com/thesofproject/sof/files/7659928/full_dmesg.txt)

kv2019i, 2021-12-07:
D'oh, my mistake completely. I was assuming arch/xtensa/configs/tgph_defconfig would be taken in, but that's not the case at all. I'll come up with a fix.

kv2019i, 2021-12-07:
Fix is available, but requires both Zephyr and SOF changes: 
https://github.com/thesofproject/sof/pull/5059
https://github.com/zephyrproject-rtos/zephyr/pull/40960

marc-hb, 2021-12-10:
Change of plan, @kv2019i submitted a revert to 2 cores: zephyrproject-rtos/zephyr/pull/41013

miRoox, 2021-12-17:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/41013

closed 2021-12-17 by miRoox