thesofproject/sof #5077: How to get dsp log from host?

lenghonglin, 2021-12-13:
Hi Guys, i bought IMX8MP to study SOF.
And i follow this [article](https://thesofproject.github.io/latest/getting_started/nxp/sof_imx_user_guide.html) to build firmware. I get `sof-imx8m.ldc` and `sof-imx8m.ri` finally, Then copy to `/lib/firmware/imx/sof/`.

lgirdwood, 2021-12-13:
@lenghonglin Have you enabled debugFS in your kernel. @dbaluta may have a specific kernel Kconfig for the IMX8MP platform. Also please check https://thesofproject.github.io/latest/index.html for details. If they are missing we can add.

dbaluta, 2021-12-13:
@lenghonglin what kernel are you using? For the SOF debugfs to appear you need:

```
CONFIG_EXPERT=y
CONFIG_SND_SOC_SOF_DEVELOPER_SUPPORT=y
CONFIG_SND_SOC_SOF_DEBUG=y
CONFIG_SND_SOC_SOF_DEBUG_ENABLE_FIRMWARE_TRACE=y
CONFIG_SND_SOC_SOF_DEBUG_ENABLE_DEBUGFS_CACHE=y

```

lenghonglin, 2021-12-13:
[image]
I don't see `sof`.

[image]

[image]

But there are  sof files. So i doubt sof is not be loaded.

And this is my device tree, there is no `sof`， whether sof dts is not loaded?

[image]

dbaluta, 2021-12-13:
First 5.4 is ancient. 

You could use https://source.codeaurora.org/external/imx/linux-imx?h=lf-5.10.y

We are working on having 5.15 by th end of the year.

Make sure you boot your board with imx8mp-mek-sof-wm9860.dtb otherwise SOF will not be enabled.

lenghonglin, 2021-12-13:
> First 5.4 is ancient.
> 
> You could use https://source.codeaurora.org/external/imx/linux-imx?h=lf-5.10.y
> 
> We are working on having 5.15 by th end of the year.
> 
> Make sure you boot your board with imx8mp-mek-sof-wm9860.dtb otherwise SOF will not be enabled.

OK, i will try to build 5.15

dbaluta, 2021-12-15:
@lenghonglin where did you buy i.MX8MP and how much did it cost?

@ranj063 maybe we can figure out a way to send you an i.mx8mp for codec_adapter patches.

An alternate way, would be to enable compress patches for Intel (I can point you to NXP internal tree, current kernel is 5.15). The only way we need to figure out after that is to create a topology for one of your intel boards.

ranj063, 2021-12-15:
> @lenghonglin where did you buy i.MX8MP and how much did it cost?
>
> @ranj063 maybe we can figure out a way to send you an i.mx8mp for codec_adapter patches.
>
> An alternate way, would be to enable [...]

I like the alternate suggestion @dbaluta

dbaluta, 2021-12-15:
@ranj063 I think you can try to cherry-pick this patch https://github.com/thesofproject/linux/pull/3222/commits

into your tree and then adapt this topology https://github.com/thesofproject/sof/blob/main/tools/topology/topology1/development/sof-imx8-compr-mp3-wm8960.m4

for intel DAI and codec.

With this kernel is ready! Then on the firmware side you need to set this config options:

CONFIG_COMP_CODEC_ADAPTER
CONFIG_CADENCE_CODEC
CONFIG_CADENCE_CODEC_MP3_DEC_LIB

Lets have a meeting tomorrow to see how it goes. I will be available in your morning time.

lenghonglin, 2021-12-16:
> First 5.4 is ancient.
> 
> You could use https://source.codeaurora.org/external/imx/linux-imx?h=lf-5.10.y
> 
> We are working on having 5.15 by th end of the year.
> 
> Make sure you boot your board with imx8mp-mek-sof-wm9860.dtb otherwise SOF will not be enabled.

Sorry for an awkward question that How to include imx8mp-mek-sof-wm9860.dtb?

dbaluta, 2021-12-16:
@lenghonglin lets chat on. https://gitter.im/thesofproject/community

marc-hb, 2022-01-25:
Is this still an issue?

lgirdwood, 2022-01-27:
Can reopen if still an issue.

closed 2022-01-27 by lgirdwood