thesofproject/sof #5079: [BUG] [v2.0] FW replied with error. Error code: -22 in test 07_06_TestKdMemLeakStaticPipelines

kkarask, 2021-12-13:
**Describe the bug**
Error occurs in test 07_06_TestKdMemLeakStaticPipelines:
FW replied with error.
                 Error code: -22
                 0x00000005 0x00000000 0x00004000

**Topology**
```
 pipe_plb
               +------------------+
        +----+ | +---+  +-------+ |
        |Host|-->|Buf|->|SSP Dai|----------------------------------+
        +----+ | +---+  +-------+ |                                |
               +----------------- +                                |
                                                                   |
                   +-------------------------+                     |
        +--------+ | +---+   +-----+   +---+ |                     |
        |KD Dummy|<->|Buf|<->|ChSel|<->|Buf|<->+                   |
        +--------+ | +---+   +-----+   +---+ | |  pipe_cap         |
                   +-------------------------+ |                   |
                                               |                   |
                             +-----------------+                   |
        +--------------------|-----------------------------------+ |
        |+----+  +------+  +---+  +---+  +---+  +---+  +-------+ | |
        ||Host|<-|Buffer|<-|Kpb|<-|Buf|<-|Vol|<-|Buf|<-|SSP Dai|---+
        |+----+  +------+  +---+  +---+  +---+  +---+  +-------+ |
        +--------------------------------------------------------+
```

**To Reproduce**
Run test 07_06_TestKdMemLeakStaticPipelines

**Reproduction Rate**
Sporadic. 

**Environment**
1) Firmware branch name and commit
Git branch: v2.0, hash: 97cc729e1dcf553d4c119a339771e8ff837e2a3b

3) Name of the platform(s) on which the bug is observed.
    * Platform: APL

Logs:
[APL_07_06_TestKdMemLeakStaticPipelines.zip](https://github.com/thesofproject/sof/files/7703866/APL_07_06_TestKdMemLeakStaticPipelines.zip)

bkokoszx, 2021-12-14:
@kkarask Do we have a reproduction also on main branch?

lgirdwood, 2021-12-14:
@kkarask this looks a good test, is there anyway we can get this test topology in topology format for developer debug ? Fyi @lyakh will be looking at connected pipeline issues and logic in the next few weeks so this topology would be needed for the debug.

bkokoszx, 2021-12-14:
@lgirdwood maybe we are missing some patch from main branch on stable-v2.0?

lgirdwood, 2021-12-16:
> @lgirdwood maybe we are missing some patch from main branch on stable-v2.0?

@kkarask can you confirm this works on main branch but not on v2.0 ?
I think we have all the fixes, but if we can confirm then I can check for missing patches.

kkarask, 2021-12-17:
@lgirdwood This bug could be related with #4961. I have checked one more time and this issue occurs only if playback iterations is greater than 1 then the Error code: -22 shows up. 
Sum up this issue doesn't work on main branch either. Sorry for misleading you.

wszypelt, 2023-01-02:
TGL issue does not occur
Tested FW hash: [4943aaa](https://github.com/thesofproject/sof/commit/4943aaa8907149cf6c02e9046e2ef5d40ba72d74)

closed 2023-01-02 by wszypelt