thesofproject/sof #5082: [BUG]  FW replied with error. Error code: -22 0x00000005 0x00000000 0x00004000 in test 07_06_TestKdMemLeakStaticPipelines

kkarask, 2021-12-14:
**Describe the bug**
Error code occurred in test 07_06_TestKdMemLeakStaticPipelines
 FW replied with error. Error code: -22 0x00000005 0x00000000 0x00004000

**Topology**
```
 pipe_plb
               +------------------+
        +----+ | +---+  +-------+ |
        |Host|-->|Buf|->|SSP Dai|----------------------------------+
        +----+ | +---+  +-------+ |                                |
               +----------------- +                                |
                                                                   |
                   +-------------------------+                     |
        +--------+ | +---+   +-----+   +---+ |                     |
        |KD Dummy|<->|Buf|<->|ChSel|<->|Buf|<->+                   |
        +--------+ | +---+   +-----+   +---+ | |  pipe_cap         |
                   +-------------------------+ |                   |
                                               |                   |
                             +-----------------+                   |
        +--------------------|-----------------------------------+ |
        |+----+  +------+  +---+  +---+  +---+  +---+  +-------+ | |
        ||Host|<-|Buffer|<-|Kpb|<-|Buf|<-|Vol|<-|Buf|<-|SSP Dai|---+
        |+----+  +------+  +---+  +---+  +---+  +---+  +-------+ |
        +--------------------------------------------------------+
```

**To Reproduce**
Run test 07_06_TestKdMemLeakStaticPipelines with Diagnostic driver

**Reproduction Rate**
100%

**Environment**
1) Firmware branch name and commit
Branch: main, Hash: cb280c4c5968800e56e5c5519088c6138a30ec2f

2) Name of the platform(s) on which the bug is observed.
    * Platform: APL

Logs:
[APL_07_06_TestKdMemLeakStaticPipelines.zip](https://github.com/thesofproject/sof/files/7706954/APL_07_06_TestKdMemLeakStaticPipelines.zip)

bkokoszx, 2021-12-21:
@kkarask I assume this is a regression. Could you provide exact commit, which introduces the regression?

wszypelt, 2023-01-02:
unsupported platform
on TGL issue does not occur
Tested FW hash: [4943aaa](https://github.com/thesofproject/sof/commit/4943aaa8907149cf6c02e9046e2ef5d40ba72d74)

closed 2023-01-02 by wszypelt