thesofproject/sof #5083: [BUG] Glitch in the whole file on JSL platform

kkarask, 2021-12-14:
**Describe the bug**
Glitch was found in the whole file in test 07_04_TestKdDmic16000Hz24b32b2ch

**Topology**
```
+----------+   +------------------+
        |    OS    |-->|   AUDIO DEVICE   |-----------------------+
        +----------+   +------------------+                       |
                                                                  |
                   +-------------------------+                    |
        +--------+ | +---+   +-----+   +---+ |                    |
        |KD Dummy|<->|Buf|<->|ChSel|<->|Buf|<->+                  |
        +--------+ | +---+   +-----+   +---+ | |  pipe_cap        |
                   +-------------------------+ |                  |
                                               |                  |
                             +-----------------+                  |
        +--------------------|----------------------------------+ |
        |+----+  +------+  +---+  +---+  +---+  +---+  +------+ | |
        ||Host|<-|Buffer|<-|Kpb|<-|Buf|<-|Vol|<-|Buf|<-| DMIC |---+
        |+----+  +------+  +---+  +---+  +---+  +---+  +------+ |
        +-------------------------------------------------------+
```

**To Reproduce**
Run test 07_04_TestKdDmic16000Hz24b32b2ch with Diagnostic driver

**Reproduction Rate**
1/75.

**Environment**
1) Firmware branch name and commit
Tag: v2.0, Hash: b678a4dcaa1e7ad0c6a056e17dd38bfe095a4464

2) Name of the platform(s) on which the bug is observed.
    * Platform: JSL

**Screenshots or console output**

[image]

[JSL_07_04_TestKdDmic16000Hz24b32b2ch.zip](https://github.com/thesofproject/sof/files/7707236/JSL_07_04_TestKdDmic16000Hz24b32b2ch.zip)

lgirdwood, 2021-12-14:
@kkarask can you retest with #5049 as we have discovered a big optimization for KPB that may help here.

kkarask, 2021-12-16:
@lgirdwood FW was build: stable-v2.0 with #5049 and reproduction of this bug is 1/500.

singalsu, 2021-12-21:
Is the glitch happening in | Host | or | KD Dummy| ? Select is usually 1ch but it can operate with 2ch bypass so I'm wondering if e917018f62abd380e99c64b12f7f2b5299340e5f could impact.

kkarask, 2021-12-22:
@singalsu Output wav file is on the Host side according to current knowledge. After detection KD Dummy should find out it and go to the Host. Is there any way to confirm it? Maybe some additional logs?
I have checked both commits you pointed and there is no impact. All 500 iterations passed.

lgirdwood, 2021-12-22:
The glitch is very periodic and looks to be 1 per ms, this could point to the DMA (host or DMIC). @kkarask what's the buffer size for each DMA ?

kkarask, 2021-12-23:
@lgirdwood If you mean buffer size on host side it is 1MB

lgirdwood, 2021-12-23:
> @lgirdwood If you mean buffer size on host side it is 1MB

Sorry, I mean on the DSP. We need at least 3 * 1ms on the HDA GW DMA and the same again for the DMIC DMA.

lgirdwood, 2022-01-03:
> @lgirdwood Do you mean buffer on FW side? Unfortunately, I have no such knowledge.

Yes, we need a way to upstream these test topologies so they can be reproduced by developers to check/debug.

wszypelt, 2023-01-02:
unsupported platform
on TGL issue does not occur

closed 2023-01-02 by wszypelt