thesofproject/sof #5093: [FEATURE][Zephyr] Make Zephyr version visible in extended manifest and/or version logs

kv2019i, 2021-12-15:
**Is your feature request related to a problem? Please describe.**
Currently it is impossible to tell from FW behaviour and default Linux kernel logs, whether the FW was built with XTOS or Zephyr as RTOS. Furthermore, it's not possible to discover the version of Zephyr that has been used.

**Describe the solution you'd like**
Possible solutions:
 - adding to FW_READY 
 - adding to extended manifest
 - printing a FW trace that identifies the RTOS and version (not ideal as FW traces are not always enabled, but this will help e.g. in analyzing logs in CI)

**Additional context**
Example problem: one machine in Intel SOF CI was mistakenly configured to XTOS although DUT name was "Zephyr". This gave the impression test results were shown for Zephyr build, while in fact an XTOS build was used.

FYI @marc-hb @lgirdwood @lyakh

marc-hb, 2021-12-15:
There is a `is_zephyr` test in sof-test but I agree it's a hack:

It searches for "zephyr" strings in the .ldc file and  that file is of course optional.
https://github.com/thesofproject/sof-test/commit/007683c95d7bca00

So I don't think you can "cleanly" tell from Linux right now. The FW_READY field seems limited, any suggestion on how to extend it? How about something (new?) in `/sys/kernel/debug/sof/` instead?

From the sof-logger it's much easier: if you see `zll-scheduler` then it's Zephyr, if you see `ll-scheduler` then it's XTOS. That is admittedly by accident but it's reliable. I will submit a PR that adds the information in the FW ABI banner.

marc-hb, 2021-12-17:
Note the Zephyr commit is visible in the build logs, see for instance f8cd20fa3e74 in https://sof-ci.01.org/sofpr/PR5066/build11393/build/tgl_zph_error.txt

Ideally, each test run should be unambiguously linked to well defined build, at least in CI it should be. But I agree it's not always the case and it's rare to have "too much" version control.

lgirdwood, 2021-12-21:
extended manifest ?

marc-hb, 2022-01-14:
Some progress at last, the very first step is to fix the clash and confusion between Zephyr's version.h and SOF's version.h, please review #5207. I have a number of unsubmitted fixes that depend on #5207

lgirdwood, 2022-01-27:
@marc-hb both PRs merged, happy to close or do you have other updates pending ?

marc-hb, 2022-01-27:
There's at least one more fix I'm working on.

dbaluta, 2022-02-08:
@marc-hb so after your fixes we will have a banner with zephyr version + SOF version, right?

lgirdwood, 2022-02-09:
> This banner is already there, see links above. There's still a dictionary checksum issue and I also want to take a look at the manifest.
> 
> Also note:
> 
> * [cmake: version.h generation performed at build time zephyrproject-rtos/zephyr#42527](https://github.com/zephyrproject-rtos/zephyr/pull/42527)

@marc-hb whats teh ETA for the above ? Will be creating a stable v2.1 in about two weeks and I think this will be needed.

marc-hb, 2022-02-10:
Just noticed that #5224 is now broken, now it shows: `zephyr:BUILD_VERSION` as in:

`INFO DMA: FW ABI 0x3014001 DBG ABI 0x5003000 tags SOF:v2.0-rc1-432-g364a34ae3b00 zephyr:BUILD_VERSION` :-(
Example: https://sof-ci.01.org/sofpr/PR5340/build11976/devicetest/?model=TGLU_UP_HDA_ZEPHYR&testcase=check-sof-logger

Probably because of
- zephyrproject-rtos/zephyr/pull/42527

> @marc-hb whats teh ETA for the above ? Will be creating a stable v2.1 in about two weeks and I think this will be needed.

It depends whether this is higher priority than building and testing IPC4

lgirdwood, 2022-02-11:
> > @marc-hb whats teh ETA for the above ? Will be creating a stable v2.1 in about two weeks and I think this will be needed.
> 
> It depends whether this is higher priority than building and testing IPC4

No, lets track this for v2.1, it can be fixed in next 5 weeks before v2.1 final.

marc-hb, 2022-02-24:
> Just noticed that https://github.com/thesofproject/sof/pull/5224 is now broken, now it shows: zephyr:BUILD_VERSION

Fix submitted in #5409 , please review. I will backport to 2.1. Sorry it took so long, I had no idea the fix would be so simple. The commit message of the regression ("cmake... performed at build time") scared me and there was always something more urgent to do like actual test failures and IPC4 stuff.

lgirdwood, 2022-03-02:
@marc-hb if this is merged I can label for v2.1 and close ?

marc-hb, 2022-03-02:
The dictionary checksum is still not implemented for Zephyr, using the git SHA as a fallback. I have some work in progress to fix this, I still want to try to make it for 2.1. I think it's important considering how "hybrid" 2.1 is going to be.

marc-hb, 2022-03-08:
> The dictionary checksum is still not implemented for Zephyr, using the git SHA as a fallback. I have some work in progress to fix this, I still want to try to make it for 2.1. I think it's important considering how "hybrid" 2.1 is going to be.

Fix submitted in #5485. I love CMake (not)

plbossart, 2022-06-07:
Heads up to Zephyr folks: the firmware version information is typically the first thing we look at when trying to support the community. We absolutely need to have the ability to add all the information that's currently only visible with sof-logger in the kernel console logs. @marc-hb @kv2019i @lgirwood this needs to be added 'soon' that the upstream kernel logs this information.

````
INFO DMA: FW ABI 0x3016001 DBG ABI 0x5003000 tags SOF:v2.0-rc1-949-gebc222dab568 zephyr:zephyr-v3.1.0-274-g0a758c36c864 src hash 0x1e737c57 (ldc hash 0x1e737c57)

````

kv2019i, 2023-01-11:
Implemented in https://github.com/thesofproject/sof/pull/6378 . Closing.

closed 2023-01-11 by kv2019i