thesofproject/sof #5094: [FEATURE][Zephyr] Implement arch_cpu_disable_core and cpu_power_down_core in wrapper.c

kv2019i, 2021-12-15:
**Is your feature request related to a problem? Please describe.**
Currently secondary cores are only powered down when DSP is put to suspend, or via out-of-band means (on Intel cAVS1.5, host controls power directly for the DSP cores).

Zephyr now implements interface to start/stop xtensa cores, so this should be used in wrapper.c.

lgirdwood, 2021-12-16:
We need to start using native Zephyr APIs and wrap the xtos version of these functions.

bkokoszx, 2021-12-21:
@mwasko fyi

kv2019i, 2022-02-01:
Implemented with https://github.com/thesofproject/sof/pull/5298 .

The native interface will be a Zephyr replacement of SOF runtime PM for DSP cores.

lgirdwood, 2022-02-02:
@kv2019i can we close now ?

kv2019i, 2022-02-02:
Yes, closing.

kv2019i, 2022-02-02:
Unfortunately have to reopen this immediately back :( -> https://github.com/thesofproject/sof/pull/5303

kv2019i, 2023-06-20:
With @juimonen 's PR https://github.com/thesofproject/sof/pull/7183 merged, we can close this.

closed 2023-06-20 by kv2019i