thesofproject/sof #5096: [BUG] DSP panic with HDaudio topologies w/ deep-buffer

plbossart, 2021-12-15:
**Describe the bug**

With deep-buffer enabled, the DSP panics (v1.9.3)

**To Reproduce**

https://github.com/thesofproject/sof/pull/4611 (copy topology on target)
https://github.com/thesofproject/linux/pull/3331

 TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg ~/sof-test/test-case/multiple-pipeline-playback.sh -l 1

**Reproduction Rate**

100%

**Expected behavior**

no DSP panic.

**Impact**

Difficult to progress on kernel patches...

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: sof-hda-generic-4ch.tplg (modified with deep-buffer)
3) Name of the platform(s) on which the bug is observed.
    * Platform: Up Extreme

**Screenshots or console output**

````
[  104.527329] sof-audio-pci-intel-cnl 0000:00:1f.3: hda_link_dma_trigger: cmd=1
[  104.527342] sof-audio-pci-intel-cnl 0000:00:1f.3: hda_dai_trigger: cmd=1
[  104.527348] sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 31 dir 0 cmd 1
[  104.527362] sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x180]=0x2024001e successful
[  104.527373] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  104.527493] sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  104.527503] sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x2014001e successful
[  104.527544] sof-audio-pci-intel-cnl 0000:00:1f.3: format_val=49, rate=48000, ch=2, format=10
[  104.528040] sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  104.528051] sof-audio-pci-intel-cnl 0000:00:1f.3: DSP panic!
[  104.528057] sof-audio-pci-intel-cnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  104.528068] sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[  104.528155] sof-audio-pci-intel-cnl 0000:00:1f.3: reason: runtime exception (0x6)
[  104.528162] sof-audio-pci-intel-cnl 0000:00:1f.3: trace point: 0x00004000
[  104.528169] sof-audio-pci-intel-cnl 0000:00:1f.3: panic at :0
[  104.528176] sof-audio-pci-intel-cnl 0000:00:1f.3: error: DSP Firmware Oops
[  104.528182] sof-audio-pci-intel-cnl 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  104.528190] sof-audio-pci-intel-cnl 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000000 PS       0x00060f25 SAR     0x00000000
[  104.528199] sof-audio-pci-intel-cnl 0000:00:1f.3: EPC1     0xbe00e103 EPC2     0xbe00d5fe EPC3     0x00000000 EPC4    0x00000000
[  104.528207] sof-audio-pci-intel-cnl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  104.528215] sof-audio-pci-intel-cnl 0000:00:1f.3: EPS2     0x00060120 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[  104.528222] sof-audio-pci-intel-cnl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[  104.528229] sof-audio-pci-intel-cnl 0000:00:1f.3: stack dump from 0xbe085be0
[  104.528238] sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe085be0: 00000000 00000000 00000000 00000000
[  104.528246] sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe085bf0: 00000000 00000000 00000000 00000000
[  104.528253] sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe085c00: be085be0 00000000 00000000 00000000
[  104.528261] sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe085c10: 00000000 00000000 00000000 00000000
[  104.528268] sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe085c20: 7e01f155 be085c60 00000000 0000001f
[  104.528275] sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe085c30: be00e103 00000000 00000000 00000000
[  104.528282] sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe085c40: 00000000 00000000 00000000 00000000
[  104.528289] sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe085c50: be01f23b be085e60 00000000 fe01f155
[  104.528295] sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
````

I realize this is an experimental setup, but the trigger error and panic make me think of a firmware issue and possibly trigger changes.

This worked a couple of months ago, albeit with xruns....

@lyakh @ranj063 FYI

lgirdwood, 2021-12-16:
Looks like an assert()

plbossart, 2021-12-16:
I vote with two hands to remove all asserts from the firmware. It makes our life miserable for no good reason.

lgirdwood, 2021-12-16:
> I vote with two hands to remove all asserts from the firmware. It makes our life miserable for no good reason.

Yep, this is agreed - I think its an ongoing process.

lyakh, 2021-12-17:
@lgirdwood @plbossart for cases where there's really no meaningful path forward in the firmware - would it be possible to save some debug information in an area of RAM, preserved over resets, reset and then read out and send that debug information to the host via normal logging?

plbossart, 2021-12-17:
@lyakh isn't that what the etrace is about, saving stuff in memory windows so that they can be read on errors?

ranj063, 2021-12-17:
> @lgirdwood @plbossart for cases where there's really no meaningful path forward in the firmware - would it be possible to save some debug information in an area of RAM, preserved over resets, reset and then read out and send that debug information to the host via normal logging?

@lyakh we can use memory window 0 in this case no?

lyakh, 2021-12-20:
@plbossart @ranj063 makes sense, so we just have to implement that :-)

lgirdwood, 2021-12-21:
panic() should do this today.

lgirdwood, 2022-02-16:
FW crash so P1

jsarha, 2022-02-23:
I tried to reproduce this on my up-extreme-i11, but could not. The test case fails when I run it (due to excessive number of detected underruns?), but there is not DSP crash dump in dmesg and no obvious problem in dtrace.

This is what I did when I tried to reproduce:

1. Set my sof repo head to plbossart/sdw/add-headset-deep-buffer
2. Build fw and topologies and copy sof-tgl.ri, sof-tgl.ldc, and
   sof-hda-generic-4ch.tplg to DUT
3. Reset linux repo to 7f41ec721b3ae5514d60ebcf8bb3845c7449ab9e, build and
   install to DUT (I tried resetting to plb/fix/hda-dai-suspend too)
4. Reboot DUT and run:
   sudo TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg /home/oku/work/sof-test/test-case/multiple-pipeline-playback.sh -l 1

Was there something I missed?

lgirdwood, 2022-02-23:
@plbossart does @jsarha need any specific kernel branch or commits here to reproduce ?

lgirdwood, 2022-02-28:
@XiaoyunWu6666 have you tested or seen this ?

plbossart, 2022-03-11:
can't reproduce this, closing.

closed 2022-03-11 by plbossart