thesofproject/sof #5098: [SOF] sof/script/run-mocks.sh with error

bawanqin1987, 2021-12-16:
**Describe the bug**
when we wanna run ./modules/audio/sof/scripts/run-mocks.sh with ZEPHYR_WORK_SPACE for testting cmock ut case.
1.config the SOF as below,BUILD_UNIT_TESTS=ON,BUILD_UNIT_TESTS_HOST=OFF
[image]

2.and run the sh,there is error:
[image]
show xt-xcc is unknown.but i had already define the xtensa toolchain path.

**Expected behavior**
it's ok to test pass

**Impact**
i can not use the xt-xcc ut case with simulation

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {ZEPHYR OS}
    * SOF: {run with hifi4}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

- [ ] 

- [ ] 

- [ ] 

- [ ]

bawanqin1987, 2021-12-16:
and then ,when i ON the BUILD_UNIT_TESTS_HOST,and the test ok.
i check the log,it's use local gcc,but it's not xt-xcc

bawanqin1987, 2021-12-16:
yes,i had already install the hifi4 xtensa toolchain:
[image]

because i use the the xtensa_build_zephyr.sh is ok to build
[image]

bawanqin1987, 2021-12-16:
i check the run_cmock sh and xtensa_build_all.sh ,The environment Settings are the same
[image]

bawanqin1987, 2021-12-16:
Failed to compile here, but wonder why show config failed  
[image]

bawanqin1987, 2021-12-16:
and i check the process,sof build cmock with host and xtensa supported,you can check the process.
They all have macros to control them
native compiler and toolchain compiler also supported.

bawanqin1987, 2021-12-16:
i set the PATH with my run cmock.sh,as below:
[image]
and run the shell:
[image]

why the process dual with xt-gcc,i can not set this tool.

bawanqin1987, 2021-12-16:
ok, and i modify the run cmock -D
[image]

and check the compile log:
[image]

but there some files report error,I suspect that files compiled with XT-XCC do not match precompiled data such as file data types  
[image]

lgirdwood, 2021-12-16:
@bawanqin1987 can you try and update the SOF module to the latest HEAD.
We will need to wait for @marc-hb to comment here on the configuration required to use XT toolchain.

marc-hb, 2021-12-17:
This bug is really too confusing, please close it and file a new, clearer and simpler bug if needed (probably not).

- Don't use screenshots, use copy/paste. Screenshots are not searchable and not indexed and they block copy/paste of your commands which is required to reproduce your issue.
 
- Copy/paste only the relevant bits.

- If you make code changes, at the very least share your modifications in a github branch or if it's very small copy/paste the git diff. We don't have time to scrutinize your screenshots to identify what you changed. The branch does not have to be pull request. There are of course exceptions but in general code changes that have not been reviewed are not supported.

- Describe one and only problem and failure per issue. There are 8 comments and each one seem to describe some different experiment.

- When reporting a bug, provide the exact and complete sequence of instructions that will other people absolutely sure they can reproduce the exact same error. No guessing.

- What is ZEPHYR_WORK_SPACE and what has it to do with any of this?

- These wrapper scripts provide default CMake parameters. If you need different CMake parameters it's often simpler NOT to use the wrapper script at all and invoke CMake directly.

- Finally, there is AFAIK is no wrapper script to run unit tests with the xtensa toolchain. To run unit tests with the xtensa toolchain, do not use any wrapper script and follow the official documentation: https://thesofproject.github.io/latest/developer_guides/unit_tests.html

http://www.catb.org/~esr/faqs/smart-questions.html

bawanqin1987, 2021-12-17:
ok  ,i  submit another case for you

closed 2021-12-17 by bawanqin1987