thesofproject/sof #5102: [SOF]Run cmock with xtensa toochain,build and excute error

bawanqin1987, 2021-12-17:
**sof/script/run-mocks.sh with xtensa toolchain report some error**
1. i check the run cmock process,it's support HOST and XTENSA toolchain(xt-gcc/xt-xcc)
2. so,i modify the run cmock configs,as below:
larryba@debian:~/project/zephyrproject$ cd modules/
audio/          bsim_hw_models/ crypto/         debug/          fs/             hal/            lib/            tee/
larryba@debian:~/project/zephyrproject$ cd modules/audio/sof/scripts/
larryba@debian:~/project/zephyrproject/modules/audio/sof/scripts$ pwd
/home/larryba/project/zephyrproject/modules/audio/sof/scripts
larryba@debian:~/project/zephyrproject/modules/audio/sof/scripts$ ls
build_csk6_xcc  build_ut       cmake                     docker_build    docker-run.sh  host-build-all.sh  kconfig        README.docker         run-mocks.sh  sof_fw.kb                sof-pre-commit-hook.sh  spelling.txt         xtensa-build-test.sh
build-tools.sh  checkpatch.pl  const_structs.checkpatch  docker-qemu.sh  gen-doc.sh     host-testbench.sh  qemu-check.sh  rebuild-testbench.sh  scan          sof-post-commit-hook.sh  sof-target-install.sh   xtensa-build-all.sh  xtensa-build-zephyr.sh
larryba@debian:~/project/zephyrproject/modules/audio/sof/scripts$ git status
位于分支 feature/ut_test
您的分支与上游分支 'origin/feature/ut_test' 一致。

尚未暂存以备提交的变更：
  （使用 "git add/rm <文件>..." 更新要提交的内容）
  （使用 "git restore <文件>..." 丢弃工作区的改动）
        修改：     ../CMakeLists.txt
        修改：     run-mocks.sh
        修改：     xtensa-build-all.sh
        修改：     ../test/test-all-defconfigs.sh
        删除：     ../zephyrproject/modules/audio/sof

未跟踪的文件:
  （使用 "git add <文件>..." 以包含要提交的内容）
        ../test/cmocka_git/

修改尚未加入提交（使用 "git add" 和/或 "git commit -a"）
larryba@debian:~/project/zephyrproject/modules/audio/sof/scripts$ git diff run-mocks.sh
diff --git a/scripts/run-mocks.sh b/scripts/run-mocks.sh
index 61cb9a7d3..20324d1f8 100755
--- a/scripts/run-mocks.sh
+++ b/scripts/run-mocks.sh
@@ -24,7 +24,7 @@ EOF
 parse_opts()
 {
        # default config if none supplied
-       CONFIG="tgph_defconfig"
+       CONFIG="csk6_defconfig"
        VALGRIND_CMD=""

        while getopts "vc:" flag; do
@@ -49,9 +49,24 @@ rebuild_ut()
         # -DINIT_CONFIG is ignored after the first time. Invoke make (or
         # -ninja) after this for faster, incremental builds.
        rm -rf build_ut/
+       OLDPATH=$PATH
+        XTENSA_TOOLS_ROOT="/home/larryba/hifi_install/xtensa/XtDevTools"
+       XTENSA_CORE="venus_hifi4"
+       XTENSA_TOOLS_VERSION="RI-2020.4-linux"
+       XTENSA_BUILDS_DIR="$XTENSA_TOOLS_ROOT/install/builds/$XTENSA_TOOLS_VERSION"
+       #TOOLCHAIN=xt
+       ROOT="$XTENSA_BUILDS_DIR/$XTENSA_CORE/xtensa-elf"
+       export XTENSA_SYSTEM=$XTENSA_BUILDS_DIR/$XTENSA_CORE/config
+       printf 'XTENSA_SYSTEM=%s\n' "${XTENSA_SYSTEM}"
+       XTENSA_TOOLS_DIR="$XTENSA_TOOLS_ROOT/install/tools/$XTENSA_TOOLS_VERSION"
+       PATH=$XTENSA_TOOLS_DIR/XtensaTools/bin:$OLDPATH
+       #COMPILER="xcc"
+       #printf "COMPILER %s\n" "$COMPILER"
+       #cmake -S "$SOFTOP" -B build_ut -DBUILD_UNIT_TESTS=ON -DBUILD_UNIT_TESTS_HOST=OFF -DTOOLCHAIN=xt -DROOT_DIR=/home/larryba/hifi_install/xtensa/XtDevTools/install/builds/RI-2020.4-linux -DINIT_CONFIG=csk6_defconfig \
+       #     -DINIT_CONFIG="${CONFIG}" -DROOT_DIR=/home/larryba/hifi_install/xtensa/XtDevTools/install/builds/RI-2020.4-linux/venus_hifi4/xtensa-elf

-       cmake -S "$SOFTOP" -B build_ut -DBUILD_UNIT_TESTS=ON -DBUILD_UNIT_TESTS_HOST=ON \
-             -DINIT_CONFIG="${CONFIG}"
+       cmake -S "$SOFTOP" -B build_ut -DBUILD_UNIT_TESTS=ON -DBUILD_UNIT_TESTS_HOST=OFF -DBUILD_UNIT_TESTS_XTENSA_GCC=OFF -DINIT_CONFIG="${CONFIG}" -DTOOLCHAIN=xt \
+             -DROOT_DIR="$XTENSA_BUILDS_DIR/$XTENSA_CORE/xtensa-elf"

        cmake --build build_ut -- -j"$(nproc --all)"
 }
larryba@debian:~/project/zephyrproject/modules/audio/sof/scripts$

**To Reproduce**
1. with zephyrworkspace:
larryba@debian:~/project/zephyrproject$ ls
bootloader  build_csk6_xcc  build_ut  modules  tools  zephyr
larryba@debian:~/project/zephyrproject$./modules/audio/sof/run_cmocks.sh
[sof run cmock with xt-xcc error.txt](https://github.com/thesofproject/sof/files/7731648/sof.run.cmock.with.xt-xcc.error.txt)

please check my attachment file.

bawanqin1987, 2021-12-17:
the modify i attach the file,So you can reproduce the problem
[modify run cmocks sh file.txt](https://github.com/thesofproject/sof/files/7731684/modify.run.cmocks.sh.file.txt)

marc-hb, 2021-12-17:
Thanks for the report, so much clearer than #5098.

As I wrote at the bottom of #5098, I'm not aware of any CMake wrapper script that supports cross-compiled unit tests.

First things first: does it work when you follow the official documentation and use CMake directly? https://thesofproject.github.io/latest/developer_guides/unit_tests.html

bawanqin1987, 2021-12-17:
of course run cmock support ,also i check the unit_tests.html
1.if we use the xtensa crosstool,we should config some development :
Export XTENSA_TOOLS_ROOT as you normally do when building the firmware.

Build the firmware using ./xtensa-build-all.sh and take note of the following variables in the build log: PATH, XTENSA_SYSTEM and the -DROOT_DIR parameter.

export the PATH and XTENSA_SYSTEM values found above.

Run cmake with -DBUILD_UNIT_TESTS=ON, the -DROOT_DIR parameter above, -DINIT_CONFIG and a new build directory

Build and run the tests with make test or ninja test.

2. the unit test html show: 
mkdir build_ut
cmake -DBUILD_UNIT_TESTS=ON -DTOOLCHAIN=xt -DINIT_CONFIG=apollolake_defconfig \
    -DROOT_DIR=/xcc/install/builds/RG-2017.8-linux/X4H3I16w2D48w3a_2017_8/xtensa-elf ..
make -j4 && ctest -j8   
and this command,it's same to run cmock sh script ,as below:
[image]
The run cmock script also use cmake command ,first build,then ctest run,it's same 1 issue reference command,u can check it.

3.i check the script the process,from the whole ,it's support the host and crosstoolchain configs. it's sample and clearly excution.

marc-hb, 2021-12-17:
You lost me sorry. Can you answer this question with "yes" or "no"?

> First things first: does it work when you follow the official documentation and use CMake directly? https://thesofproject.github.io/latest/developer_guides/unit_tests.html

bawanqin1987, 2021-12-17:
Even if you compile an executable using cross-compilation, just because it doesn't run locally doesn't mean it can't compile  
but now it's build error .some files preprocesser report datatype check error...i think these files 
syntax does not adapter with our xtensa crosstool chain...

for another,i had already use script xtensa-build-zephyr.sh/xtensa-build-all.sh success for building.

bawanqin1987, 2021-12-17:
> You lost me sorry. Can you answer this question with "yes" or "no"?
> 
> > First things first: does it work when you follow the official documentation and use CMake directly? https://thesofproject.github.io/latest/developer_guides/unit_tests.html

yes.i had already tried.host on is ok .host off and xtensa xt-xcc/xt-gcc  crosstoolchain is not pass.

marc-hb, 2021-12-17:
Good. So there is no bug, I'm removing the "bug" label. I don't understand everything you wrote but I believe you are requesting a more convenient way to cross-compile and run the unit tests. This is a valid request but it's not a priority right now. If you find how to do it, please submit a pull request.

bawanqin1987, 2021-12-17:
wait.but and you have method to use the https://thesofproject.github.io/latest/developer_guides/unit_tests.html
with xtensa crosstoolchain ,The documentation says yes, but actually the command used in the documentation failed.
so That's why I tried to modify run Cmock because I saw that the process supports cross-compilation .
so ,how i can i do using xtensa crosstool with cmock case??
the unit test html show:
mkdir build_ut
cmake -DBUILD_UNIT_TESTS=ON -DTOOLCHAIN=xt -DINIT_CONFIG=apollolake_defconfig
-DROOT_DIR=/xcc/install/builds/RG-2017.8-linux/X4H3I16w2D48w3a_2017_8/xtensa-elf ..
make -j4 && ctest -j8

marc-hb, 2021-12-17:
You wrote:

> >  First things first: does it work when you follow the official documentation and use CMake directly?
> >  https://thesofproject.github.io/latest/developer_guides/unit_tests.html

> yes.i had already tried.

But now you are saying "no"? This is extremely hard to follow.

bawanqin1987, 2021-12-17:
i mean unit_tests.html ,host tool gcc is ok to support ,but the xtensa toolchain failed to follow the documentation recommendations.

how can i do?

bawanqin1987, 2021-12-17:
Example: Running tests for APL¶
mkdir build_ut && cd build_ut
cmake -DBUILD_UNIT_TESTS=ON -DTOOLCHAIN=xt -DINIT_CONFIG=apollolake_defconfig \
    -DROOT_DIR=/xcc/install/builds/RG-2017.8-linux/X4H3I16w2D48w3a_2017_8/xtensa-elf ..
make -j4 && ctest -j8

and this example,I think the description here is problematic,when we create the ut_test dir and cd the dir,there is no cmakelist,Of course, the cmake command cannot be executed

marc-hb, 2021-12-21:
> and this example,I think the description here is problematic,

What problem? The error messages in the `sof run cmock with xt-xcc error.txt` attachment are in Chinese. The common language for this project is English.

If you want others to understand your bugs, you must provide the absolute minimum steps required to reproduce. Build with a single thread `-j1` so there is a single error message not interleaved with unrelated stuff, then provide the concise error message in the description, not in a large attachment with a ton of noise. 

> cmake -DBUILD_UNIT_TESTS=ON -DTOOLCHAIN=xt -DINIT_CONFIG=apollolake_defconfig
-DROOT_DIR=/xcc/install/builds/RG-2017.8-linux/X4H3I16w2D48w3a_2017_8/xtensa-elf ..
> when we create the ut_test dir and cd the dir,there is no cmakelist,Of course, the cmake command cannot be executed

"Of course" there is `../CMakeLists.txt`.

I tried the instructions again and I found a `uintptr_t` error. I submitted a workaround in https://github.com/thesofproject/sof-docs/pull/395. With this workaround I can build with XCC without any other issue.

bawanqin1987, 2021-12-21:
sorry , i can not found where excute the command :
from the URL : https://thesofproject.github.io/latest/developer_guides/unit_tests.html :
i check the steps:
Example: Running tests for APL
1. mkdir build_ut && cd build_ut (i can not understand why cd build_ut??and where can i mkdir the build_ut dir??)
2. cmake -DBUILD_UNIT_TESTS=ON -DTOOLCHAIN=xt -DINIT_CONFIG=apollolake_defconfig \
    -DROOT_DIR=/xcc/install/builds/RG-2017.8-linux/X4H3I16w2D48w3a_2017_8/xtensa-elf .. 
make -j4 && ctest -j8

lgirdwood, 2021-12-21:
@mengdonglin @marc-hb I wonder if we have a configuration difference between the xt tooling or the environment that Intel is using and the xt tooling / environment that @bawanqin1987 is using ? This could explain why it's not working at all for @bawanqin1987.
@mengdonglin could someone with the xt tooling run the cmocka script and past all the commands and the output here so that @bawanqin1987 can compare the output generated with his output ?

aiChaoSONG, 2021-12-23:
@bawanqin1987 I am not able to reproduce the issue you reported here. Below is the output of `run-mocks.sh`. the tests are successfully built.
[cmocka_build_with_xtensa.txt](https://github.com/thesofproject/sof/files/7766660/cmocka_build_with_xtensa.txt)
```
➜  sof git:(main) ✗ readelf -h ./build_ut/test/cmocka/src/list/list_item_append
ELF Header:
  ...
  Machine:                           Tensilica Xtensa Processor
  ...
```
I used the script you modified, the difference are the toolchain version and the processor config. In the log you attached, `malloc.h` is not found, and this file is from the processor config(builds). Maybe your processor config is not properly installed?

bawanqin1987, 2021-12-23:
i used the hifi4 XtDevTools\install\builds\RI-2020.4-win32\venus_hifi4，and this tool i used the script xtensa-build-test.sh
/xtensa-build-zephyr.sh with xtensa toolchain is ok to build.

aiChaoSONG, 2021-12-23:
@bawanqin1987  can you find the file `malloc.h` under `.../venus_hifi4/xtensa-elf/include`?

bawanqin1987, 2021-12-23:
[image]

bawanqin1987, 2021-12-23:
so there is no malloc.h ,but this xtensa_elf,We have already used this version of the core configuration file and it is all good 
but sof cmock used.

aiChaoSONG, 2021-12-23:
@bawanqin1987  somewhere in #5098, you seem to enable clang, maybe you should try to enable it here in the cmoka test?

bawanqin1987, 2021-12-23:
Xt-xcc/XT-clang both toolchains have been used, except for the xT-XCC build here .
you mean i need config xt-clang?

bawanqin1987, 2021-12-23:
which files i need to modify?but i can not found sof/cmock support xt-clang .

aiChaoSONG, 2021-12-23:
I mean export XCC_USE_CLANG like what you do in the below picture.
[image]

please try to remove `XCC_USE_CLANG=1` from xtensa-build-all.sh, and see you still build SOF successfully?

bawanqin1987, 2021-12-23:
i know , I mean other files also have related processing, I see only XCC processing in the Cmake file  ：
[image]

bawanqin1987, 2021-12-23:
XCC_USE_CLANG=1  i check this mean use xt_clang

bawanqin1987, 2021-12-23:
Is the correct setting，You want me to remove it and then./run cmock.sh?

bawanqin1987, 2021-12-23:
when i remove the XCC_USE_CLANG=1,and run the script run_cmocks.sh ,it 's also error:
[image]

bawanqin1987, 2021-12-23:
[image]
build all,before run cmock ,it's always sucess,no error,With or without XCC USE CLANG!

aiChaoSONG, 2021-12-23:
@bawanqin1987 I think your environment for building SOF/Zephyr is correct. and the scripts in sof repo are also good. The problem here is that when you want to build cmocka test,  the tests uses `malloc.h`, but it doesn't exist in your processor config.

aiChaoSONG, 2021-12-23:
@bawanqin1987 for the tests that include `malloc.h`, for example sof/test/cmocka/src/audio/mixer/mixer_test.c, can you replace `#include <malloc.h>` with `#include <stdlib.h>` and try again?

aiChaoSONG, 2021-12-23:
@bawanqin1987  Please help to check if #5126 fixes your issue.

bawanqin1987, 2021-12-23:
ok, i will merge the malloc build error issue, but still have others error!....please check my attachment build log.
[image]

aiChaoSONG, 2021-12-23:
Please don't attach a lot of pictures, instead, upload the full log in a text file, this gives more context.

bawanqin1987, 2021-12-23:
[Uploading cmock build error with xtensa tool.txt…]()

bawanqin1987, 2021-12-23:
[Uploading stdint.txt…]()

bawanqin1987, 2021-12-23:
[image]
it's cadence hifi4 sdk package include files,we can not modify it.

bawanqin1987, 2021-12-23:
We don't have permission to modify it,So if we need to modify this bug, we need sof to adapt.

aiChaoSONG, 2021-12-24:
@bawanqin1987  could you please check the files you uploaded here

> Uploading cmock build error with xtensa tool.txt…

and here

> Uploading stdint.txt…

They are not uploaded at all.

PR #5126 (2021-12-25): test: improve compatibility for malloc.h
The C header file malloc.h does not exist on
some Xtensa processor configurations or host OSes,
include stdlib.h instead because they both provide
the prototype for 'malloc' function.

Fixes: #5102
commit b673d4a3 (2021-12-25): test: improve compatibility for malloc

The C header file malloc.h does not exist on
some Xtensa processor configurations or host OSes,
include stdlib.h instead because they both provide
the prototype for 'malloc' function.

Fixes: #5102
closed 2021-12-25 by lgirdwood