thesofproject/sof #5112: Realtek ALC711-CG in Dell Precision 5760 does not work

Thorval, 2021-12-18:
Dear SOF developers,

I have a new Dell Precision 5760 laptop with a built-in Realtek ALC711-CG sound card which does not work with the SOF firmware. This sound card is apparently built into new Dell XPS and Precision laptops and probably will affect many users.
I am running Debian 11 "Bullseye" with a 5.10 kernel. I have tried the SOF firmware versions 1.7.1, 1.9.1 from Debian repositories and version 1.9.3 from this site. The Debian maintainer of the firmware-sof-signed package could not help and suggested that I should contact the upstream developers.

Here is the lspci output for this sound card:

# lspci -k -nn -v -s 00:1f.3
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Device [8086:43c8] (rev 11)
        Subsystem: Dell Device [1028:0a5e]
        Flags: bus master, fast devsel, latency 64, IRQ 16, IOMMU group 17
        Memory at 618d1d8000 (64-bit, non-prefetchable) [size=16K]
        Memory at 618d000000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [50] Power Management version 3
        Capabilities: [80] Vendor Specific Information: Len=14 <?>
        Capabilities: [60] MSI: Enable- Count=1/1 Maskable- 64bit+
        Kernel driver in use: sof-audio-pci
        Kernel modules: snd_hda_intel, snd_sof_pci

This is the output of 'dmesg | egrep "snd|sof|audio"':

[    3.456411] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.456438] snd_hda_intel 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[    3.474643] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.474660] sof-audio-pci 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[    3.474704] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.474920] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.480694] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.487680] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.707791] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 4
[    3.708557] sof-audio-pci 0000:00:1f.3: firmware: direct-loading firmware intel/sof/sof-tgl.ri
[    3.708562] sof-audio-pci 0000:00:1f.3: warning: unknown sof_ext_man header type 6 size 0x20
[    3.708562] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:9:0-fa857
[    3.708563] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:19:0 Kernel ABI 3:17:0
[    3.708564] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    3.708566] sof-audio-pci 0000:00:1f.3: warning: unknown sof_ext_man header type 3 size 0x30
[    3.708566] sof-audio-pci 0000:00:1f.3: warning: unknown sof_ext_man header type 5 size 0x20
[    5.786354] sof-audio-pci 0000:00:1f.3: error: firmware boot failure
[    5.787166] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -5
[    5.789199] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -5

An "aplay -l" does not find any sound card. A "pavucontrol" only shows a non-functional "dummy" sound card.

I have also tried the newer 5.14 kernel from the Bullseye backports repository with SOF 1.7.1, but had to revert to the 5.10 kernel because of problems with the Nvidia kernel module.
I have also tried the following kernel parameters upon booting: "snd-hda-dspcfg.dsp_driver=0", "1", "2" and "3" without success. Passing "1" boots without any "sof"-error, an "aplay -l" shows 10 HDMI devices, but "pavucontrol" still shows only a non-functional "dummy" sound card.

Could you please check this? I would be happy to provide any additional information that you need!

Best regards,

Dirk.

mengdonglin, 2021-12-21:
@Thorval Dell Precision 5760 is a supported device model of SOF. We'll double check mainline kernel v5.10 and SOF upstream versions and come back to you.

lgirdwood, 2021-12-21:
> [ 5.786354] sof-audio-pci 0000:00:1f.3: error: firmware boot failure

@Thorval Not sure why FW would fail to boot here, Linux 5.10 should have all the support for TGL DSP and soundwire. Did 5.14 work with SOF v1.9.3 ?
@plbossart @ranj063 @kv2019i any ideas ?

kv2019i, 2021-12-21:
Hmm, this seems odd. Only change affecting the boot are the ICCMAX patches, but those are in 5.10, so should be good. 

Just to double check, @Thorval can you check that the Intel Management Engine (ME) is not disabled in BIOS. Please see https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html#user-space-and-filesystem-requirements and https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#make-sure-the-me-is-enabled 

Another is to try enabling more logs from kernel with dyndbgs:
https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#enable-dynamic-debug
I'd expect to see multiple tries to boot the DSP and dump of DSP state when failure is seen.

And last, one easy test to do is to try a Fedora 35 live image and see whether the FW boot proceeds further. Just to double-check nothing is missing from the Debian kernel.

Thorval, 2021-12-22:
Many thanks for your suggestions, of which the following two led to a working sound card!

1), Fedora 35 Live Session
Kernel 5.14.10, SOF firmware 1.8.2
I had to choose the SOF soundwire speaker in the Gnome3 settings.

2). Debian "Bullseye" with Kernel from Backports and SOF firmware from "Testing"
Kernel 5.14.0, SOF firmware 1.9.1
Worked out-of-the-box in KDE Plasma5.

So, at least a Kernel 5.14 together with SOF >= 1.8.2 worked!
The outputs of lspci, aplay and dmesg were practically identical in both tests: Here are the outputs from the second test:
 $ lspci -v -nn -k -s 00:1f.3
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Device [8086:43c8] (rev 11)
        Subsystem: Dell Device [1028:0a5e]
        Flags: bus master, fast devsel, latency 64, IRQ 206, IOMMU group 17
        Memory at 618d1d8000 (64-bit, non-prefetchable) [size=16K]
        Memory at 618d000000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: <access denied>
        Kernel driver in use: sof-audio-pci-intel-tgl
        Kernel modules: snd_hda_intel, snd_sof_pci_intel_tgl
$ aplay -l
**** Liste der Hardware-Geräte (PLAYBACK) ****
Karte 0: sofsoundwire [sof-soundwire], Gerät 0: Jack Out (*) []
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: sofsoundwire [sof-soundwire], Gerät 2: Speaker (*) []
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: sofsoundwire [sof-soundwire], Gerät 5: HDMI 1 (*) []
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: sofsoundwire [sof-soundwire], Gerät 6: HDMI 2 (*) []
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: sofsoundwire [sof-soundwire], Gerät 7: HDMI 3 (*) []
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0

$ dmesg | egrep "sof|snd|audio"
[    2.696709] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.696795] snd_hda_intel 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[    2.930778] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.930855] sof-audio-pci-intel-tgl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[    2.930942] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.931211] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.933098] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    2.939875] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    3.144881] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 4
[    3.147576] sof-audio-pci-intel-tgl 0000:00:1f.3: firmware: direct-loading firmware intel/sof/sof-tgl-h.ri
[    3.147582] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 1:9:0-fa857
[    3.147583] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:19:0 Kernel ABI 3:18:0
[    3.147585] sof-audio-pci-intel-tgl 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    3.147588] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[    3.233002] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 1:9:0-fa857
[    3.233005] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:19:0 Kernel ABI 3:18:0
[    3.233006] sof-audio-pci-intel-tgl 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    3.357223] sof-audio-pci-intel-tgl 0000:00:1f.3: firmware: direct-loading firmware intel/sof-tplg/sof-tgl-rt711-rt1308-rt715.tplg
[    3.357231] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:19:0 Kernel ABI 3:18:0
[    3.357233] sof-audio-pci-intel-tgl 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    3.373498] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    4.500903] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
[    4.500906] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 4
[    4.500907] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 5
[    4.500907] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 6
[    4.500908] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 7
[    4.500909] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 8
[    4.518828] input: sof-soundwire Headset Jack as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input16
[    4.518855] input: sof-soundwire HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input17
[    4.518876] input: sof-soundwire HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input18
[    4.518893] input: sof-soundwire HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input19

If you need any additional information, I would be happy to provide them!

Best regards!

XiaoyunWu6666, 2021-12-23:
> Linux 5.10 should have all the support for TGL DSP and soundwire

Hi @Thorval  

Mainline kernel 5.10 may not 

FYI, if you have to use kernel 5.10 , then may you try kernel of this tag 'release-sof-tgl-h-beta' ?

Link :https://github.com/thesofproject/linux/tree/release-sof-tgl-h-beta
I've tried kernel built with this tag (it's a 5.10 rc5)  + firmware of v1.5 (or even the newest 2.0) , the sound card works.

libinyang, 2021-12-28:
@Thorval As per the link @XiaoyunWu6666 pointed, please try to backport the commit https://github.com/thesofproject/linux/commit/a0cbd7bf112eccbc81c2d94261607e5f4f678378 and have a try.

Thorval, 2021-12-30:
Sorry, compiling and testing a kernel is beyond what I am able to do. Meanwhile, I have changed to the more recent Debian "Bookworm", where both sound card and graphics driver work with the more recent Kernel 5.15. I have informed the Debian SOF firmware package maintainer about a possible problem with Kernel 5.10 in Debian "Bullseye".
Many thanks again for your suggestions!

plbossart, 2022-01-10:
@Thorval I don't think we can make this TigerLake device work with 5.10, there are just too many changes that happened for that platform in 2021.

plbossart, 2022-12-12:
no update in almost a year, closing

closed 2022-12-12 by plbossart